hildogjr/KiCost #146: Mouser web scraping not working

xesscorp, 2018-01-15:
I changed the web scraper to account for the new format of Mouser pages, but the scraping is still not working. This may be because Mouser is blocking my IP address because they think I'm a robot. Somebody else will have to try scraping and see if the Mouser information is appearing in the spreadsheet.

hildogjr, 2018-01-15:
This explain, on my tests, why Mouser was always returning `None`.

xesscorp, 2018-01-16:
Previous tests with Mouser were returning no data because they changed their web page format so it didn't match the scraper patterns. Current lack of data is caused by something else.

hildogjr, 2018-01-17:
For some minor components is working, so I don't the extensions of the page change.
manf# = 3296Z-1-254LF  --> 652-3296Z-1-254LF 
manf# = 1888687 --> 651-1888687

hildogjr, 2018-03-05:
Today, 2018-03-04 BR clock, I had no more problem with Mouser scrape module.

closed 2018-03-06 by hildogjr