hildogjr/KiCost #178: New release on Pypi?

xesscorp, 2018-02-10:
When do you think you would like to make a new release on Pypi?

hildogjr, 2018-02-10:
Hi @xesscorp, I would like (utopian idea) to fix the all three bugs #174, #161 and #146 and also finish #144 enhancement, but I think it is may be long way.
About the bug #174, I don't know how many BOM files (in the test folder) is being affected by. The #161 and #146, mainly, is annoying me.
So, we could focus fix the bugs, even if the GUI is not full functional yet, postpone it and leave as alpha version for now.

About the GUI, I would like to modify the `setup.py` [in some way](https://stackoverflow.com/questions/18725137/how-to-obtain-arguments-passed-to-setup-py-from-pip-with-install-option), maybe in one next release, to be possible to install more dependences when is asked to install GUIDE (install and check wxPython, maybe create a desktop shortcut) or a developer version of KiCost (may install some additional debug tool in the future), having some installation as:

    pip install kicost # Default installation.
    pip install --install-option="--gui" kicost # To install GUI interface dependences.
    pip install --install-option="--dev" kicost # To install developer debug dependences.

About the CSV submodule and multifile functionality, I tested with KiCad, hand made CSV and others EDA softwares CSVs and was able to create a integrated unique spreadsheet (feature interesting for my country because of the customs duties).

hildogjr, 2018-03-05:
@xesscorp , @devbisme
I tested now and got a pretty good scrape on Mouser (maybe the module still need some enhancement #146 ).
The man issue now, still #161, that had the web page changed recently.

hildogjr, 2018-03-07:
I think, just fixing the RS module will be ok to new release.
Other important think is update the page https://xesscorp.github.io/KiCost/docs/_build/singlehtml/index.html

xesscorp, 2018-03-07:
I've tested the latest release. It fails on tests/StickIt-Hat.xml. That file is processed correctly on the current 0.1.42 PyPi release. The problem is caused by some change related to handling custom part data.

hildogjr, 2018-03-08:
Fixed.
I think this happened when I ask to move way all the web routines from the `kicost.py`.

hildogjr, 2018-03-10:
Ok @diorcety.
Check and release (I will not merge new code). I start to study the scrape module to be able to fix the RS one. And I think there are some improvements to be made and standardize the codes. I am concerned about the `Currency_Converter` library used in some too.

xesscorp, 2018-03-12:
I'm checking commit 95f4bfe58 and it seems to pass my tests. Good job!

Have you added all the changes for this release to HISTORY.rst? Has the docs/usage.rst been updated with all the new features? Once those are done, is it time to release 0.1.43?

hildogjr, 2018-03-12:
@xesscorp, yes. I was always adding the new feature to HISTORY (leaving as possible 0.1.43) and updating USAGE.rst.
I just checked them now and fixed some spell errors. You can release.

xesscorp, 2018-03-15:
KiCost 0.1.43 has been released to PyPI.

closed 2018-03-15 by xesscorp