hildogjr/KiCost #184: Format Designators list with comma separated individual entries

whatnick, 2018-02-25:
I am using Kicost generated BOM directly in MacroFab. However MacroFab parser cannot grasp the mixed designator lists with comma separated entries and range entries, however macrofab does use a similar format when showing the imported data. Would a command line switch be possible to disable the compact range formatting and include every designator in the  Refs field.
[image]
 See above for partially understood KiCost generated BOM in MacroFab.

hildogjr, 2018-02-26:
Hi @whatnick, until we analyze the need or extensions of the change, you can comment the lines

    # Convert the list of numbers for each ref prefix into ranges.
    for prefix in list(prefix_nums.keys()):
        prefix_nums[prefix] = convert_to_ranges(prefix_nums[prefix])

on `collapse_refs()` from the `eda_tools.py` file and achieve this.

whatnick, 2018-02-27:
Thanks for the tip. I will make a fork and a command line option for my
MacroFab purposes. The tip can remain in the issue for others having
similar roadblocks.

hildogjr, 2018-02-27:
The modification needed appear to be small. But before do this in the main branch I am waiting to see if is enough for you (to make compatible with MacroFab, I didn't know this company) and if more user relate something related.
I was think to create some `--not_collapse` parameter in the call, what you think?
Cold you see if this modification in the references help with other services like MacroFab?

whatnick, 2018-02-27:
Hi,

I will check with Seeedstudio and Aisler. Who are the other services who
offer automatic BOM parsing and assembly job creation.

Regards,

Tisham.

hildogjr, 2018-02-28:
Thanks @whatnick.
Tell me after if this feature/command make compatible with all them.

commit 766ccc4f (2018-03-03): Merge pull request #185 from hildogjr/master

Fixed #184, #126 (#110) and #77
commit df1b74e9 (2018-03-03): Fixed #184 and improved GUI and other thinks.
closed 2018-03-03 by hildogjr