hildogjr/KiCost #194: [BUG] Excel file invalid when not performing distributors search.

mdeweerd, 2018-03-11:
I run KiCOST while excluding all distributors to check that all grouping, part numbers, etc are ok before scraping.

I get the following error with Excel:

[image]

Excel can repair the file, so this is not extremely critical but it is of course a limitation.

I run kicost 1.42 like this:

> kicost -d 10 -i Manettes.xml -e digikey newark mouser farnell rs tme -o BOM_9309dummy.xlsx -var '^(.*)$' --overwrite -np 4 -f comment S1MN S1PN S2MN S2PN

The "Manettes.xml" file is enclosed:
[Manettes.zip](https://github.com/xesscorp/KiCost/files/1800427/Manettes.zip)

hildogjr, 2018-03-12:
Hi @mdeweerd.
Are you using the last official release or the last GitHub code? (If last release, could you check with the last GitHub code? You can use `--no_scrape` instead of `--exclude digikey mouser ... allthem`).
I have some clue of the source of this error, but I could reproduce it on LibreOffice.

hildogjr, 2018-03-12:
I did some improvement to deal with this specific case of just "assembly list".
[BOM_9309dummy.xlsx](https://github.com/xesscorp/KiCost/files/1801332/BOM_9309dummy.xlsx)
Check it in Excel, please.

mdeweerd, 2018-03-12:
@hildogjr 
I was using the latest Github version & the file you joined in your last message opens in Excel without a probelm.

hildogjr, 2018-03-12:
So, fixed on merge https://github.com/xesscorp/KiCost/commit/3401830412545258815e7cd72aa9628d9f6958c9.

commit 53063ec8 (2018-03-12): Fixed #194 on GUI
commit 95f4bfe5 (2018-03-12): Merge pull request #197 from hildogjr/master

Fixed #194 on GUI
closed 2018-03-12 by hildogjr