hildogjr/KiCost #195: Variant field, desc field not ignored?

mdeweerd, 2018-03-11:
I run KiCost 1.42 like this on the attached file:

`kicost -d 10 -i Manettes.xml -e digikey newark mouser farnell rs tme -o BOM_3370dummy.xlsx -var '^(.*)$' --overwrite -np 4 --ignore_fields variant -f comment S1MN S1PN S2MN S2PN`

Notice that it is requested that the 'variants' field.
Yet the generated BOM file does not group the 1uF capacitors.

Also, the 1.41 documentation says:

> KiCost groups similar parts together and places their information on a single line of the generated spreadsheet. For parts to be grouped, they must:
> 
> - come from the same library (e.g., “device”),
> - be the same part (e.g., “R”),
> - have the same value (e.g., “10K” but note that this would not match “10000” or “10K0”), and
> - have the same footprint (e.g., “Resistors_SMD:R_0805_HandSoldering”).

It is not indicated that they should have the same description, the same variant, etc.

As you can see, the description field is empty.

When I request KiCost to ignore the "Desc" field to allow grouping on it, it fails:
`kicost -d 10 -i Manettes.xml -e digikey newark mouser farnell rs tme -o BOM_3734dummy.xlsx -var '^(.*)$' --overwrite -np 4 --ignore_fields desc variant -f comment S1MN S1PN S2MN S2PN`

> Traceback (most recent call last):
>   File "/usr/local/bin/kicost", line 9, in <module>
>     load_entry_point('kicost==0.1.42', 'console_scripts', 'kicost')()
>   File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.42-py2.7.egg/kicost/__main__.py", line 260, in main
>     scrape_retries=args.retries, throttling_delay=args.throttling_delay)
>   File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.42-py2.7.egg/kicost/kicost.py", line 262, in kicost
>     user_fields, '-'.join(variant) if len(variant)>1 else variant[0])
>   File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.42-py2.7.egg/kicost/spreadsheet.py", line 174, in create_spreadsheet
>     parts, user_fields, collapse_refs)
>   File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.42-py2.7.egg/kicost/spreadsheet.py", line 413, in add_globals_to_worksheet
>     desc_col = columns['desc']['col']
> KeyError: 'desc'

I would like to join all 1uF capacitors (in this example) on the same line.

[Manettes.zip](https://github.com/xesscorp/KiCost/files/1800522/Manettes.zip)
[Manettes_dummy.xlsx](https://github.com/xesscorp/KiCost/files/1800525/Manettes_dummy.xlsx)

hildogjr, 2018-03-14:
@mdeweerd try the last modification in my personal branch (using the `--group_fields` and not the `--ignore_fields`). I am waiting @xesscorp do the v0.1.43, this fix will be in the v0.1.44 next month, when I want to, by default, merging the description, as your understanding of the user documentation (that I think is the correct).

hildogjr, 2018-03-14:
I used:

     kicost -i ./Manettes.xml -q -w --no_scrape -grp var desc libpart -var '^(.*)$' -grp variant desc

hildogjr, 2018-03-17:
@mdeweerd, I change more the code (for the next version, probably, but available on my Git). This version deal automatically with this situation, not need `-grp` of `--ignore_field`.

mdeweerd, 2018-03-17:
Hi @hildogjr 
Thank you for the update - I may wait for the update to test this.
Kind regards
Mario

mdeweerd, 2018-03-21:
Hi @hildogjr 
I tried the version from you hildogr/master today.
The command
 kicost -d 10 -i Manettes.xml -e digikey newark mouser farnell rs tme -o BOM_3501dummy.xlsx -var '^(.
*)$' --overwrite -np 4 -f comment S1MN S1PN S2MN S2PN

Generated this:
[image]

- There are two lines with 470nF values (one starting with C10, the other starting with C7).
- C19 has the desc field filled in, the others have no desc field - so the other should "inherit" the desc field from C19, not be shown as an empty "desc".

Other lines have similar behaviors.

[Other "issue": when opening the excel file, the multiline fields appear as 1 line.  I have to enter the field and fake editing it to have two lines shown
Here is the XML file:
[ManettesXML.zip](https://github.com/xesscorp/KiCost/files/1834257/ManettesXML.zip)].

hildogjr, 2018-03-21:
Missing just the `desc` multiline issue.

hildogjr, 2018-03-22:
All fixed, was just a missing `[...]`.
Other tip: use `--no_scrape` instead of `-e digikey newark mouser farnell rs tme`

commit 25e17899 (2018-03-14): Merge pull request #200 from hildogjr/master

Fix #199, Fix #195
commit b9a742f9 (2018-03-14): Fix #199, Fix #195
commit c325f01a (2018-03-21): Fixed #195, missing `desc` issue
commit d7f3bd95 (2018-03-21): Merge pull request #210 from hildogjr/master

Fixed #195, missing `desc` issue
closed 2018-03-22 by hildogjr