hildogjr/KiCost #201: Use of multi-digit quantities in part number field corrupts spreadsheet

xesscorp, 2018-03-14:
In the attached XML file (attached as a .txt file), I use a quantity in the manf# field for J3 as follows:

    <comp ref="J3">
      <value>BLE113_Debug_Header</value>
      <footprint>Connectors:CNC-Tech_3220-10-0300-00</footprint>
      <fields>
        <field name="manf#">10:3220-10-0300-00</field>
      </fields>
      <libsource lib="Connectors" part="BLE113_Debug_Header"/>
      <sheetpath names="/" tstamps="/"/>
      <tstamp>57AA8AC8</tstamp>
    </comp>

Using the command:

    kicost -i multipart2.xml -qw --inc digikey

creates a corrupted spreadsheet (attached) with multiple cost-calculation and number-of-parts-found cells that overlap the part data area.

Increasing the quantity in the manf# field (say from 10 to 100 to 1000) leads to increasing amounts of corruption in the spreadsheet. The spreadsheet is fine, however, as long as the quantity is a single digit. I've traced the problem to the following section of eda_tools.py (lines 328-336):

            for key, val in list(components[ref].items()):
                if key == 'manf#_qty':
                    try:
                        for i in range(len(val)):
                            grp_fields['manf#_qty'][i] += '+' + val[i] # DUMMY way and need improvement to realy do arithmetic and not string cat. #TODO
                            val[i] = grp_fields['manf#_qty'][i] # Make the firt values take also equal.
                    except:
                        grp_fields['manf#_qty'] = val
                    continue

The problem is on line 331 which takes the length of `val` which is the subpart quantity. Since that quantity is stored as a string returned by the `manf_code_qtypart()` function, the `for` loop seems to be processing each character of the quantity. I don't know why that is done. The spreadsheet is corrupted whenever the subpart quantity has more than one digit.

I don't understand this code, so I can't fix it. This is the last problem I've found. Once this is fixed, I can release version 0.1.43.

[multipart2.txt](https://github.com/xesscorp/KiCost/files/1811063/multipart2.txt)

[multipart2.xlsx](https://github.com/xesscorp/KiCost/files/1811044/multipart2.xlsx)

hildogjr, 2018-03-14:
The error was in `spreadsheet.py`. When checked `len(field('manf#_qty')` for 10 as quantity, was returning `2`.
First I needed to test if `field('manf#_qty')` was as `isinstance( ..., list)`, if don't, the length (quantity of files read is `1`).

hildogjr, 2018-03-14:
About the `#DUMMY #TODO`, it is because a quantity sum is generated concatenating the quantities of each part when group, e.g.: `1+2+1/4+0.5+10`.
I think is a little dummy, because could produce `12+1/4+0.5`. That is a formula more simple to Excel / LibreOffice, so could need some improvement, but is full functional.

commit 340783be (2018-03-14): Fixed #201
closed 2018-03-14 by hildogjr