hildogjr/KiCost #215: Exit all threads when close the main one

hildogjr, 2018-03-24:
When KiCost scrape using multithread, create multiple processes to the OS, and when one is force (manually) to be finalized occur a error. That's ok, but if the main process is closed all the others have to be force to be close, for safety (this on CLI mode yet).
This will be also interesting for next enhancement on the GUIDE.

hildogjr, 2018-06-02:
Fix on @mmmaisel thread configuration.
Now is enrolled just one process to the OS with multiple threads, this saves the computer memory use (validated on Linux).

New issue founded in `num_processes = min(num_processes, len(distributor_dict))` on `kicost.py` the ` len(distributor_dict)` have to be `len(dist_list)`. The first way creates a more threads than necessary when asked to not scrape one distributor. `dist_list` keep a list of just the distributors that will be used to scrape in that particular execution of KiCost.

mmmaisel, 2018-06-04:
I don't think that there are more threads created than needed.
In [my kicost.py](https://github.com/mmmaisel/KiCost/blob/master/kicost/kicost.py), when `num_processes = min(num_processes, len(distributor_dict))` is calculated in line 215, all excluded entries are already popped from `distributor_dict` (line 112). After initialization, failed distributors are popped again from `distributor_dict` and num_processes is adjusted as needed (lines 260 and 292).

hildogjr, 2018-06-05:
Agree. Checking more deep now, this change will be related with #186.

hildogjr, 2018-06-06:
Fixed on #242, merging soon.

closed 2018-06-06 by hildogjr