hildogjr/KiCost #221: New release on Pypi? (discussion v0.1.44)

hildogjr, 2018-03-24:
@devbisme 
I am working on next version (0.1.44), in this road map is planed (and need some help to):
#4 (finish to Digikey and need to be replicated for the others) #193 #185 and #144

xesscorp, 2018-05-03:
I need your username on PyPi (or you need to open an account there). Then I can add you as a maintainer and you can release KiCost when needed.

hildogjr, 2018-05-05:
I am added as collaborator on PyPI but not as manager, so I can't release a new KiCost.

hildogjr, 2018-05-29:
Just to keep here registered:
I upload v0.1.44 to pypi and I am releasing on KiCad forum.
I was missing `long_description_content_type="text/markdown",` in `setup.py` for correct interpretation of the markdown description.
Because there was a changing in this interpretation in 16/March/2018 (https://dustingram.com/articles/2018/03/16/markdown-descriptions-on-pypi), the description on the PyPI KiCost page may be with some extra `====`.

@xesscorp  / @devbisme , I am closing this "issue" and intend to fix the `===` issue soon.

closed 2018-05-29 by hildogjr