hildogjr/KiCost #229: Thread exception if no parts found for a symbol

GaryBoone, 2018-04-01:
KiCost fails with a thread exception apparently if a symbol has no parts found.

To reproduce:
1. Make a new KiCad project and add a diode.
2. Set the diode's MPN to: INL5AB30
3. Run KiCost, via CLI with kicost --overwrite -i test.xml
4. See successful run
5. Back in Kicad, edit the diode and replace the diodes' MPN with a blank.
6. Run KiCost, via CLI with 

$ kicost --overwrite -i test.xml
Progress:   0%|                                                                    | 0/1 [00:00<?, ?part/s]Part ['D1'] not found at rs
Part ['D1'] not found at farnell
Part ['D1'] not found at newark
Part ['D1'] not found at tme
Part ['D1'] not found at mouser
Progress: 100%|████████████████████████████████████████████████████████████| 1/1 [00:09<00:00,  9.07s/part]
Exception in thread Thread-1 (most likely raised during interpreter shutdown):

This is on MacOs High Sierra 10.13.4. 
KiCost installed today with 
$ sudo easy_install kicost

hildogjr, 2018-04-01:
Please @GaryBoone, provide more information / the xml problem files. I could not reproduce this issue on Ubuntu Py3 using the last code version here on GitHub.
About (5), thats is the "blank", because in my KiCad (nightly) when I try leave blank the field is automatically deleted.

hildogjr, 2018-04-01:
To install the last code (here) version, you could use:

    sudo +H pip install git+https://github.com/xesscorp/KiCost.git

GaryBoone, 2018-04-01:
I tried with Python 3.4.6 and Python 2.7.13 in virtual environments and could _not_ reproduce the exception on exit. 

I can continue to reproduce with Python 2.7.13 outside the virtual environment. However, I notice that the error is intermittent. Sometimes it exits without error, sometimes with the exception. See below for a sample run. 

I also uninstalled and reinstalled as you suggested (I assumed you meant -H instead of +H). The error persists.

I'm guessing there's a race condition in the exit code of 2.7 versions. 

*** Uninstall, reinstall, reproduce ***
$ sudo pip uninstall kicost
[:~/GoogleDrive … ad Projects/Test/test] 1 $ which kicost
[:~/GoogleDrive … ad Projects/Test/test] 1 $ sudo -H pip install --upgrade git+https://github.com/xesscorp/KiCost.git
$ which kicost
/opt/local/Library/Frameworks/Python.framework/Versions/2.7/bin/kicost
[:~/GoogleDrive … ad Projects/Test/test] $ kicost --overwrite -i test.xml
Progress:   0%|                                             | 0/1 [00:00<?, ?part/s]Part ['D1'] not found at mouser
Part ['D1'] not found at rs
Part ['D1'] not found at farnell
Part ['D1'] not found at newark
Part ['D1'] not found at tme
Progress: 100%|█████████████████████████████████████| 1/1 [00:09<00:00,  9.56s/part]
Exception in thread Thread-4 (most likely raised during interpreter shutdown):

*** Testing with Python 3.4.6 in a virtual environment, I could _not_ reproduce the error. ***
$ sudo pip install virtualenv
$ virtualenv -p /opt/local/bin/python3 kicost_test
$ source kicost_test/bin/activate
[kicost_test] $ sudo -H pip install --upgrade git+https://github.com/xesscorp/KiCost.git
[kicost_test] $ python --version
Python 3.4.6
[kicost_test] $ which kicost
/Users/gary/installations/kicosttest/kicost_test/bin/kicost
[kicost_test] $ kicost --overwrite -i test.xml
Progress:   0%|                                             | 0/1 [00:00<?, ?part/s]Part ['D1'] not found at newark
Part ['D1'] not found at rs
Part ['D1'] not found at tme
Part ['D1'] not found at farnell
Part ['D1'] not found at mouser
Progress: 100%|█████████████████████████████████████| 1/1 [00:09<00:00,  9.42s/part]
[:~/GoogleDrive … cadProjects/Test/test] [kicost_test]

Note clean exit. With repeated tries, I did not see an error with Python 3.4.6.

*** Runs with 2.7.13 outside of a virtual environment. ***
$ which kicost
/opt/local/Library/Frameworks/Python.framework/Versions/2.7/bin/kicost
[:~/GoogleDrive … cadProjects/Test/test] $ python --version
Python 2.7.13
[:~/GoogleDrive … cadProjects/Test/test] $ kicost --overwrite -i test.xml
Progress:   0%|                                             | 0/1 [00:00<?, ?part/s]Part ['D1'] not found at rs
Part ['D1'] not found at newark
Part ['D1'] not found at tme
Part ['D1'] not found at mouser
Part ['D1'] not found at farnell
Progress: 100%|█████████████████████████████████████| 1/1 [00:07<00:00,  7.78s/part]
[:~/GoogleDrive … cadProjects/Test/test] $ kicost --overwrite -i test.xml
[**deleted same output as above about parts not found and progress**]
[:~/GoogleDrive … cadProjects/Test/test] $ kicost --overwrite -i test.xml
[**deleted same output as above about parts not found and progress**]
Exception in thread Thread-4 (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 801, in __bootstrap_inner
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/tqdm-4.19.9-py2.7.egg/tqdm/_monitor.py", line 53, in run
[:~/GoogleDrive … cadProjects/Test/test] $ kicost --overwrite -i test.xml
[**deleted same output as above about parts not found and progress**]
[:~/GoogleDrive … cadProjects/Test/test] $ kicost --overwrite -i test.xml
[**deleted same output as above about parts not found and progress**]
[:~/GoogleDrive … cadProjects/Test/test] $ kicost --overwrite -i test.xml
[**deleted same output as above about parts not found and progress**]
[:~/GoogleDrive … cadProjects/Test/test] $ kicost --overwrite -i test.xml
[**deleted same output as above about parts not found and progress**]
Exception in thread Thread-4 (most likely raised during interpreter shutdown):
[:~/GoogleDrive … cadProjects/Test/test] $ kicost --overwrite -i test.xml
[**deleted same output as above about parts not found and progress**]
Exception in thread Thread-4 (most likely raised during interpreter shutdown):
[:~/GoogleDrive … cadProjects/Test/test] $

hildogjr, 2018-04-01:
Related with #174

hildogjr, 2018-04-01:
There is a note on the documentation https://xesscorp.github.io/KiCost/docs/_build/singlehtml/index.html

    Note that if you install KiCost using pip on a Windows system running Python 2.7,
    using the default option that web scrapes with parallel processes may cause MANY
    errors. You can avoid this problem by:
    kicost -i %file& -s

KiCost will run in a single process, will be less fast but will give us clue about the error. Try this, please.

GaryBoone, 2018-04-02:
Additionally, I confirmed that all pip-installed libraries were up-to-date in both virtual and non-vertual environments for Python 2.7.13.

$ sudo -H pip list --outdated --format=freeze

All libraries were up-to-date.

GaryBoone, 2018-04-02:
$ kicost --overwrite -i test.xml -s
global name 'label' is not defined
$ kicost --overwrite -i test.xml --serial
global name 'label' is not defined

Well, that was unexpected. 

$ grep -i label test.xml
$

GaryBoone, 2018-04-02:
BTW, I'm running on Mac

hildogjr, 2018-04-02:
@GaryBoone, I fix the `--serial` problem. Please, try again with the new code version to check if your main issue is related with the parallel mode.

hildogjr, 2018-04-02:
Yes, we are tracking the Mac-OS problem with other user on issue #174. Because this (I don't have a Mac available here) I have to ask you to help with the debug (try the new code, python 2 or 3, ...).

GaryBoone, 2018-04-03:
Summary of log below:
1. pip install didn't seem to grab the latest version. 
2. uninstalled/reinstalled kicost
3. Ran latest version
Result 1: Label bug gone.
Result 2: Exception on exit remains.

Request: Could we have build numbers in the version string? That way, it would be clear what version is actually present.

One thing I notice is that the exception is not printing its stacktrace as I would expect. It ends with a colon, leading me to expect the stacktrace.

$ sudo -H pip install git+https://github.com/xesscorp/KiCost.gitPassword:
Collecting git+https://github.com/xesscorp/KiCost.git
  Cloning https://github.com/xesscorp/KiCost.git to /private/tmp/pip-mTW218-build
  Requirement already satisfied (use --upgrade to upgrade): kicost==0.1.43 from git+https://github.com/xesscorp/KiCost.git in /opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages
Requirement already satisfied: beautifulsoup4>=4.3.2 in /opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/beautifulsoup4-4.6.0-py2.7.egg (from kicost==0.1.43)
Requirement already satisfied: XlsxWriter>=0.7.3 in /opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/XlsxWriter-1.0.2-py2.7.egg (from kicost==0.1.43)
Requirement already satisfied: future>=0.15.0 in /opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/future-0.16.0-py2.7.egg (from kicost==0.1.43)
Requirement already satisfied: lxml>=3.7.2 in /opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages (from kicost==0.1.43)
Requirement already satisfied: yattag>=1.5.2 in /opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/yattag-1.10.0-py2.7.egg (from kicost==0.1.43)
Requirement already satisfied: tqdm>=4.4.0 in /opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/tqdm-4.19.9-py2.7.egg (from kicost==0.1.43)
Requirement already satisfied: CurrencyConverter>=0.5 in /opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/CurrencyConverter-0.13.3-py2.7.egg (from kicost==0.1.43)
Requirement already satisfied: pycountry>=18.2 in /opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages (from kicost==0.1.43)
$ kicost --overwrite -i test.xml -s
global name 'label' is not defined
$ kicost --version
KiCost 0.1.43
$ sudo -H pip uninstall kicost
[output elided]
$ sudo -H pip install git+https://github.com/xesscorp/KiCost.git
[output elided]
$ kicost --version
KiCost 0.1.43
$ kicost --overwrite -i test.xml -s
        No currency/country configuration for Newark
        No currency/country configuration for TME
        No currency/country configuration for Farnell
        No currency/country configuration for RS Components
        No currency/country configuration for Mouser
Progress:   0%|                                                                    | 0/1 [00:00<?, ?part/s]Part ['D1'] not found at rs
Part ['D1'] not found at rs
Progress:   0%|                                                                    | 0/1 [00:02<?, ?part/s]Part ['D1'] not found at tme
Part ['D1'] not found at tme
Progress:   0%|                                                                    | 0/1 [00:05<?, ?part/s]Part ['D1'] not found at newark
Part ['D1'] not found at newark
Progress:   0%|                                                                    | 0/1 [00:06<?, ?part/s]Part ['D1'] not found at mouser
Part ['D1'] not found at mouser
Progress:   0%|                                                                    | 0/1 [00:07<?, ?part/s]Part ['D1'] not found at farnell
Part ['D1'] not found at farnell
Progress: 100%|████████████████████████████████████████████████████████████| 1/1 [00:08<00:00,  9.00s/part]
Exception in thread Thread-1 (most likely raised during interpreter shutdown):
$

hildogjr, 2018-04-03:
It is a good idea. Please, create a issue with something like "create a __VERSION__ control for beta versions before PYPI release".
About not getting the new version may be necessary to use `-U` (update on `pip`):

   sudo -H pip install -U git+https://...

hildogjr, 2018-04-03:
@devbisme, this issue appear related with the #174 related by @timoalho.
@timoalho, are you still having the hang issue?

timoalho, 2018-04-03:
@hildogjr the issue persists, see my update in #174 .

hildogjr, 2018-04-11:
@gare, did you could install the last version from GitHub?

     sudo -H pip install -U git+https://...

After this, please, report with the `--serial`option to check if the cause is the same of the other issue on Mac-OS.

hildogjr, 2018-04-11:
Marked as duplicated of #174 (Kicost hangs when parallel scraping on Mac-OS). Re-open case is a different issue.

commit 4c12b184 (2018-04-02): Merge pull request #230 from hildogjr/master

Fixed #229 serial problem
commit efe5e341 (2018-04-02): Fixed #229 serial problem
closed 2018-04-11 by hildogjr