hildogjr/KiCost #233: ImportError: wxPython package not recognised.

Phara0h, 2018-04-09:
Latest 1.4.3 version of Linux (Ubuntu) I get this error after installing it through pip. 

Traceback (most recent call last):
  File "/usr/local/bin/kicost", line 11, in <module>
    load_entry_point('kicost==0.1.42', 'console_scripts', 'kicost')()
  File "/home/jtwhissel/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 572, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/home/jtwhissel/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2755, in load_entry_point
    return ep.load()
  File "/home/jtwhissel/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2408, in load
    return self.resolve()
  File "/home/jtwhissel/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2414, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.42-py2.7.egg/kicost/__main__.py", line 39, in <module>
    from .kicost_gui import * # User guide.
  File "/usr/local/lib/python2.7/dist-packages/kicost-0.1.42-py2.7.egg/kicost/kicost_gui.py", line 33, in <module>
    raise ImportError('wxPython package not recognised.')
ImportError: wxPython package not recognised.

Phara0h, 2018-04-09:
Added 

    import wxversion
    wxversion.select('3.0')

before the import wx instide of kicost_gui.py and it fixed the issue.

hildogjr, 2018-04-09:
Doing this, the GUI works perfect on python 2.7 using the already installed wx from KiCad? This need to be tested in others platforms/OS and use some exception:

    import sys.version_info as py_version
    if py_version<(3,0):
        try: # Because the new implementation of wxPython doesn't provide `wxversion`.
            import wxversion
            wxversion.select('3.0')
        exception:
            pass
    import wx

I am using on Python3, so I installed manually the wxPython package (and the machines that I don't use GUI have no problem).

hildogjr, 2018-04-09:
Reference about this issue
https://docs.wxpython.org/MigrationGuide.html#the-wxversion-module

hildogjr, 2018-04-11:
I could not reproduce the issue, appear related with the official Pypi version (bug already fixed on current Git code).
Re-open case not.

closed 2018-04-11 by hildogjr