hildogjr/KiCost #236: Alternate parts

volafh, 2018-04-16:
With a similar goal in mind like issue #17 I'd like to suggest the possibility of entering alternatives to parts. Especially when looking at generic passive components you might have tested several alternatives for your design and want to use the component with best availability.

There is something similar with Subparts. You can of course use subparts in order to automatically get storage and price of alternate parts and then manually delete the non-stocked or more expensive.

hildogjr, 2018-04-16:
Something in this line could be:
1) Use more than one `manf#`cone in the design (e.g. manf# = PART1; PART5; PART4);
2) Set `ALIAS` variable (e.g. alias = 1; 3, in this example PART1 and PART4 are alias);
3) Create the spreadsheet as usual;
4) Create other spreadsheet page as 'minimum values of PART1 and PART4'.

This is just a idea, but now, I am trying to use time on the base feature as multiple currency #65 and bugs.

hildogjr, 2018-11-17:
@mobilinkd, about the discussion in #17.

The propose of use `|` as "or" is interesting, but we have to keep in mind that the different distributors could have best price in different `manf#` part that compose your selection. Therefore the spreadsheet have to deal with the best price (not the internal KiCost code), **this will ask for a new spreadsheet format**.

The work around is using the multipart options and select by your self.

I am re-opening this issue for new propose about, if you have some. But, for now, we have without people to fix the main recent issues of KiCost (https://forum.kicad.info/t/kicost-call-of-developers/13271), so I have to focus me time in keep KiCost operational (not working in #236, #17).

closed 2019-03-16 by hildogjr