hildogjr/KiCost #237: Information on "Unknown error"

metropt, 2018-04-17:
`Progress:  93%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████▏          | 27/29 [00:04<00:00,  5.32part/s]Unknown error for 571-440129-4 from mouser
Unknown error for 571-440129-4 from mouser                                                                                                                                                                   
Progress:  93%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████▏          | 27/29 [00:05<00:00,  5.32part/s]Part ['J3#4'] not found at mouser
Part ['J3#4'] not found at mouser                                                                                                                                                                            
Progress:  93%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████▏          | 27/29 [00:05<00:00,  5.32part/s]No Mouser part number found!
No Mouser part number found!                                                                                                                                                                                 
Progress:  93%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████▏          | 27/29 [00:05<00:00,  5.32part/s]No Mouser part quantity found!
No Mouser part quantity found!                                                                                                                                                                               
Progress:  93%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████▏          | 27/29 [00:05<00:00,  5.32part/s]No Mouser pricing information found!
No Mouser pricing information found!                                                                                                                                                                         
Progress:  97%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████▌     | 28/29 [00:05<00:00,  4.96part/s]mouser ['X1#4']
mouser ['X1#4']                                                                                                                                                                                              
Progress:  97%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████▌     | 28/29 [00:05<00:00,  4.96part/s]Unknown error for M20-7820646 from mouser
Unknown error for M20-7820646 from mouser   `

Hi,
How can I debug what is happening when get an `Unknow error for part`?

Thank you

hildogjr, 2018-04-17:
Use the debug option.

   kicost --help

Will appear the call parameter `--debug`. Use 0 (less) to 8 (more deep debug). If still have some problem with the software, create the issue related or re-open this one.

hildogjr, 2018-04-17:
Could you attach the BoM file? I just tested the M20-7820646 as bellow and worked fine.

[test.csv.zip](https://github.com/xesscorp/KiCost/files/1920509/test.csv.zip)
[test.xlsx](https://github.com/xesscorp/KiCost/files/1920507/test.xlsx)

metropt, 2018-04-17:
I forgot to mention that the error shows up for all the parts and I just pasted a part of the output.
[project.xml.zip](https://github.com/xesscorp/KiCost/files/1920547/project.xml.zip)

metropt, 2018-04-17:
for no reason it worked now :s

hildogjr, 2018-04-17:
We realize that site distributors block multiple consecutive access by the same IP.
For this, you could use `--throttling_delay` to give a time between access to the same distributor.

closed 2018-04-17 by hildogjr