hildogjr/KiCost #242: Mouser scraping fails for some parts

mmmaisel, 2018-04-22:
Scraping some Mouser parts fails with "Unknown error for <part number> from mouser" while other parts or distributors just work fine. This was tested with git revision e7e4243f on Ubuntu 16.04 64bit (Python 2.7) and Ubuntu 18.04 64bit (Python 3.6).

KiCost was invoked with the following command:
```
~/.local/bin/kicost -i test.xml -d99 -w -s --throttling_delay 1
```

PDB shows that the variables "html" and "tree" in distributors/mouser/mouser.py:178 contain some  [javascript garbage](https://github.com/xesscorp/KiCost/files/1935790/js_garbage.txt) instead of the intented HTML source code.

The following minimal KiCad xml demonstrates the error:
[test.xml.txt](https://github.com/xesscorp/KiCost/files/1935792/test.xml.txt)

Terminal output: 
[term_output.txt](https://github.com/xesscorp/KiCost/files/1935791/term_output.txt)

hildogjr, 2018-04-22:
I could not reproduce this error, for me the result was just fine: [test.xlsx](https://github.com/xesscorp/KiCost/files/1936170/test.xlsx).
Waiting for more test / someone more to confirm (I'm not a specialist on web / javascript).

mmmaisel, 2018-05-29:
It looks like the root-cause of this issue is that mouser returns a captcha after scraping some parts.

I did some further experiments and figured out that if you correctly handle cookies, the captcha can be avoided. The following (partial?) solution worked for me if the `-s` and `--throttling_delay=3` options are used:

* Create a `http.cookiejar` object and a `urllib.request.HTTPCookieProcessor` and install it as global opener (`urllib.request.install_opener`).
* Init the `Request` object with user-agent and keep this user-agent for the whole scraping process.
* Access `www.mouser.com` without search query and store all cookies for subsequent scraping apsses.
* Scrape all parts and update cookies after every part if required.

See attached [mouser.py.txt](https://github.com/xesscorp/KiCost/files/2049045/mouser.py.txt) for an example implementation of those hacks.

Now, the question is how to cleanly implement the following in KiCost's code:
* Keep state information between calls to `get_part_html_tree` and between different threads.
* Avoid global opener object, maybe use the opener object directly instead of `urlopen`.
* Add support for the mouser `preferences='ps=www2&pl=en-US&pc_www2=USDe'` cookie (this is not supported by my hack).
* Implement this for all distributors.

hildogjr, 2018-05-29:
@mmmaisel thanks by debug and discussion. We (@devbisme and I) know that this multiple-sequencial scrape could be interpreted as robots and block by the distributors.
And debug by users we help KiCost to improve.

(Since the last code modifications #223 somethink could be implemented on `fake_brownser(foo)`).

mmmaisel, 2018-05-30:
@hildogjr yes, fake_browser (as a class) would be a good point for storing various things like the cookiejar, selected user_agent, domain name, urllib opener, scrape retries and other things.
However, this fake_browser object has to be passed to every function that accesses URLs like get_part_html_tree and various AJAX using sub-functions of some distributors. It also has to be instanced somewhere before the scraping starts.

At this point, I think the code needs some redesign because currently KiCost first iterates over all parts (eventually distributed over several threads) and then iterates over all known distributors for a single part (with random user-agent and without cookies from previous requests). This needs to be changed to iterate distributors first, then parts to keep the state information.
I think one thread per distributor is enough as the scrape operation is limited by network I/O speed and throttling_delay settings not CPU speed. Allowing multiple threads per distributor would only increase state synchronization complexity and risk of getting banned.

Maybe, it would also make sense to change all distributor modules to classes which hold all required state information and common values like the fake_browser object, currency settings and maybe others. This would reduce the amount of function arguments that has to passed around as well.

While I'm not a Python specialist, I can help implementing the changes if desired.

hildogjr, 2018-05-30:
@mmmaisel nice too ear some tips. I am not expert too but, all standardization that help the next user / developer to add and correct features will be welcome.
When I entered to the KiCost project I started talking with @devbisme and we split the code in several modules (before KiCost as "like one big code file"). This allow us to improve the spreadsheet, expand to more EDAs, add the multiple parts component...
Since web-scrape is not my strong skill on Python, I am focused on the other code parts (spreadsheet, BOM readers, merge BOMs, ...).

If @devbisme (owner of the package) agree and pass us some directives, we could use the `pickle_module` or one `reformulation_classes` in my Git to this development to release when stable.

In my view, the points of this implementation is (more your tips) allow enhacement #4 and #65 (in beta on Digikey). Also if allow to search some part by "value/footprint/tolerance/so on" and get by, for example, more chipper, could me allow to future solve #17 (for resistor and capacitor at least).

Yes, I agree, the distributors modules is now the "bottleneck" of KiCost.

hildogjr, 2018-05-31:
@mmmaisel, what you think:
1) A main class with `browser` definitions, maximum number of retries, cookies, ajax / java of plugins used to interpret (beautifulsoup4 also);
2) Each distributor inherits (1) add language, currency, method to match when table results (close match as now or chipper and prioritize parts recommend to laytout);
3) Change the way to iterate, if possible (first distributors, second by parts), as you said to decreased the ban risk;
4) Return extra of the component page (footprint, image link, ...). As did already for Digikey;
5) The distributors class have to keep the functionality of automatic import (today just adding the subfolder if the `__init__.py` on `distributors`, it is recognized as a new distributor module).

mmmaisel, 2018-06-01:
I started implementing the changes in my [local fork](https://github.com/mmmaisel/KiCost).
First tests with mouser work. However, there may be new bugs introduced by refactoring.

Currently implemented are:
- Class approach for all distributors and fake_browser.
- Parts are scraped in distributor -> part order.
- One (IO limited) python thread per distributor and simplified locking.
- Automatic module import is now handeled from kicost.py.

hildogjr, 2018-06-02:
@mmmaisel, appear that the Farnell issue was fixed on your branch.
I just had this issues / reports:
1) KiCost is spending almost 30s to start scrape on my computer (may be related in the class importation and initial configuration?);
2) Digikey is not working (return always empty). But I saw that there is a #TODO in the file.
3) TME got this error:

Traceback (most recent call last):
  File "/home/h/.local/bin/kicost", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.5/dist-packages/kicost/__main__.py", line 274, in main
    local_currency=args.currency)
  File "/usr/local/lib/python3.5/dist-packages/kicost/kicost.py", line 315, in kicost
    res_dist = res_proc.get()
  File "/usr/lib/python3.5/multiprocessing/pool.py", line 608, in get
    raise self._value
  File "/usr/lib/python3.5/multiprocessing/pool.py", line 119, in worker
    result = (True, func(*args, **kwds))
  File "/usr/local/lib/python3.5/dist-packages/kicost/kicost.py", line 300, in mt_scrape_part
    retval.append(inst.scrape_part(i, parts[i]))
  File "/usr/local/lib/python3.5/dist-packages/kicost/distributors/distributor.py", line 160, in scrape_part
    qty_avail = self.dist_get_qty_avail(html_tree)
  File "/usr/local/lib/python3.5/dist-packages/kicost/distributors/tme/tme.py", line 149, in dist_get_qty_avail
    ajax_tree, qty_str = self.__ajax_details(pn)
  File "/usr/local/lib/python3.5/dist-packages/kicost/distributors/tme/tme.py", line 76, in __ajax_details
    r = r.decode('utf-8')  # Convert bytes to string in Python 3.
UnboundLocalError: local variable 'r' referenced before assignment

mmmaisel, 2018-06-03:
@hildogjr 
1. Init time issue: should be fixed now.
2. I'm now working on the digikey issue.
3. The TME "use of undefined local variable 'r'" error was already there before my refactoring, see [line 66](https://github.com/xesscorp/KiCost/blob/7b66a2f3693838d79e2aff631866316da82cd5b4/kicost/distributors/tme/tme.py)).

hildogjr, 2018-06-06:
Fixed on #242, merging soon.
This discussion could be re-open any moment for more improvements.

closed 2018-06-06 by hildogjr