hildogjr/KiCost #243: Crash with "ValueError: could not convert string to float:"

linuxcf, 2018-04-30:
Running kicost from the command line on stock 64-bit Ubuntu 16.04.4.  If I run the attached BOM xml file with '--include digikey', the run completes without errors.  However, if I let kicost check all distributors it can, it crashes per the output below.  Although I am running kicost in the anaconda, I get the same error if I run it in the native Python environment.

(base) gus@X230t:~/development/kicad/projects/Pi-Hat-double$ kicost   --overwrite --throttling_delay 1 -i pihat_template-error_demo.xml 
Progress:   0%|                                        | 0/10 [00:00<?, ?part/s]Part ['J2', 'J4'] not found at farnell
Part ['TP2', 'TP3', 'TP5', 'TP6', 'TP13', 'TP14', 'TP4', 'TP1', 'TP9', 'TP10', 'TP11', 'TP12', 'TP7', 'TP8', 'TP15'] not found at farnell
Part ['C13', 'C3', 'C6', 'C8', 'C11'] not found at farnell
Part ['TP2', 'TP3', 'TP5', 'TP6', 'TP13', 'TP14', 'TP4', 'TP1', 'TP9', 'TP10', 'TP11', 'TP12', 'TP7', 'TP8', 'TP15'] not found at rs
Part ['J3'] not found at farnell
Part ['C13', 'C3', 'C6', 'C8', 'C11'] not found at tme
Part ['J2', 'J4'] not found at tme
Part ['J1'] not found at tme
Part ['TP2', 'TP3', 'TP5', 'TP6', 'TP13', 'TP14', 'TP4', 'TP1', 'TP9', 'TP10', 'TP11', 'TP12', 'TP7', 'TP8', 'TP15'] not found at tme
Part ['J1'] not found at farnell
Part ['U2'] not found at farnell
Part ['C5', 'C4', 'C2', 'C12', 'C1', 'C10', 'C7', 'C9'] not found at tme
Progress:  10%|███▏                            | 1/10 [00:16<02:31, 16.80s/part]Part ['R5', 'R6', 'R7', 'R3', 'R4', 'R8', 'R10', 'R11'] not found at tme
Part ['R5', 'R6', 'R7', 'R3', 'R4', 'R8', 'R10', 'R11'] not found at farnell
Progress:  20%|██████▍                         | 2/10 [00:19<01:41, 12.72s/part]Part ['R5', 'R6', 'R7', 'R3', 'R4', 'R8', 'R10', 'R11'] not found at rs
Progress:  30%|█████████▌                      | 3/10 [00:21<01:05,  9.32s/part]Part ['C13', 'C3', 'C6', 'C8', 'C11'] not found at newark
Progress:  40%|████████████▊                   | 4/10 [00:21<00:40,  6.69s/part]Part ['U1', 'U5', 'U4', 'U3'] not found at tme
Part ['R1', 'R2'] not found at farnell
Part ['J1'] not found at newark
Part ['C5', 'C4', 'C2', 'C12', 'C1', 'C10', 'C7', 'C9'] not found at farnell
Part ['U1', 'U5', 'U4', 'U3'] not found at rs
Part ['U1', 'U5', 'U4', 'U3'] not found at farnell
Progress:  60%|███████████████████▏            | 6/10 [00:28<00:19,  4.83s/part]Part ['J3'] not found at tme
Progress:  70%|██████████████████████▍         | 7/10 [00:28<00:10,  3.56s/part]Part ['R1', 'R2'] not found at tme
Part ['U2'] not found at tme
Part ['U2'] not found at rs
Progress:  80%|█████████████████████████▌      | 8/10 [00:33<00:08,  4.04s/part]Part ['R1', 'R2'] not found at rs
Progress:  90%|████████████████████████████▊   | 9/10 [00:36<00:03,  3.58s/part]multiprocessing.pool.RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/home/gus/development/anaconda3/lib/python3.6/multiprocessing/pool.py", line 119, in worker
    result = (True, func(*args, **kwds))
  File "/home/gus/development/anaconda3/lib/python3.6/site-packages/kicost/distributors/web_routines.py", line 159, in scrape_part
    price_tiers[d] = dist_module.get_price_tiers(html_tree)
  File "/home/gus/development/anaconda3/lib/python3.6/site-packages/kicost/distributors/mouser/mouser.py", line 60, in get_price_tiers
    unit_price = float(re.sub('[^0-9.]', '', unit_price_tree.text))
ValueError: could not convert string to float: 
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/gus/development/anaconda3/bin/kicost", line 11, in <module>
    sys.exit(main())
  File "/home/gus/development/anaconda3/lib/python3.6/site-packages/kicost/__main__.py", line 260, in main
    scrape_retries=args.retries, throttling_delay=args.throttling_delay)
  File "/home/gus/development/anaconda3/lib/python3.6/site-packages/kicost/kicost.py", line 250, in kicost
    id, url, part_num, price_tiers, qty_avail = result.get()
  File "/home/gus/development/anaconda3/lib/python3.6/multiprocessing/pool.py", line 644, in get
    raise self._value
ValueError: could not convert string to float:
[github-kicost.zip](https://github.com/xesscorp/KiCost/files/1962437/github-kicost.zip)

hildogjr, 2018-04-30:
Hi @linuxcf, I tried now no error. But could be something related with `anaconada` and/or some specific part (the format of Mouser HTML of this part).
[pihat_template-error_demo.xlsx](https://github.com/xesscorp/KiCost/files/1962456/pihat_template-error_demo.xlsx)
Could you run KiCost as `kicost -i YOURFILE -np 1 --debug 8 > logfile.txt`.
This will expend more time to execute than usual (will not can multiple web scrapes, just one by time) but will give me a full report log.

linuxcf, 2018-05-01:
I installed kicost 0.1.43 with pip.  Anaconda tracks dependencies quite tightly.  I ran kicost per your recommendation and got the same error.  Looking at the log file, it may be the Keystone 5017 part that is throwing the kicost off.  I have attached the log file.

[logfile.txt](https://github.com/xesscorp/KiCost/files/1962535/logfile.txt)

hildogjr, 2018-05-01:
The last part scraped appear to be '649-66953-020LF' at Mouser, it web page have "quotes" as last price row (https://www.mouser.com/ProductDetail/Amphenol-FCI/66953-020LF?qs=%2fha2pyFadug2XGUHDLtReyzRpusVX%252bd4vQgOvrv5nvo%3d).
If is this, this issues is already fix #234, but the fixed version is not on pypi.
So, I recommend you to install KiCost from the GitHub repository to check (use pip was you are habituated):

    pip install -U git+https://github.com/xesscorp/KiCost.git

Or just replace the files on the folder by the files here in GitHub.

hildogjr, 2018-05-01:
Contact here if having some problem to do that.

closed 2018-05-01 by hildogjr

linuxcf, 2018-05-01:
Yup, the pip update from git did it.  Thank you!  Will the git version make it to pypi/pip?

hildogjr, 2018-05-01:
Yes, sure. @devbisme is owner, as soon as possible he run some checks and update the pypi with last version.
This version includes this list of fix and new features https://github.com/xesscorp/KiCost/blob/master/HISTORY.rst