hildogjr/KiCost #245: KiCost - No Farnell results

iahim67, 2018-05-03:
I have created a KiCad 4.07 schematics and used Farnell components (manf# new field).
Seems that KiCost is able to get the parts in my BOM from all distributors except Farnell.
I have installed and used both 0.1.43 and beta KiCost from github (https://github.com/xesscorp/KiCost) with same results. I tried resistors, diodes, capacitors, connectors from Farnell but there are always 0 results from Farnell while there are more or less results from the rest of the distributors.
[image]

[tutorial1.xlsx](https://github.com/xesscorp/KiCost/files/1970029/tutorial1.xlsx)

[tutorial1.zip](https://github.com/xesscorp/KiCost/files/1970031/tutorial1.zip)

#### Issue / Problem report
Add:
1. Terminal error report / txt;
2. Use ``--debug`` option on KiCost call;
2. Write the command used to call KiCost (or the graphical interface configuration);
3. One BoM to reproduce the error (with the EDA version).

#### Doubts

#### Feature / Enhancement request
- Explain it will help the other users.

The team thanks your contribution.

xesscorp, 2018-05-03:
This happens because Farnell returns a page with Javascript to redirect to the product page that kicost has requested. Kicost can't execute the Javascript or find the link to the page, so no part information can be scraped. The only solution to this may be to use a Python package like Selenium and mimic the operation of an actual browser. Without doing that, kicost will not be able to scrape Farnell.

hildogjr, 2018-05-03:
A little more complicated that I was thinking. I just check the HTML tags and appear fine.

iahim67, 2018-05-03:
Does it mean there is no solution in the near future for Farnell?

hildogjr, 2018-05-03:
Sure there is, but for now @xesscorp can help quick than I.
I have to study a little, javascript and its interaction in the web is not my strong.

iahim67, 2018-05-03:
OK, thank's for now  :-)

hildogjr, 2018-05-12:
Now, with #223 enhancement will be more simple to do this improvement to all distributors submodules (also fixing probable #242).

hildogjr, 2018-05-26:
Checking the documentation about `selenium` is needed some thirty browser / actual browser plugin (in the case of Firefox / Chrome) plus (in my tests) a configuration of the installed browser folder (that change in each OS).
These leave me concerned.

mmmaisel, 2018-05-31:
I did some experiments in an empty browser session and it seems that the root cause of this is missing cookies as in issue #242.

Additionally, the Farnell search URL has changed from `http://it.farnell.com/webapp/wcs/stores/servlet/Search?catalogId=15001&langId=-4&storeId=10165&gs=true&st=` to `http://it.farnell.com/Search?storeId=10165&catalogId=15001&categoryName=&selectedCategoryId=&langId=-4&categoryIdBox=&st=`.

Steps to reproduce my test:
* Clear all browser cookies
* Access KiCost's search URL with a part number -> get javascript redirect
* Clear cookies again, then access `it.farnell.com` main site
* Access KiCost's search URL with a part number -> it works

So fixing #242 should fix this as well, no need to implement javascript.

closed 2018-05-31 by hildogjr