hildogjr/KiCost #258: ImportError when using multiple threads

dsa-t, 2018-05-20:
I installed kicost on windows with "pip install git+https://github.com/xesscorp/KiCost.git".
If "Parallel process" is set to >1, kicost produces an error with CSV or KiCad input files.
KiCad version 5.0.0-rc2-dev-727-ga9dc41731-dirty.
Traceback:
`Process SpawnPoolWorker-3:
Process SpawnPoolWorker-6:
Traceback (most recent call last):
  File "c:\program files\python36\lib\multiprocessing\process.py", line 258, in _bootstrap
    self.run()
  File "c:\program files\python36\lib\multiprocessing\process.py", line 93, in run
    self._target(*self._args, **self._kwargs)
  File "c:\program files\python36\lib\multiprocessing\pool.py", line 108, in worker
    task = get()
  File "c:\program files\python36\lib\multiprocessing\queues.py", line 337, in get
    return _ForkingPickler.loads(res)
  File "c:\program files\python36\lib\site-packages\kicost\distributors\web_routines.py", line 35, in <module>
    from ..eda_tools.eda_tools import order_refs # To better print the warnings about the parts.
  File "c:\program files\python36\lib\site-packages\kicost\eda_tools\__init__.py", line 51, in <module>
    eda_modules[module] = __import__(module, globals(), locals(), [], level=1)
  File "c:\program files\python36\lib\site-packages\kicost\eda_tools\altium\__init__.py", line 29, in <module>
    from .altium import get_part_groups
  File "c:\program files\python36\lib\site-packages\kicost\eda_tools\altium\altium.py", line 38, in <module>
    from ...kicost import distributor_dict
  File "c:\program files\python36\lib\site-packages\kicost\kicost.py", line 54, in <module>
    from .distributors.web_routines import scrape_part, config_distributor
ImportError: cannot import name 'scrape_part'`

hildogjr, 2018-05-21:
By this repost appear to  be using Windows.
I could not reproduce the error.
Could you execute

    kicost -i %YOUR_FILE% -d 8 > log.txt

and attach here the log file?

dsa-t, 2018-05-21:
Windows does not write anything to the log.txt with this command, but here is console output:
[log_.txt](https://github.com/xesscorp/KiCost/files/2021338/log_.txt)
with -np 1 it works fine.

dsa-t, 2018-05-21:
And here is a log from -np 1 run (interrupted):
[log.txt](https://github.com/xesscorp/KiCost/files/2021352/log.txt)

hildogjr, 2018-06-02:
On hold. Possible fix with #242.

hildogjr, 2018-06-06:
Fixed on #242, merging soon.

closed 2018-06-06 by hildogjr