hildogjr/KiCost #282: Getting no result in Mouser

hildogjr, 2018-06-29:
Again, Mouser distributor module is return no results.
Need more investigation.

mmmaisel, 2018-07-02:
Mouser started using Distil Networks to block bots and tracker blockers.

Basically, if you now open the mouser page, you first get a invisible dummy page which runs some javascript which reads lots of your systems information, including the following things, and sends them to Distils servers:

* user-agent string
* system language
* browser plugins
* "do not track" settings
* screen resolution and color depth
* timezone
* WebGL canvas hash
* touch capabilities
* installed audio and video codecs
* installed system fonts
* Javascript and HTML/CSS engine name
* Availability of lots of Javascript API functions
* Various other things I couldn't decode

The script itself is loaded from a dynamically generated URL and is compressed with aptimizer. It also includes CryptoJS and zlib libraries. The whole communication with Distils servers uses lots of non-standard HTTP headers which start with "X-Distil-".

After that, if Distil gives their OK, you will be redirected to the Mouser page with temporary access based on session cookies. However the session is still permanently monitored by Distil.

If you don't support javascript or block their trackers, your access to Mouser gets blocked, (in my case also in Firefox).

FYI: I found a [python V8 javascript engine](https://github.com/sqreen/PyMiniRacer) implementation. Maybe this can help.

hildogjr, 2018-07-02:
It will be quite complicated to solve, it will demand some time of study and test (in my not web experiment developer).
I intend to release 0.1.46 this month, may be without this fixed.
Thanks by inlight me.

hildogjr, 2018-09-15:
The scrape module are now deprecated #314

closed 2018-09-15 by hildogjr