hildogjr/KiCost #290: No wxPython 3.x.x suport

marcos-paltatech, 2018-07-04:
marcos@silversurfer:~$ kicost

(kicost:10304): Gtk-WARNING **: Unable to locate theme engine in module_path: “adwaita”,

(kicost:10304): Gtk-WARNING **: Unable to locate theme engine in module_path: “adwaita”,
Traceback (most recent call last):
File “/usr/local/bin/kicost”, line 11, in 
sys.exit(main())
File “/usr/local/lib/python2.7/dist-packages/kicost/main.py”, line 233, in main
kicost_gui() # Use the user guide.
File “/usr/local/lib/python2.7/dist-packages/kicost/kicost_gui.py”, line 1051, in kicost_gui
frame = formKiCost(None)
File “/usr/local/lib/python2.7/dist-packages/kicost/kicost_gui.py”, line 252, in init
self.SetSizeHints( wx.DefaultSize, wx.DefaultSize )
File “/usr/lib/python2.7/dist-packages/wx-3.0-gtk2/wx/_core.py”, line 9718, in SetSizeHints
return core.Window_SetSizeHints(*args, **kwargs)
TypeError: in method ‘Window_SetSizeHints’, expected argument 2 of type ‘int’
marcos@silversurfer:~$

From kicad forum "GUI was developed with compatibility with wxPython 4.x.x and KiCad still using 3.x.x on v5"

Thanks!

hildogjr, 2018-07-04:
As report by @marcos-paltatech, this is related with https://docs.wxpython.org/classic_vs_phoenix.html 3 to 4 migration.
Maybe the docs give us some way to deal with of just use the work around `if int(wx.__version__[0])>3)`.

hildogjr, 2018-07-10:
Waiting confirmation to be officially close.

commit 620144ff (2018-07-10): Fix #290
commit 795e233d (2018-07-10): Fix #290 on self.run(foo)
closed 2018-09-03 by hildogjr