hildogjr/KiCost #291: Digikey crash with "NameError: name 'pn' is not defined"

linuxcf, 2018-07-06:
Running KiCost with only Digikey selected (since now Mouser's policies prevent scraping, as per issue #282 ), I get an error about name 'pn' not defined, occurring @line 385 of 'digikey.py', like so:

-------------------------------------------------------------------------------------------------------------------
```
(base) gus@X230t:~/development/kicad/projects/AR0521-large_bd$ kicost --debug -w --include digikey  -eda kicad -i AR0521-large_bd.xml > kicostlog20180705.txt
Progress:  26%|████████▎                       | 6/23 [02:25<06:57, 24.55s/part]Traceback (most recent call last):
  File "/home/gus/development/anaconda3/lib/python3.6/site-packages/kicost/distributors/digikey/digikey.py", line 382, in merge_price_tiers
    for tr in alt_tree.find('table', id='product-dollars').find_all('tr'):
AttributeError: 'NoneType' object has no attribute 'find_all'

During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/home/gus/development/anaconda3/bin/kicost", line 11, in <module>
    sys.exit(main())
  File "/home/gus/development/anaconda3/lib/python3.6/site-packages/kicost/__main__.py", line 282, in main
    local_currency=args.currency)
  File "/home/gus/development/anaconda3/lib/python3.6/site-packages/kicost/kicost.py", line 289, in kicost
    scrape_result = distributor_dict[d]['instance'].scrape_part(i, parts[i])
  File "/home/gus/development/anaconda3/lib/python3.6/site-packages/kicost/distributors/distributor.py", line 128, in scrape_part
    html_tree, url = self.get_part_html_tree(part)
  File "/home/gus/development/anaconda3/lib/python3.6/site-packages/kicost/distributors/distributor.py", line 166, in get_part_html_tree
    return self.dist_get_part_html_tree(part.fields[key], extra_search_terms)
  File "/home/gus/development/anaconda3/lib/python3.6/site-packages/kicost/distributors/digikey/digikey.py", line 308, in dist_get_part_html_tree
    self.merge_price_tiers(tree, ap_tree)
  File "/home/gus/development/anaconda3/lib/python3.6/site-packages/kicost/distributors/digikey/digikey.py", line 385, in merge_price_tiers
    self.logger.log(DEBUG_OBSESSIVE, 'Problem merging price tiers for Digikey part {} with alternate packaging!'.format(pn))
NameError: name 'pn' is not defined

```
-----------------------------------------------------------------------------------------------
As you can see, the command used was 'kicost --debug -w --include digikey  -eda kicad -i AR0521-large_bd.xml > kicostlog20180705.txt'.

I am attaching the BOM (in the zipped archive)  and the logfile output.

[kicostzip20180706.zip](https://github.com/xesscorp/KiCost/files/2169171/kicostzip20180706.zip)
[kicostlog20180705.txt](https://github.com/xesscorp/KiCost/files/2169179/kicostlog20180705.txt)

closed 2018-07-06 by hildogjr

linuxcf, 2018-07-06:
Thank you!  It worked!

I have one, maybe dumb, question.  You recommend replacing the 'digikey.py' file "with the new one here."  However, all I can find here is the diff file.  Is there any easy way to download the diff file itself, so I can apply the 'patch' command on the original?  Simple copying/pasting does not seem to work.  The way I got around that (other than patching the 'digikey.py' file by hand) was to run 'pip install -U git+https://github.com/xesscorp/KiCost.git', which I think is an overkill.  I don't use github frequently, so I may be missing something.

hildogjr, 2018-07-06:
I am not a expert on git / distributions yet. But, the corrections that I did was:

    def merge_price_tiers(self, main_tree, alt_tree, pn):
    def merge_qty_avail(self, main_tree, alt_tree, pn):

That before were:

    def merge_price_tiers(self, main_tree, alt_tree):
    def merge_qty_avail(self, main_tree, alt_tree):

commit 64015f15 (2018-07-06): Merge pull request #292 from hildogjr/master

Fix #291
commit d69237bc (2018-07-06): Fix #291