hildogjr/KiCost #304: Crashes because of regex non-match when creating xlsx

MrSurly, 2018-08-02:
I pulled the fix for #302.

I instrumented the code in spreadsheet.py, added a debug breakpoint:

```python
    # Then, order the part references with priority ref prefix, ref num, and subpart num.
    def get_ref_key(part):
        match = re.match(PART_REF_REGEX, part.collapsed_refs)
        try:
            return [match.group('prefix'), match.group('ref_num'), match.group('subpart_num')]
        except Exception as e:
            import pdb; pdb.set_trace()
            raise e

```

In this case, `part.collapsed_refs` is an empty string (i.e. `repr(part.collapsed_refs)` yields `''`).

```
$ kicost -i main.xml --include digikey  -var bat_pwr --overwrite -d1 --throttling_delay 0 --currency USD
Exchange rate: 1 EUR = 1.18 USD
# Getting from XML 'main.xml' KiCad BoM...
Getting authorship data...
Getting parts library...
Getting components...
# Removing do not populate parts...
Splitting subparts in the manufacture / distributors codes...
# Grouping parts...
Getting groups of identical components...
Checking the seemingly identical parts group...
Merging field asked in the identical components groups...
Propagating field values to identical components...
Create HTML page for parts with custom pricing...
Initialising digikey                                                                                                                                                                                               
Starting new HTTPS connection (1): www.digikey.com:443                                                                                                                                                             
https://www.digikey.com:443 "GET / HTTP/1.1" 200 45736                                                                                                                                                             
# Configuring the distributors locale and currency...                                                                                                                                                              
https://www.digikey.com:443 "GET /en/resources/international HTTP/1.1" 200 None                                                                                                                                    
https://www.digikey.com:443 "GET / HTTP/1.1" 200 45736                                                                                                                                                             
# Scraping part data for each component group...                                                                                                                                                                   
Scraping digikey                                                                                                                                                                                                   
https://www.digikey.com:443 "GET /products/en?keywords=TLV8544PWR HTTP/1.1" 200 None                                                                                                                               
https://www.digikey.com:443 "GET /product-detail/en/texas-instruments/TLV8544PWR/296-47855-2-ND/6910133 HTTP/1.1" 200 None                                                                                         
https://www.digikey.com:443 "GET /product-detail/en/texas-instruments/TLV8544PWR/296-47855-1-ND/8133204 HTTP/1.1" 200 None                                                                                         
https://www.digikey.com:443 "GET /product-detail/en/texas-instruments/TLV8544PWR/296-47855-6-ND/8133377 HTTP/1.1" 200 None                                                                                         
https://www.digikey.com:443 "GET /products/en?keywords=1276-1946-2-ND HTTP/1.1" 403 None                                                                                                                           
Starting new HTTPS connection (1): www.digikey.com:443                                                                                                                                                             
https://www.digikey.com:443 "GET / HTTP/1.1" 200 45736                                                                                                                                                             
Received 403, scraper possibly detected: Starting new session for https://www.digikey.com                                                                                                                          
https://www.digikey.com:443 "GET /products/en?keywords=1276-1946-2-ND HTTP/1.1" 200 None                                                                                                                           
https://www.digikey.com:443 "GET /product-detail/en/samsung-electro-mechanics/CL10B105KP8NNNC/1276-1946-1-ND/3890032 HTTP/1.1" 200 None                                                                            
https://www.digikey.com:443 "GET /product-detail/en/samsung-electro-mechanics/CL10B105KP8NNNC/1276-1946-6-ND/3892460 HTTP/1.1" 200 None                                                                            
https://www.digikey.com:443 "GET /products/en?keywords=C3216X5R0J336M130AC HTTP/1.1" 200 None                                                                                                                      
https://www.digikey.com:443 "GET /product-detail/en/tdk-corporation/C3216X5R0J336M130AC/445-4060-2-ND/1965648 HTTP/1.1" 200 None                                                                                   
https://www.digikey.com:443 "GET /product-detail/en/tdk-corporation/C3216X5R0J336M130AC/445-4060-1-ND/1965707 HTTP/1.1" 200 None                                                                                   
https://www.digikey.com:443 "GET /product-detail/en/tdk-corporation/C3216X5R0J336M130AC/445-4060-6-ND/1965766 HTTP/1.1" 200 None                                                                                   
https://www.digikey.com:443 "GET /products/en?keywords=RMCF0603FT4K70TR-ND HTTP/1.1" 403 None                                                                                                                      
Starting new HTTPS connection (1): www.digikey.com:443                                                                                                                                                             
https://www.digikey.com:443 "GET / HTTP/1.1" 200 45736                                                                                                                                                             
Received 403, scraper possibly detected: Starting new session for https://www.digikey.com                                                                                                                          
https://www.digikey.com:443 "GET /products/en?keywords=RMCF0603FT4K70TR-ND HTTP/1.1" 200 None                                                                                                                      
https://www.digikey.com:443 "GET /product-detail/en/stackpole-electronics-inc/RMCF0603FT4K70/RMCF0603FT4K70CT-ND/1943033 HTTP/1.1" 200 None                                                                        
https://www.digikey.com:443 "GET /product-detail/en/stackpole-electronics-inc/RMCF0603FT4K70/RMCF0603FT4K70DKR-ND/1943939 HTTP/1.1" 200 None                                                                       
https://www.digikey.com:443 "GET /product-detail/en/stackpole-electronics-inc/RMCF0603FG4K70/RMCF0603FG4K70TR-ND/1714016 HTTP/1.1" 200 None                                                                        
https://www.digikey.com:443 "GET /products/en?keywords=AK9753AE HTTP/1.1" 200 None                                                                                                                                 
https://www.digikey.com:443 "GET /product-detail/en/akm-semiconductor-inc/AK9753AE/974-1157-2-ND/7388309 HTTP/1.1" 200 None                                                                                        
https://www.digikey.com:443 "GET /product-detail/en/akm-semiconductor-inc/AK9753AE/974-1157-1-ND/7388315 HTTP/1.1" 200 None                                                                                        
[... cut ...]
  File "/home/epoulsen/.local/lib/python2.7/site-packages/kicost-0.1.47-py2.7.egg/kicost/spreadsheet.py", line 468, in add_globals_to_worksheet
    parts.sort(key=get_ref_key)
  File "/home/epoulsen/.local/lib/python2.7/site-packages/kicost-0.1.47-py2.7.egg/kicost/spreadsheet.py", line 467, in get_ref_key
    return [match.group('prefix'), match.group('ref_num'), match.group('subpart_num')]
AttributeError: 'NoneType' object has no attribute 'group'
```
[main.zip](https://github.com/xesscorp/KiCost/files/2255115/main.zip)

hildogjr, 2018-08-02:
Your issue is related with some behavior of your XML file. Debugging, after KiCost group the part, it return:

...
['D301', 'D302']
D301 {'libpart': 'device:D', 'manf#_qty': '1', 'value': 'BAS16GWJ', 'manf#': 'BAS16GWJ', 'spn': '1727-7320-2-ND', 'manf': 'Nexperia USA Inc.', 'footprint': 'Diodes_SMD:D_SOD-123'}
D302 {'libpart': 'device:D', 'manf#_qty': '1', 'value': 'BAS16GWJ', 'manf#': 'BAS16GWJ', 'spn': '1727-7320-2-ND', 'manf': 'Nexperia USA Inc.', 'footprint': 'Diodes_SMD:D_SOD-123'}
[]
[]
['J9']
J9 {'digikey#_qty': '1', 'libpart': 'conn:Conn_01x02', 'manf#_qty': '1', 'value': 'B2B-PH-SM4-TB(LF)(SN)', 'digikey#': '455-1734-2-ND', 'manf': 'JST', 'manf#': 'B2B-PH-SM4-TB(LF)(SN)', 'spr': 'digikey', 'footprint': 'Connectors_JST:JST_PH_B2B-PH-SM4-TB_02x2.00mm_Straight'}
...

The `[]` means two empty groups of components. I never saw this behavior before, could you provide more information of your schematic? (If used some not conventional filed in the parts, KiCad version and the "Edit table field" list).

I am trying to track this behavior.

hildogjr, 2018-08-03:
This issue appear to be created because the group algorithm and component without `digikey#` field.
The error just appear when Digikey distributor is active generating the empty groups.

MrSurly, 2018-08-03:
I do a fair amount of Python -- Is this something I could dig into, or is it pretty involved?  Maybe give me a few hints around where these hooks are?

hildogjr, 2018-08-04:
The problem is in the `group_parts()` function in the file `eda_tools.py`, need some correcting / re-manufacture. And I need to merge here the #266.
It is a function to group the the parts and propagate the the `manf#` / `digikey#` / ... to the semi-identical part (by default KiCost propagate the `manf#` to parts of same kind, footprint and value, that are empty).
All help and new developers are welcome.

hildogjr, 2018-08-04:
KiCost works in this sequence:
1. Read the file (files);
1. Merge the groups;
2. Group the parts and propagate the `manf#` (the function here are the issue);
3. Scrape the distributors (as multi-thread);
3a. Configure the currency and language of the distributor (in beta / just working for Digikey);
3b. Look for the specific part number;
3c. Look for tree of part that match if 3b doesn't return;
4. Assembly the spreadsheet.

mobilinkd, 2018-08-19:
I am hitting this bug. Are there any work-arounds for this?

hildogjr, 2018-09-15:
This bug are not present on new implementation #314  (will be released in few days).
If you would like to have a try, replace the current code installation by https://github.com/xesscorp/KiCost/tree/octopart

Re-open the discussion if the bug persists.

closed 2018-09-15 by hildogjr