hildogjr/KiCost #309: KiCost GUI is crashing when I am running it

ramprasadajay, 2018-08-05:
I recently installed Kicost on my ubuntu 18.04. When I run it, it's giving the following error.

```
(kicost:6862): Gtk-CRITICAL **: 16:57:57.123: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkScrollbar
[xcb] Unknown sequence number while processing queue
[xcb] Most likely this is a multi-threaded client and XInitThreads has not been called
[xcb] Aborting, sorry about that.
python3: ../../src/xcb_io.c:259: poll_for_event: Assertion `!xcb_xlib_threads_sequence_lost' failed.
Aborted (core dumped)
```
kicost is installed on python 3
4.0.3 gtk3 (phoenix) wxWidgets 3.0.5

hildogjr, 2018-08-05:
This problem is caused in the thread that the GUI use to run CLI because it is not able to change the GUI control from inside the thread.
It's a kwoned issue from wxPython and may be simple to solve (it as not happing on wxPython 4.0.1).

hildogjr, 2018-09-15:
@ramprasadajay 
KiCost is been reshaped in the new branch https://github.com/xesscorp/KiCost/tree/octopart, not released yet.
This, test it and re-open this issue if the error persists.

closed 2018-09-15 by hildogjr