hildogjr/KiCost #311: name 'unicode' is not defined

romain145, 2018-08-15:
using KiCost in the Linux Subsystem on Windows 10.
Command line: kicost -i isolator.xml --overwrite > kicost.log
Using kicost 0.1.46 from pip3 install kicost

Error:
Traceback (most recent call last):
  File "/home/romain/.local/bin/kicost", line 11, in <module>
    sys.exit(main())
  File "/home/romain/.local/lib/python3.5/site-packages/kicost/__main__.py", line 282, in main
    local_currency=args.currency)
  File "/home/romain/.local/lib/python3.5/site-packages/kicost/kicost.py", line 348, in kicost
    user_fields, '-'.join(variant) if len(variant)>1 else variant[0])
  File "/home/romain/.local/lib/python3.5/site-packages/kicost/spreadsheet.py", line 198, in create_spreadsheet
    prj_info[i_prj]['title'], wrk_formats['proj_info'])
  File "/home/romain/.local/lib/python3.5/site-packages/xlsxwriter/worksheet.py", line 66, in cell_wrapper
    return method(self, *args, **kwargs)
  File "/home/romain/.local/lib/python3.5/site-packages/xlsxwriter/worksheet.py", line 404, in write
    return self._write(row, col, *args)
  File "/home/romain/.local/lib/python3.5/site-packages/xlsxwriter/worksheet.py", line 437, in _write
    if token_type is unicode:
NameError: name 'unicode' is not defined

hildogjr, 2018-08-15:
Could you attach the XML file?

romain145, 2018-08-16:
XML file attached.

[isolator.xml.txt](https://github.com/xesscorp/KiCost/files/2292886/isolator.xml.txt)

romain145, 2018-08-16:
The scrape finished now, I've rerun it, but the xlsx file doesn't seem to be complete (prices missing).
Could it be a problem with my install?
[isolator.xlsx](https://github.com/xesscorp/KiCost/files/2293131/isolator.xlsx)

romain145, 2018-08-16:
After updating the schematic and exporting the BOM again, I get a different error:

Z:\Dropbox\Funtronic\Projects\Electronica\isolator>kicost -i isolator.xml --overwrite --debug > kicost.log
Progress:  95%|################################################################8   | 206/216 [15:39<04:26, 26.62s/part]Traceback (most recent call last):
  File "c:\python27\lib\runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "c:\python27\lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "C:\Python27\Scripts\kicost.exe\__main__.py", line 9, in <module>
  File "c:\python27\lib\site-packages\kicost\__main__.py", line 282, in main
    local_currency=args.currency)
  File "c:\python27\lib\site-packages\kicost\kicost.py", line 329, in kicost
    res_dist = res_proc.get()
  File "c:\python27\lib\multiprocessing\pool.py", line 567, in get
    raise self._value
NameError: global name 'pn' is not defined

XML file:
[isolator.xml.txt](https://github.com/xesscorp/KiCost/files/2293890/isolator.xml.txt)

KiCost log:
[kicost.log](https://github.com/xesscorp/KiCost/files/2293892/kicost.log)

hildogjr, 2018-08-16:
Are you using KiCost from the PYPI or from the last code version here?
If PYPI, please, replace with the new file here or update the installation (I just released 0.1.47 right now and this 'pn' problem was already fixed).
>> pip3 install --upgrade kicost
[isolator.xlsx](https://github.com/xesscorp/KiCost/files/2293976/isolator.xlsx)

closed 2018-08-16 by hildogjr