hildogjr/KiCost #312: xlsx has problems on opening and is incomplete

romain145, 2018-08-16:
Using KiCost 0.1.47 on linux subsystem on Windows 10:
$ kicost -i isolator.xml --overwrite --exclude digikey > kicost.log

I excluded DigiKey because it was generating a lot of "Received 403, scraper possibly detected: Starting new session for https://www.digikey.com"

XML file attached:
[isolator.xml.txt](https://github.com/xesscorp/KiCost/files/2295479/isolator.xml.txt)

XLSX file attached:
[isolator.xlsx](https://github.com/xesscorp/KiCost/files/2295475/isolator.xlsx)

romain145, 2018-08-22:
Using KiCost 0.1.47 on linux subsystem on another Windows 10 machine:
$ kicost -wi isolator.xml

it generates the same incomplete XLSX file (although I don't have the 403 error this time).
The cells are not "Named" and the conditional formatting isn't set.

hildogjr, 2018-08-23:
Duplicated from #313

closed 2018-08-23 by hildogjr