hildogjr/KiCost #313: Scraping fails without error

romain145, 2018-08-17:
KiCost ends the scraping before reaching the end without error.

romain@vps307675:~$ kicost -i isolator.xml --debug > kicost.log
Progress:  67%|██████████████████████████▋             | 144/216 [13:01<24:31, 20.44s/part]romain@vps307675:~$ tail kicost.log
Part ['U6'] not found at digikey.
Received 403, scraper possibly detected: Starting new session for https://www.digikey.com
Received 403, scraper possibly detected: Starting new session for https://www.digikey.com
Part ['D1'] not found at digikey.
Received 403, scraper possibly detected: Starting new session for https://www.digikey.com
Received 403, scraper possibly detected: Starting new session for https://www.digikey.com
Part ['Q1', 'Q2', 'Q4', 'Q5', 'Q6', 'Q7'] not found at digikey.
Received 403, scraper possibly detected: Starting new session for https://www.digikey.com
Received 403, scraper possibly detected: Starting new session for https://www.digikey.com
Part ['J4'] not found at digikey.
romain@vps307675:~$ kicost -v
KiCost 0.1.47
romain@vps307675:~$ uname -a
Linux vpsxxx.ovh.net 3.16.0-4-amd64 #1 SMP Debian 3.16.43-2+deb8u2 (2017-06-26) x86_64 GNU/Linux

[kicost.log](https://github.com/xesscorp/KiCost/files/2298456/kicost.log)
[isolator.xml.txt](https://github.com/xesscorp/KiCost/files/2298457/isolator.xml.txt)

hildogjr, 2018-08-18:
@romain145 could you provide a complete log?
>> kicost -wi %YOUR_FILE% --include digikey -s --debug 8 > log_file.txt

romain145, 2018-08-18:
```
romain@vps307675:~$ kicost -wi isolator.xml --include digikey -s --debug 8 > kicost.log
Started KiCost v.0.1.47 on Linux-3.16.0-4-amd64-x86_64-with-debian-9.5(64bit) Python 2.7.13
Exchange rate: 1 EUR = 1.16 USD
# Getting from XML 'isolator.xml' KiCad BoM...
Getting authorship data...
Getting parts library...
Getting components...
# Removing do not populate parts...
Splitting subparts in the manufacture / distributors codes...
Q1 >> ['manf#']
part/qty>> IPB044N15N5          part>>IPB044N15N5       qty>>
{u'datasheet': 'https://www.wolfspeed.com/media/downloads/145/C3M0065090J.pdf', u'libpart': 'mos_to267-7:IPB044N15N5', u'manf#': 'IPB044N15N5', u'footprint': 'isolator:TO-263-7_Tab_NoPin4', 'value': 'IPB044N15N5'}
LED1 >> ['manf#']
part/qty>> 150060RS55040                part>>150060RS55040     qty>>
{u'datasheet': '~', u'libpart': 'Device:LED', u'manf#': '150060RS55040', u'footprint': 'LED_SMD:LED_0805_2012Metric_Pad1.15x1.40mm_HandSolder', 'value': 'RED'}
Q3 >> ['manf#']
part/qty>> 2N7002               part>>2N7002    qty>>
{u'datasheet': 'https://www.fairchildsemi.com/datasheets/2N/2N7002.pdf', u'libpart': 'Transistor_FET:2N7002', u'manf#': '2N7002', u'footprint': 'Package_TO_SOT_SMD:SOT-23', 'value': '2N7002'}
U7 >> ['manf#']
part/qty>> TMP36GRT             part>>TMP36GRT  qty>>
{u'datasheet': 'SOT-23-5 Analog Devices', u'libpart': 'TMP36:TMP36', u'manf#': 'TMP36GRT', u'footprint': 'Package_TO_SOT_SMD:SOT-23-5', 'value': 'TMP36'}
U6 >> ['manf#']
part/qty>> PC3H4J00001H         part>>PC3H4J00001H      qty>>
{u'footprint': 'isolator:PC3H4', u'libpart': 'pc3h4:PC3H4', 'value': 'PC3H4', u'manf#': 'PC3H4J00001H'}
U1 >> ['manf#']
part/qty>> 171021501            part>>171021501 qty>>
{u'footprint': 'isolator:VDRM_171021501', u'libpart': 'MagICModule:MagIC_VDRM', 'value': 'MagIC_171021501', u'manf#': '171021501'}
U3 >> ['manf#']
part/qty>> VOM1271T             part>>VOM1271T  qty>>
{u'libpart': 'VOM1271T:VOM1271T', u'field8': 'VOM1271T', u'field6': 'Unavailable', u'field7': 'None', u'field4': 'Vishay Semiconductor', u'field5': 'SOP-4 Vishay', u'datasheet': 'Relay SSR 50mA 1.6V DC-IN 4-Pin SOP T/R', 'value': 'VOM1271T', u'manf#': 'VOM1271T', u'footprint': 'Package_SO:SO-4_4.4x4.3mm_P2.54mm'}
U2 >> ['manf#']
part/qty>> CY8C4245AXI-483              part>>CY8C4245AXI-483   qty>>
{u'manf#': 'CY8C4245AXI-483', u'libpart': 'psoc:CY8C4245AXI-483', 'manf': 'Cypress', u'footprint': 'Package_QFP:TQFP-44_10x10mm_P0.8mm', 'value': 'CY8C4245AXI-483'}
C4 >> ['manf#']
part/qty>> CC0603KRX7R8BB105            part>>CC0603KRX7R8BB105 qty>>
{u'datasheet': '~', u'libpart': 'Device:C', u'manf#': 'CC0603KRX7R8BB105', u'footprint': 'Capacitor_SMD:C_0603_1608Metric', 'value': '1u'}
C11 >> ['manf#']
part/qty>> CC0603KRX7R9BB104            part>>CC0603KRX7R9BB104 qty>>
{u'datasheet': '~', u'libpart': 'Device:C', u'manf#': 'CC0603KRX7R9BB104', u'footprint': 'Capacitor_SMD:C_0603_1608Metric', 'value': '100n'}
C6 >> ['manf#']
part/qty>> 875105144010         part>>875105144010      qty>>
{u'footprint': 'Capacitor_SMD:CP_Elec_6.3x5.4', u'libpart': 'Device:CP', 'value': '330u 6.3V', u'manf#': '875105144010'}
R10 >> ['manf#']
part/qty>> PCF0603R-24K3BT1             part>>PCF0603R-24K3BT1  qty>>
{u'datasheet': '~', u'libpart': 'Device:R', u'manf#': 'PCF0603R-24K3BT1', u'footprint': 'Resistor_SMD:R_0603_1608Metric', 'value': '24.3k 0.1%'}
R11 >> ['manf#']
part/qty>> RC0603FR-0710KL              part>>RC0603FR-0710KL   qty>>
{u'datasheet': '~', u'libpart': 'Device:R', u'manf#': 'RC0603FR-0710KL', u'footprint': 'Resistor_SMD:R_0603_1608Metric', 'value': '10k'}
D3 >> ['manf#']
part/qty>> BZA856A              part>>BZA856A   qty>>
{u'footprint': 'Package_TO_SOT_SMD:SOT-353_SC-70-5', u'libpart': 'bza856al:BZA856AL', 'value': 'BZA856A', u'manf#': 'BZA856A'}
R34 >> ['manf#']
part/qty>> RC0603FR-072K2L              part>>RC0603FR-072K2L   qty>>
{u'datasheet': '~', u'libpart': 'Device:R', u'manf#': 'RC0603FR-072K2L', u'footprint': 'Resistor_SMD:R_0603_1608Metric', 'value': '2.2k'}
R4 >> ['manf#']
part/qty>> ERJ3GEYJ115V         part>>ERJ3GEYJ115V      qty>>
{u'datasheet': '~', u'libpart': 'Device:R', u'manf#': 'ERJ3GEYJ115V', u'footprint': 'Resistor_SMD:R_0603_1608Metric', 'value': '1.1M 1%'}
R6 >> ['manf#']
part/qty>> CPF0603B174KE1               part>>CPF0603B174KE1    qty>>
{u'datasheet': '~', u'libpart': 'Device:R', u'manf#': 'CPF0603B174KE1', u'footprint': 'Resistor_SMD:R_0603_1608Metric', 'value': '174k 0.1%'}
R7 >> ['manf#']
part/qty>> KTR10EZPF3903                part>>KTR10EZPF3903     qty>>
{u'footprint': 'Resistor_SMD:R_0805_2012Metric', u'libpart': 'Device:R', 'value': '390k', u'manf#': 'KTR10EZPF3903'}
R1 >> ['manf#']
part/qty>> RN73C1J52K3BTDF              part>>RN73C1J52K3BTDF   qty>>
{u'datasheet': '~', u'libpart': 'Device:R', u'manf#': 'RN73C1J52K3BTDF', u'footprint': 'Resistor_SMD:R_0603_1608Metric', 'value': '52.3k 0.1%'}
R2 >> ['manf#']
part/qty>> KTR10EZPF3903                part>>KTR10EZPF3903     qty>>
{u'footprint': 'Resistor_SMD:R_0805_2012Metric', u'libpart': 'Device:R', 'value': '390k', u'manf#': 'KTR10EZPF3903'}
[... 80 lines cut ...]
Qty>> ['R10']    1*PCF0603R-24K3BT1
Qty>> ['R7', 'R2']       1*KTR10EZPF3903
Qty>> ['R28', 'R25', 'R31']      1*ERJ-3GEYJ181V
Qty>> ['R36', 'R30', 'R3', 'R33', 'R11', 'R18', 'R19', 'R24', 'R27', 'R20']      1*RC0603FR-0710KL
Qty>> ['R35', 'R34']     1*RC0603FR-072K2L
Qty>> ['R4']     1*ERJ3GEYJ115V
Qty>> ['R6']     1*CPF0603B174KE1
[... cut ...]
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
  File "/home/romain/.local/lib/python2.7/site-packages/tqdm/_monitor.py", line 72, in run
  File "/home/romain/.local/lib/python2.7/site-packages/tqdm/_tqdm.py", line 106, in __enter__
  File "/home/romain/.local/lib/python2.7/site-packages/tqdm/_tqdm.py", line 99, in acquire
  File "/usr/lib/python2.7/threading.py", line 168, in acquire
<type 'exceptions.TypeError'>: 'NoneType' object is not callable
```

romain145, 2018-08-20:
Doing the same test on LSS under W10, the scraping finishes and the generated xlsx is valid (with only digi key results of course). Is it an issue with Python 2.7? How can I force KiCost to use Python 3.5 on the VPS?

hildogjr, 2018-08-23:
My system is also Py3. Could you test on it, @romain145?

romain145, 2018-08-23:
Test ran on LSS under Win10 with the default Python (2.7.12):
```
$ uname -a
Linux VIGLEN 4.4.0-17134-Microsoft #137-Microsoft Thu Jun 14 18:46:00 PST 2018 x86_64 x86_64 x86_64 GNU/Linux
$ python -V
Python 2.7.12
$ kicost -v
KiCost 0.1.47
$ kicost -wi isolator.xml --include digikey -s --debug 8 > log_file.txt
Started KiCost v.0.1.47 on Linux-4.4.0-17134-Microsoft-x86_64-with-Ubuntu-16.04-xenial(64bit) Python 2.7.12
Exchange rate: 1 EUR = 1.16 USD
# Getting from XML 'isolator.xml' KiCad BoM...
Getting authorship data...
Getting parts library...
Getting components...
# Removing do not populate parts...
Splitting subparts in the manufacture / distributors codes...
Q1 >> ['manf#']
part/qty>> IPB044N15N5          part>>IPB044N15N5       qty>>
{u'datasheet': 'https://www.wolfspeed.com/media/downloads/145/C3M0065090J.pdf', u'libpart': 'mos_to267-7:IPB044N15N5', u'manf#': 'IPB044N15N5', u'footprint': 'isolator:TO-263-7_Tab_NoPin4', 'value': 'IPB044N15N5'}
LED1 >> ['manf#']
part/qty>> 150060RS55040                part>>150060RS55040     qty>>
{u'datasheet': '~', u'libpart': 'Device:LED', u'manf#': '150060RS55040', u'footprint': 'LED_SMD:LED_0805_2012Metric_Pad1.15x1.40mm_HandSolder', 'value': 'RED'}
Q3 >> ['manf#']
part/qty>> 2N7002               part>>2N7002    qty>>
{u'datasheet': 'https://www.fairchildsemi.com/datasheets/2N/2N7002.pdf', u'libpart': 'Transistor_FET:2N7002', u'manf#': '2N7002', u'footprint': 'Package_TO_SOT_SMD:SOT-23', 'value': '2N7002'}
U7 >> ['manf#']
part/qty>> TMP36GRT             part>>TMP36GRT  qty>>
{u'datasheet': 'SOT-23-5 Analog Devices', u'libpart': 'TMP36:TMP36', u'manf#': 'TMP36GRT', u'footprint': 'Package_TO_SOT_SMD:SOT-23-5', 'value': 'TMP36'}
U6 >> ['manf#']
part/qty>> PC3H4J00001H         part>>PC3H4J00001H      qty>>
{u'footprint': 'isolator:PC3H4', u'libpart': 'pc3h4:PC3H4', 'value': 'PC3H4', u'manf#': 'PC3H4J00001H'}
U1 >> ['manf#']
part/qty>> 171021501            part>>171021501 qty>>
{u'footprint': 'isolator:VDRM_171021501', u'libpart': 'MagICModule:MagIC_VDRM', 'value': 'MagIC_171021501', u'manf#': '171021501'}
U3 >> ['manf#']
part/qty>> VOM1271T             part>>VOM1271T  qty>>
{u'libpart': 'VOM1271T:VOM1271T', u'field8': 'VOM1271T', u'field6': 'Unavailable', u'field7': 'None', u'field4': 'Vishay Semiconductor', u'field5': 'SOP-4 Vishay', u'datasheet': 'Relay SSR 50mA 1.6V DC-IN 4-Pin SOP T/R', 'value': 'VOM1271T', u'manf#': 'VOM1271T', u'footprint': 'Package_SO:SO-4_4.4x4.3mm_P2.54mm'}
U2 >> ['manf#']
part/qty>> CY8C4245AXI-483              part>>CY8C4245AXI-483   qty>>
{u'manf#': 'CY8C4245AXI-483', u'libpart': 'psoc:CY8C4245AXI-483', 'manf': 'Cypress', u'footprint': 'Package_QFP:TQFP-44_10x10mm_P0.8mm', 'value': 'CY8C4245AXI-483'}
C4 >> ['manf#']
part/qty>> CC0603KRX7R8BB105            part>>CC0603KRX7R8BB105 qty>>
{u'datasheet': '~', u'libpart': 'Device:C', u'manf#': 'CC0603KRX7R8BB105', u'footprint': 'Capacitor_SMD:C_0603_1608Metric', 'value': '1u'}
C11 >> ['manf#']
part/qty>> CC0603KRX7R9BB104            part>>CC0603KRX7R9BB104 qty>>
{u'datasheet': '~', u'libpart': 'Device:C', u'manf#': 'CC0603KRX7R9BB104', u'footprint': 'Capacitor_SMD:C_0603_1608Metric', 'value': '100n'}
C6 >> ['manf#']
part/qty>> 875105144010         part>>875105144010      qty>>
{u'footprint': 'Capacitor_SMD:CP_Elec_6.3x5.4', u'libpart': 'Device:CP', 'value': '330u 6.3V', u'manf#': '875105144010'}R10 >> ['manf#']
part/qty>> PCF0603R-24K3BT1             part>>PCF0603R-24K3BT1  qty>>
{u'datasheet': '~', u'libpart': 'Device:R', u'manf#': 'PCF0603R-24K3BT1', u'footprint': 'Resistor_SMD:R_0603_1608Metric', 'value': '24.3k 0.1%'}
R11 >> ['manf#']
part/qty>> RC0603FR-0710KL              part>>RC0603FR-0710KL   qty>>
{u'datasheet': '~', u'libpart': 'Device:R', u'manf#': 'RC0603FR-0710KL', u'footprint': 'Resistor_SMD:R_0603_1608Metric', 'value': '10k'}
D3 >> ['manf#']
part/qty>> BZA856A              part>>BZA856A   qty>>
{u'footprint': 'Package_TO_SOT_SMD:SOT-353_SC-70-5', u'libpart': 'bza856al:BZA856AL', 'value': 'BZA856A', u'manf#': 'BZA856A'}
R34 >> ['manf#']
part/qty>> RC0603FR-072K2L              part>>RC0603FR-072K2L   qty>>
{u'datasheet': '~', u'libpart': 'Device:R', u'manf#': 'RC0603FR-072K2L', u'footprint': 'Resistor_SMD:R_0603_1608Metric', 'value': '2.2k'}
R4 >> ['manf#']
part/qty>> ERJ3GEYJ115V         part>>ERJ3GEYJ115V      qty>>
{u'datasheet': '~', u'libpart': 'Device:R', u'manf#': 'ERJ3GEYJ115V', u'footprint': 'Resistor_SMD:R_0603_1608Metric', 'value': '1.1M 1%'}
R6 >> ['manf#']
part/qty>> CPF0603B174KE1               part>>CPF0603B174KE1    qty>>
{u'datasheet': '~', u'libpart': 'Device:R', u'manf#': 'CPF0603B174KE1', u'footprint': 'Resistor_SMD:R_0603_1608Metric', 'value': '174k 0.1%'}
R7 >> ['manf#']
part/qty>> KTR10EZPF3903                part>>KTR10EZPF3903     qty>>
{u'footprint': 'Resistor_SMD:R_0805_2012Metric', u'libpart': 'Device:R', 'value': '390k', u'manf#': 'KTR10EZPF3903'}
[... 87 lines cut ...]
Qty>> ['R7', 'R2']       1*KTR10EZPF3903
Qty>> ['R28', 'R25', 'R31']      1*ERJ-3GEYJ181V
Qty>> ['R36', 'R30', 'R3', 'R33', 'R11', 'R18', 'R19', 'R24', 'R27', 'R20']      1*RC0603FR-0710KL
Qty>> ['R35', 'R34']     1*RC0603FR-072K2L
Qty>> ['R4']     1*ERJ3GEYJ115V
Qty>> ['R6']     1*CPF0603B174KE1
Qty>> ['R32', 'R22', 'R23', 'R29', 'R26', 'R21']         1*RC0603FR-071KL
Qty>> ['SW1', 'SW2']     1*KMR211GLFS
Qty>> ['U1']     1*171021501
Qty>> ['U2']     1*CY8C4245AXI-483
Qty>> ['U6']     1*PC3H4J00001H
Qty>> ['U7']     1*TMP36GRT
Qty>> ['U5', 'U4', 'U3']         1*VOM1271T
[... cut ...]
Exception KeyError: KeyError(<weakref at 0x7f31c2665d60; to 'tqdm' at 0x7f31db3dd950>,) in <bound method tqdm.__del__ of Progress: 100%|███████████████████████████████████████████████████████████████████████| 36/36 [11:17<00:00, 16.25s/part]> ignored
```

See attached xml BOM file, generated xlsx file and log file.
[isolator.xml.txt](https://github.com/xesscorp/KiCost/files/2314662/isolator.xml.txt)
[isolator.xlsx](https://github.com/xesscorp/KiCost/files/2314660/isolator.xlsx)
[log_file.txt](https://github.com/xesscorp/KiCost/files/2314661/log_file.txt)

Will now change the python version and rerun the test.

hildogjr, 2018-09-15:
@romain145 the KiCost scrape module are now deprecated. In fell day we will release the API concept.
Give a try downloading the code on the branch "octopart"
https://github.com/xesscorp/KiCost/tree/octopart

Check the result by this new code:
[isolator.xlsx](https://github.com/xesscorp/KiCost/files/2385960/isolator.xlsx)

mobilinkd, 2019-02-18:
Please consider re-opening this issue as the octopart experiment appears to be a complete bust.  I have downgraded to 0.1.47 to do anything useful with KiCost.

hildogjr, 2019-02-18:
@mobilinkd, @easyw , to reopen this discussion we will need someone with Python + web experience + java script working on.
Since the constant changes in the web distributors and the java script blocking scraps (your algorithm shall be classified as robot), keep it working stated to be a cumcubersome to the only two active developers.

I stated a campaign https://forum.kicad.info/t/kicost-call-of-developers/13271 to call some users to help us. But was unsuccessful.
Please, let us now if  someone can provide us some coding help.

easyw, 2019-02-19:
@hildogjr 
I think we would need:
1) a GH repo in which KiCost is downgraded to version 0.1.47 to let users to contribute easily to scrapers
or an option to use scrapers to the actual release

Then it would be easier to find if some user may will in contributing to scrapers as they did in the past...
As at the actual KiCost repo state, it is really difficult to contribute to scrapers...

At the moment we could even ask to @volunteerlabs if he could have a look at the digikey scraper, as he already implemented a working version for his tool...

hildogjr, 2019-02-19:
May not be working (due web site changes) but the scrape modules are available in https://github.com/xesscorp/KiCost/tree/scrape_modules

hildogjr, 2019-03-16:
Discussion continues at #346

closed 2019-03-16 by hildogjr