hildogjr/KiCost #314: octopart integration

xesscorp, 2018-09-03:
I've created a new branch called "octopart". It gets part information using only the Octopart API (no web scraping). It works for the half-dozen XML input files I've tried it on. It currently has these problems:

1. Price breaks for Digi-Key parts always seem to be for reeled quantities (3000, 6000, etc). The cut-tape, small-quantity pricing is not returned. I'm not sure if there is an option in the API for enabling this.
2. RS Components always seems to return an empty price list.
3. It only runs under Python 2.7 because it uses urllib. (This should be easy to fix.)

All the changes I made are in the kicost.py file. This may not be the best place for it, but it makes the changes obvious.

I also noticed that there is a large delay (several seconds) in kicost before the part data is fetched from Octopart (which takes less than a second). This is caused by another parallel scrape that seems to be checking for the distributor websites? If so, I guess we could remove that since Octopart is being used. (I can't say for sure: this is new code I've never seen before.)

romain145, 2018-09-03:
Python 3 uses 2 different modules for urllib parse and request:

```
diff --git a/kicost/kicost.py b/kicost/kicost.py
index df6e24c..0f99c90 100644
--- a/kicost/kicost.py
+++ b/kicost/kicost.py
@@ -362,14 +362,15 @@ def kicost(in_file, eda_tool_name, out_filename,
         logger.log(DEBUG_OVERVIEW, '# Getting part data from Octopart...')

         import json
-        import urllib
+        import urllib.parse
+        import urllib.request

         dist_xlate = {'Digi-Key':'digikey', 'Mouser':'mouser', 'Newark':'newark', 'Farnell':'farnell', 'RS Components':'rs', 'TME':'tme'}

         def get_part_info(query, parts):
-            url = 'http://octopart.com/api/v3/parts/match?queries=%s' % urllib.quote(json.dumps(query))
+            url = 'http://octopart.com/api/v3/parts/match?queries=%s' % urllib.parse.quote(json.dumps(query))
             url += '&apikey=96df69ba'
-            results = json.loads(urllib.urlopen(url).read())['results']
+            results = json.loads(urllib.request.urlopen(url).read())['results']
             for result in results:
                 i = int(result['reference'])
                 for item in result['items']:
```

Hope that helps.

xesscorp, 2018-09-03:
Regarding the API key: it's unique to an application. So I have registered KiCost with Octopart and that's the assigned key. Individual users do not need their own key. (This is something I did not realize before.)

romain145, 2018-09-03:
For some reason, the distributors parts number have a prefix, such as "2401_855-", or "2401_81-" that is probably appended by Octopart (The links are trackers that point to Octopart as well). Not sure what the prefix means though...
So the BOM copy/paste import on the distributor's website doesn't work.

xesscorp, 2018-09-03:
I just looked at my spreadsheets and they also have the prepended number. It looks like it's a unique code for each distributor. It may be safe to strip off the initial digits and the underscore to get the actual part number.

anderwm, 2018-09-04:
1) This is pretty awesome
2) Although it worked, I got a 

> Traceback (most recent call last):
  File "/home/andersonm/kicad5_pro/ve_kicost_oct/lib/python3.6/site-packages/tqdm-4.25.0-py3.6.egg/tqdm/_tqdm.py", line 885, in __del__
    self.close()
  File "/home/andersonm/kicad5_pro/ve_kicost_oct/lib/python3.6/site-packages/tqdm-4.25.0-py3.6.egg/tqdm/_tqdm.py", line 1090, in close
    self._decr_instances(self)
  File "/home/andersonm/kicad5_pro/ve_kicost_oct/lib/python3.6/site-packages/tqdm-4.25.0-py3.6.egg/tqdm/_tqdm.py", line 454, in _decr_instances
    cls.monitor.exit()
  File "/home/andersonm/kicad5_pro/ve_kicost_oct/lib/python3.6/site-packages/tqdm-4.25.0-py3.6.egg/tqdm/_monitor.py", line 52, in exit
    self.join()
  File "/usr/lib/python3.6/threading.py", line 1053, in join
    raise RuntimeError("cannot join current thread")
RuntimeError: cannot join current thread

3) I believe the special distributor part number is how Octopart is making money (and also why the API is free to use). They must get some royalty whenever somebody buys a part with their tracking part numbers. It's probably in our best interest to use their numbers if we can in the cart building process so it will remain free.

romain145, 2018-09-04:
@anderwm I've had the same error recently, see https://github.com/xesscorp/KiCost/issues/313#issuecomment-418120390

3. Then the BOM should probably be filled on [Octopart BOM tool](https://octopart.com/bom-tool). By clicking on "Create a New BOM", then there is a "paste" link on the left where to paste the data from the spreadsheet but even then it seems confused by the prefix and does not recognise the MPN.

anderwm, 2018-09-04:
3) It looks like that special number is just used by Octopart to build a link to Digikey/whoever with them as the referrer, so that is how they are making money. To get at the real digikey part number there is a field called "sku" under each "offer". It may be good enough to just include Octopart's link in the spreadsheet the way you are doing, but have the anchor be the actual sku. The way Kicost works for me I never visit the links that come out of it. I just use the comma separated text to start a cart/BOM at all the distributors I need. I'm not sure if there is a way for Octopart to make money on that or not...or even if we should care.
4) It appears that each packaging option comes in a separate "offer" in the response from Octopart. This could be nice as I have times I would like to buy all cut tape/reel and other times when I don't care. In any event, there are multiple offers from Digikey and if you don't take the "cut tape" one or "tube" one you might not get the quantities you would expect.

xesscorp, 2018-09-04:
Thanks, @anderwm. Using the sku field solves the problem with ordering from the distributors.

Thanks also for pointing out that the Octopart response includes multiple offers. By collecting the price breaks for all the offers from Digi-Key, I was able to build a complete price table.

anderwm, 2018-09-04:
No problem. Still need some way to pick which package type to use as the sku though(as Cat# in datasheet). If not you wind up with a bunch of reels in your cart that you have to change over. We could use the amount being purchased, or default to cut tape, or pass in something from the command line I guess.

xesscorp, 2018-09-04:
I'll try using the SKU from the offer with the smallest quantities in the list. If you order enough parts to consume a reel, I think they might automatically apply reel pricing.

hildogjr, 2018-09-04:
In Digikey, at least, you are advised to that (and order the next price break) when checkout the cart.

xesscorp, 2018-09-04:
Let's try it that way, then.

anderwm, 2018-09-04:
So possibly it is a Me problem then. But most of the Digikey part numbers are coming up as reel for me in the Cat#. For instance, CL21C090CBANNNC shows in the spreadshed as Digikey Cat# 1276-2558-2-ND and the -2 means reel.
I assumed that it was just because that happened to be the first or last offer that you came across from Digikey. I guess you have to keep track of which sku corresponds to the lowest minimum quantity, but that isn't the part number showing up for me.

xesscorp, 2018-09-04:
Right, I haven't fixed the SKU problem.

anderwm, 2018-09-04:
sorry...got it

xesscorp, 2018-09-04:
KiCost now selects the SKU, available quantity, and web page from the cut-tape version. I'm distinguishing the cut-tape part by looking for the offer with the smallest quantity difference between the first two entries in its pricing table. This seems to work after my extensive testing on a single example. Each part offer also has a "packaging" field that could be used if that gives more accurate results.

hildogjr, 2018-09-05:
We could set some road map:

1. Finish the Octopart main tests;
2. Pass the code into `distributors/octopart.py`;
a. Re-mafacture the the `distributor_dict` dict() if necessary;
b. Fix the `spreadsheet.py` if some bug found;
c. So add more distributors here (add Arrow, ... #116) using the Octopart API (should be easy at this point, like just some definition at the `octopart` class() initialization);
3. Merge with the main branch;
a. Rename the scrape file routines to `*_webscrap` (e.g. `digikey_webscrape.py`, `mouse_webscrape.py`);
b. Use the web scrape just in complete failure of APIs (or at least keep for history and future decisions);
c. May be possible to create the others APIs (e.g. `digikey.py`)

I like the folder structure `distributors` and `eda_tools` but each sub folder inside of `distributor/*` I think is too much to keep just 1 file inside, but could be the most pythonized way. Some opinions about?

anderwm, 2018-09-05:
Sounds good...I stole the Newark code and added Arrow just to see if it really was that easy. It seems to be:
https://github.com/anderwm/KiCost/tree/octopart

hildogjr, 2018-09-05:
@anderwm, are your `arrow/arrow.py` scrape module functional or just created to update the `distributor_dict`?

anderwm, 2018-09-05:
just to update the distributor_dict...it is all just the Newark code with find/replace on newark->arrow

hildogjr, 2018-09-05:
So OK @xesscorp, after you make the code migration to `octopart.py` and changes at `distributor_dict` (even if necessary to change the fileds, because now it is not necessary some of than). I will code something about #65.

mmmaisel, 2018-09-06:
@hildogjr I would create a new distributor module for the octopart API stuff in distributors/octopart.py.
The existing class structure / virtual methods from distributor.py should do fine for this. It may only need a new method for selecting the distributors which shall be retrieved.

xesscorp, 2018-09-07:
I pushed a new version to the octopart branch. The new code is encapsulated in the octopart.py file in the distributors directory. I also added Arrow as a distributor and stripped as much from the code as possible while still allowing it to run. Turns out we need almost nothing in those distributor files. I didn't remove any code from the original set of distributors.

Eventually, it might be best to create a PartsList object that handles part grouping and information procurement. Then the query_octopart function could be rolled into that as a method.

hildogjr, 2018-09-07:
Agreed @mmmaisel.
If you have some time to have a look in... (I really like the way that you organize the classes, easy to control and maintenance).
@xesscorp, so, could we remove the other distributors from octopart branch? (just to create and test a functional version. At this point we keep them at the main branch).

hildogjr, 2018-09-07:
Some propose for new organization of the dict() struct, the old one:

    distributor_dict.update({
            'arrow': {
                'octopart_name': 'Arrow Electronics, Inc.',
                'module': 'arrow',  # The directory name containing this file.
                'scrape': 'web',     # Allowable values: 'web' or 'local'.
                'label': 'Arrow',  # Distributor label used in spreadsheet columns.
                'order_cols': ['part_num', 'purch', 'refs'],  # Sort-order for online orders.
                'order_delimiter': ',',  # Delimiter for online orders.
                # Formatting for distributor header in worksheet.
                'wrk_hdr_format': {
                    'font_size': 14,
                    'font_color': 'white',
                    'bold': True,
                    'align': 'center',
                    'valign': 'vcenter',
                    'bg_color': '#000000'  # Arrow black.
                },
            }
    })

Could have the initialization at `octopart.py`, the `arrow` folder have not a topological function at KiCost. So:

    distributor_dict.update({
            'arrow': {
                'module': {'name': 'octopart', 'octopart_name': 'Arrow Electronics, Inc.', 'type': 'api'} ,     # Allowable values: 'scrape', 'api' or 'local'. (older 'web' make no sense)
                'order' : {'cols': ['part_num', 'purch', 'refs'], 'order_delimiter': ','}, # Sort-order for online orders & delimiter.
                # Formatting for distributor header in worksheet.
                'wrk_hdr_format': {
                    'label': 'Arrow',  # Distributor label used in spreadsheet columns.
                    'font_size': 14, 'font_color': 'white', 'bold': True,
                    'align': 'center', 'valign': 'vcenter', 'bg_color': '#000000'  # Arrow black.
                },
            }
    })

In my experience with the total code of KiCost, this will provide a better integration and separation between the informations.

xesscorp, 2018-09-07:
You can take most of the code out of the digikey, mouser, farnell, etc. modules and make them look like the arrow module. As long as that code still resides in the repo, we can get it back out if it's ever needed.

As for the organization of the distributor dictionary, I have no opinion. However, I would advise against too much refactoring at this particular time and, instead, concentrate on making a new release. The KiCost users haven't had access to any Mouser pricing for a while, and the Octopart version will restore that along with providing a 40x speed increase. And we can provide more distributors pretty easily. I think they'll care about that more than the internal organization of the code. Refactoring can be done in the background at a leisurely rate.

hildogjr, 2018-09-07:
I did some tests and I am grateful with the results.
@xesscorp, if KiCost pass in all tests we could just merge to main and release.
After, if @mmmaisel agree to make the class, I will start some code improvements. I miss the response of RS Components, but I think is because of the missing currency conversion (including this in the code improvement step).

mmmaisel, 2018-09-08:
I can start porting the octopart code to a distributor class during the next week.
In this process I would also like to merge the octopart branch to master (with an opt-in switch e.g. --octopart until it is considered stable).

I also found a regression: If a project uses only distributor part numbers instead of manufacturer part numbers there will be no results at all. However, if the distributor number is derived from the manufacturer number, there are correct results from octopart (tested on their web page) most of the time which are ignored by kicost.

mmmaisel, 2018-09-08:
I just looked at the proposed octopart implementation and realized that is it 100% incompatible with the current distributor class architecture.

The current architecture processes one distributor at a time then one part at a time, without knowing the total amount of parts in advance. Once the whole process is finished, the results are merged into the parts[id].property[distributor] dictionary. Handling local distreibutors is no problem in this process.

The proposed octopart code handles everything at once. It processes parts in batches, gets the results for all distributors at once and writes them directly to the parts dictionary without synchronization with the main thread. This completely breaks the current scheduling and local part handling (but it relies on some initializations made by the scraper classes).

In my opinion, we will definitely need a distributors/octopart class which replaces the distributors/distributor base class. Regarding the implementation of this class and its integration with kicost.py and the existing distributor classes I'm currently out of ideas.

Just removing the existing stuff looks like a bad idea for me because we'll loose the parallel processing of local distributors and we'll never know how it can be used in future.

Furthermore, we should specify a KiCost specific user-agent when using APIs. What do you think about "Mozilla/5.0 (compatible; KiCost/0.x; +https://github.com/xesscorp/KiCost)"?

xesscorp, 2018-09-08:
I've checked in the latest octopart branch. I restored the capability to get part information if only a distributor SKU is provided instead of the manufacturer's part number. My testing on it shows it to be working on a limited number of examples.

I removed the web scraping code from the distributor classes and made them look similar to the Arrow class. I don't see any downside to that since it's no longer used but it's still available in the repository if we want to resurrect it.

There is no longer any parallel code in KiCost. The only reason it was there originally was to make the scraping faster. Since Octopart relieves us of the web-scraping chore, I don't think parallel operations are currently needed.

I've run the Octopart KiCost on every example I have using Python 2.7 and 3.6 without any crashes or exceptions. I'll need to run the web-scraping version on these files and compare the results to verify the Octopart version is getting the same part information.

xesscorp, 2018-09-10:
I couldn't run the tests with the web-scraping version. Digi-key isn't showing up in the spreadsheet so they might have changed their web page structure, or maybe they've also moved to using Distil like Mouser. Scraping is taking over 20 seconds per part. In my opinion, the current version of KiCost is unusable. You might as well ready a release of the Octopart version. I've checked the spreadsheets and it appears to be getting the right prices and available quantities. Any flaws it might have will be minor inconveniences compared to the current release.

hildogjr, 2018-09-10:
I agree to release a v0.1.48 as this new system and check the users acceptance. In the middle we work to remove the not used code (it will vanish some commands such `-np`, ... and we will need to update the manuals).

mmmaisel, 2018-09-10:
As the new octopart code works very well I agree with @xesscrop that we don't need the web-scraping code anymore. Hence the distributor scraper code, its class organization and the parallelization complexity can be completely removed.

hildogjr, 2018-09-15:
I make a bigger clean up on commit #aa86f8d972945ad30071edb48fa4066b33b8d0e2
Removed folders unnecessary and put the initialization of the distributor dictionary on `octopart.py`.
@xesscorp, I restore the code of `local.py`, now in `distributor/local.py` (not in `distributor/local/local.py` anymore). The mistake of this deletion caused on first steps of migration made a not functional local module.
Also fix some secondary issues (for example, `local.py` was not working in multi-files BoM).

Next step: I need to clean up the `distributors/__init__.py` to make the automatic import (in the future may be possible to create `distributors/digikey.py` as API, for example).

hildogjr, 2018-09-15:
Trying sequential times I got

requests.exceptions.ConnectionError: HTTPConnectionPool(host='octopart.com', port=80): Max retries exceeded with url: /api/v3/parts/match?queries=%5B%7B%22mpn%22%3A+%22MFWZ+ba
...
(Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fa456059550>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',))

This have to be in the `octpart.py` code as a Exception to warning the user.

mmmaisel, 2018-09-19:
@hildogjr I think `fake_browser` is obsolete now as there is no need to fake some browser (including state/cookie tracking) for APIs.
However, I think a free standing function which setup some things (like a user-agent) prior to the `request` call may be useful.

If you remove fake_browser, I think all functions except `dist_init_distributor_dict` from "local.py", the `scrape_part` and `get_part_html_tree` functions from "distributor.py" and the dynamic module loader from `__init__.py` can be removed as well.

hildogjr, 2018-09-20:
More fix and clean up did.
1. Fix some issues on local distributor template (is marked as #TODO because the string configuration should be shared with the `eda_tools` package);
2. Implemented the #65 and #315 (currency conversion). Needs some improvement (that I have in plan) but I miss the definitions and distributor class standardization.

hildogjr, 2018-09-25:
I think this is full functional. Do you all agree? (ready to release with some more documentation check?)
I am moving forward and create
https://github.com/xesscorp/KiCost/upload/octopart_class
branch with next steps of evolution:

1) Distributors new class format;
2) Eda also became a class format;
3) Each distributor / eda will be a file and not a folder/submodule.

xesscorp, 2018-09-26:
I assume the octopart branch is ready to release. The only real change to the documentation that I can think of would be to remove these options that no longer apply:

-s
-np
--rt
--throttling_delay

Then move the current master branch to something like "web_scraping" and make the octopart branch into the master. Then release it on PyPi.

hildogjr, 2018-09-26:
@xesscorp, I already removed these from `octopart` branch (before start my work on `octopart_class`, my vision of next improve in the code). I recommend:

1. Check the documentation (I removed something from there, but a double check will be nice);
2. Create a `scrape_old` from `main` branch just for historic reason (could be interesting for future survey);
3. `octopart` -> `main`
4. Release.

5. Start working on `octopart_class`.

hildogjr, 2018-09-26:
@xesscorp, I already removed this from documentation, but could you do the double check?

Something more: since march/2018 PyPI changed the documentation file style interpreter. I had to reconfigure the style to *.MD instead *.RST (our RST doesn't pass on the compliance), what created a bad look in the page https://pypi.org/project/kicost/

xesscorp, 2018-09-26:
I looked at the documentation. Seems OK except the "-np" option is still listed in the "Command-Line Options" section. Also, the "Parallel Web Scraping" section needs to be removed.

In regards to the README.RST and HISTORY.RST files, you can easily convert those into markdown files using pandoc:

pandoc readme.rst -o readme.md
pandoc history.rst -o history.md

The problem is you still need to keep the RST files because the documentation uses those to generate the manual in the docs directory using Sphinx unless you translate the doc files into markdown as well.

hildogjr, 2018-09-27:
Did the remove in the docs. Also change some 'scrape' words to 'scrape/query'.

Could we use `panddoc` on the manual/site files?

hildogjr, 2018-09-30:
We need release the new version `octopart` (now already old, `octopart_class` will take as development branch, see #320).
The numbers of users complaining problem about scrape problem are growing, at least 4 last week contacted me.

@xesscorp, could you fix th RST files?

xesscorp, 2018-10-01:
I added readme.md and history.md to the repository. The readme.rst and history.rst files are still there. You'll need to regenerate the markdown versions whenever you modify the RST files.

hildogjr, 2018-10-03:
Released as 1.0.0
There are issues already see by me, some fix in `octopar_class` (that will not release because may new check of others here).
Starting merging the branch and organizing the repo.

hildogjr, 2018-10-07:
Re-opened because KiCost need to deal and warning with "HTTP Status Codes" of https://octopart.com/api/docs/v3/overview

hildogjr, 2018-10-23:
#331 fix will have to contemplate this enhancement.

closed 2018-10-23 by hildogjr