hildogjr/KiCost #317: octopart doesn't return out of stock components

romain145, 2018-09-10:
Using the octopart branch.

Some references were not found on octopart for some reason.
The BOM contains the capacitor: ED1E471MNN1016
Which can be found here on TME: https://www.tme.eu/en/details/ed1e471mnn1016/tht-low-imp-electrolytic-capacitors/elite/
But octopart returns nothing for some reason: https://octopart.com/search?q=ED1E471MNN1016

I realise that's not linked to kicost, but I thought I'd ask if you had any idea why?

hildogjr, 2018-10-07:
Marked to new to be investigated.

romain145, 2018-10-08:
Answer from TME:
"Our tool is working really well. Stock list is refreshed every night. We don’t take any responsibility for Octopart."

Answer from Octopart:
"Our data comes directly from the distributors who list on our site, so typically when an offer is not showing on Octopart, it's because the distributor is not including that offer in the inventory feed they send to us. 

I will ask our data team to double check on this capacitor to make sure it is not being submitted by TME with some error happening on our end, but in all likelihood, TME has chosen not to send us this MPN for some reason—anything from low stock to lack of popularity to the part being phased out and/or replaced with a newer variant. 

I'm really sorry for the inconvenience, and I want to thank you for looking into this! Please feel free to let your TME rep know that you'd like to see all their offers on Octopart :) There likely won't be an immediate fix, but it never hurts to let your favorite distributors know that you're using Octopart and want to see all their parts listed there!"

hildogjr, 2018-10-08:
We just have to wait the Octopart internal update.
Thanks @romain145 to take a look into this.

closed 2018-10-08 by hildogjr