hildogjr/KiCost #319: After octopart update, partial partnumbers does not have a hit.

skumfidus, 2018-09-27:
Feature / Enhancement request
Correct me if I'm wrong, but I believe that previously if Manf# field contained "TPS71733DRV" this could return a hit on "TPS71733DRVR". I presume the reel option (extra "R" at the end of the part number) might have been chosen because this was the cheapest option. This is a nice feature if you use a PCBA manufacture that can handle all kinds of tapes, reels, and trays. 

I'm running https://github.com/xesscorp/KiCost.git@octopart and now it requires part numbers to be exact matches to a specific hit on Octopart. It would be nice if Kicost was able to just choose the cheapest option that matches a search on the part number. Possibly adding to the file a field that tells if the part number listed is an exact match or a partial.

hildogjr, 2018-09-27:
Before, the screpe module taken the most close matches which included "TPS71733DRVR" when asked by "TPS71733DRV".
We stay improving the new module. Please check your search at https://octopart.com to see the behavior.

hildogjr, 2018-09-27:
For implementation reference
https://octopart.com/api/docs/v3/rest-api#response-schemas-partsmatchrequest

hildogjr, 2019-04-02:
Merged with #346, it will be used the `match` API instead `part`.

closed 2019-04-02 by hildogjr