hildogjr/KiCost #320: Octopart (an EDAs) class format

hildogjr, 2018-09-28:
Since #314 is almost finished (missing fix new bug that will raise), @mmmaisel, could you give a lock (some help to me) in the branch
https://github.com/xesscorp/KiCost/tree/octopart_class

I:
1. Changing the folder structs of KiCost and trying to eliminate the submodule concept (it never help me to debug or leave me import separately);
2. Change the distributors class and it `__init__.py` (where I need your tip) to import the files with class and not packages/folder anymore;
3. Trying to do the same to the EDAs.

Are this concept pythonize?

hildogjr, 2018-09-29:
@xesscorp, please check the `octopart_class` branch. After we follow the work flow:

1. `Main` became `scrape`
2. `octopart` (documentation already updated) became `main`
3. Release KiCost
4. `octopart_class` merge into `Main` (old octopart)

hildogjr, 2018-10-03:
Release and merges did. Please, @mmmaisel & @xesscorp, check `main` branch for future codes.

mmmaisel, 2018-10-06:
> * Change the distributors class and it `__init__.py` (where I need your tip) to import the files with class and not packages/folder anymore;

I can't recommend doing this. The correct code works perfectly and consists of only two lines while the auto-import thing is much longer than 20 lines, buggy and probably dangerous (imaging some bad guy places some files within that directory that gets silently executed during the next run of KiCost).

The new master code looks good now and works well. I'll create a pull request with some more code cleanups soon.

closed 2018-10-06 by hildogjr