hildogjr/KiCost #321: Digikey (and others) dos not work anymore.

melody8801, 2018-09-28:
#### Issue / Problem report
Add:
1. Terminal error report / txt;
Hay

I have a problem to catch the informations of Digikey. 
I tried it by command line and GUI in Linux and Windows. Both variants did not work for me.

Here i postet the Logfiles i created. 

KiCad2.log is the original termial output, because  not everything was redirectet to KiCad.log
[KiCad.log](https://github.com/xesscorp/KiCost/files/2429300/KiCad.log)
[KiCad2.log](https://github.com/xesscorp/KiCost/files/2429301/KiCad2.log)

3. Write the command used to call KiCost (or the graphical interface configuration);
$ kicost -w -i "C:\Users\xxx\Downloads\KiCad.xml" -s -o "C:\Users\xxx\Downloads\test2.xlsx" --include digikey --debug 8 >"C:\Users\xxx\Downloads\KiCad.log"

4. One BoM to reproduce the error (with the EDA version).
[KiCad.xml.txt](https://github.com/xesscorp/KiCost/files/2429320/KiCad.xml.txt)
I tried a XML Export form V4.07 and V5.0

There was an update by Digikey. Did that crush the plugin?

Thanks for your plugin, its awesome.
and sorry for my english.

hildogjr, 2018-09-28:
Are you using the main branch / official release? If yes, please read
https://forum.kicad.info/t/kicost-not-finding-components-still-working-in-september-2018/12741/16

melody8801, 2018-09-28:
Thanks a lot, this worked for me.

closed 2018-09-28 by melody8801