hildogjr/KiCost #323: KiCost 0.1.47 does fails to find parts

nickoe, 2018-09-29:
```
$ kicost --version
/usr/lib/python2.7/site-packages/wx-3.0-gtk2/wx/_core.py:16629: UserWarning: wxPython/wxWidgets release number mismatch
  warnings.warn("wxPython/wxWidgets release number mismatch")
KiCost 0.1.47
```
I am aware of the wxpython warning, but I don't think this should affect this bug as the GUI is not supposed to be open. I installed from pip on Archlinux.

#### Issue / Problem report
```
kicost -i stmtest.xml --overwrite --include digikey mouser -d 2

<snip>

https://www.digikey.com:443 "GET /products/en?keywords=DP83848CVVX%2FNOPB HTTP/1.1" 200 None                                                                                                                                                   
https://www.digikey.com:443 "GET /product-detail/en/texas-instruments/DP83848CVVX-NOPB/DP83848CVVX-NOPBTR-ND/1870793 HTTP/1.1" 200 None                                                                                                        
Part ['U10'] not found at digikey.  

<snip>

collapsed_refs =  'U10'
fields =  {u'datasheet': 'http://www.ti.com/lit/ds/symlink/dp83848c.pdf',
 u'footprint': 'Package_QFP:LQFP-48_7x7mm_P0.5mm',
 u'libpart': 'Interface_Ethernet:DP83848C',
 u'manf#': u'DP83848CVVX/NOPB',
 'manf#_qty': '1',
 'value': 'DP83848C'}
info_dist =  {u'digikey': {}, u'mouser': {}}
manfcat_codes =  {u'digikey#': set([None]),
 u'local_template#': set([None]),
 'manf#': set([u'DP83848CVVX/NOPB']),
 u'mouser#': set([None])}
part_num =  {u'digikey': u'', u'mouser': u''}
price_tiers =  {u'digikey': {}, u'mouser': {}}
qty_avail =  {u'digikey': None, u'mouser': None}
refs =  ['U10']
url =  {u'digikey': '', u'mouser': ''}

<snip>

Exception KeyError: KeyError(<weakref at 0x7fc805045050; to 'tqdm' at 0x7fc803aecb10>,) in <bound method tqdm.__del__ of Progress: 100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 106/106 [01:21<00:00,  4.24s/part]> ignored
```

When I use that URL and search directly on digikey in a browser I do get the part with various packaging options.

https://www.digikey.dk/products/en?keywords=DP83848CVVX%2FNOPB

I have attached the bom and it was created with the 5.0 branch of kicad specifically Version: (5.0.1-dev-152-gf591f8741), release build

#### Doubts
That exection in the end seems suspicous.

Thank you.

nickoe, 2018-09-29:
Bom example attached
[stmtest.txt](https://github.com/xesscorp/KiCost/files/2431038/stmtest.txt)

hildogjr, 2018-09-30:
Duplicated from #321.

nickoe, 2018-09-30:
@hildogjr I am not quite sure what to make of this.

But I tried to install the octopart branch and that seems to work.
`pip install -U git+https://github.com/xesscorp/KiCost.git@octopart --user`

closed 2018-09-30 by nickoe

hildogjr, 2018-09-30:
The scrape modules are now decrepit. This branch that you installed will take its place.
We are just finishing the last aspects of documentation and user help.
I am making @xesscorp here, this is the second issue about in one week (not taking into account users in KiCad forum).

commit 7f64cfaa (2018-10-03): Fix #323