hildogjr/KiCost #324: KiCost subpart not working with octopart

nickoe, 2018-10-02:
#### Issue / Problem report
As described in the docs it should be possible to speficy subparts with the "part1;42:part2" to get 1pcs of part1 and 10 pcs of part2 into the BOM output. This does not seem to work properly. I get a row matching the number of parts, but the pcs are not respected.

I added the follwoing `455-1700-1-ND;455-1197-ND;10:455-1130-1-ND` to `digikey#`.
n.

I use the octopart branch of kicost from yesterday. I am not sure how to check for the latest version with pip. KiCost 0.1.47.

Commandline used:
`kicost -i subpartbom.xml --overwrite --include digikey farnell --debug`

I get no errors. I have attacehd the input xml and the resulting xlsx.
[subpartbom.txt](https://github.com/xesscorp/KiCost/files/2439992/subpartbom.txt)
[subpartbom.xlsx](https://github.com/xesscorp/KiCost/files/2439989/subpartbom.xlsx)

Looking at the debug output it do look like it parses the digikey part and qty correctly, but it does not seem like it is used in the query to octopart.
```
collapsed_refs =  'J2#3'
fields =  {'digikey#': '455-1130-1-ND',
 'digikey#_qty': '10',
 'footprint': 'None',
 'libpart': 'test:lala',
 'manf': '',
 'manf#': 'S10B-ZR-SM4A-TF(LF)(SN) ',
 'value': 'MY_CONN - p3/3'}
info_dist =  {'digikey': {}, 'farnell': {}, 'local_template': {}}
manfcat_codes =  {'digikey#': {'455-1130-1-ND'},
 'farnell#': {None},
 'local_template#': {None},
 'manf#': {'S10B-ZR-SM4A-TF(LF)(SN) '}}
part_num =  {'digikey': '455-1700-1-ND', 'farnell': '2399333', 'local_template': ''}
price_tiers =  {'digikey': {0: 0.0,
             1: 1.01,
             10: 0.889,
             100: 0.7672,
             500: 0.6277,
             1000: 0.50724,
             2000: 0.47554,
             3000: 0.45969000000000004,
             5000: 0.44384,
             10000: 0.42165},
 'farnell': {0: 0.0,
             1: 5.745261,
             30: 5.416289000000001,
             80: 5.0755680000000005,
             250: 4.746596,
             500: 4.382377},
 'local_template': {}}
qty_avail =  {'digikey': 2599, 'farnell': 802, 'local_template': None}
qty_increment =  {'digikey': 9, 'farnell': 29, 'local_template': None}
refs =  ['J2#3']
url =  {'digikey': 'https://octopart.com/click/track?ak=96df69ba&sig=068f204&sid=459&ppid=248864&vpid=1873498&ct=offers',
 'farnell': 'https://octopart.com/click/track?ak=96df69ba&sig=0fed374&sid=819&ppid=248864&vpid=141950469&ct=offers',
 'local_template': ''}
```

nickoe, 2018-10-02:
This is a potential duplicate of https://github.com/xesscorp/KiCost/issues/266

hildogjr, 2018-10-03:
You are using:
digikey# = "455-1700-1-ND;455-1197-ND;10:455-1130-1-ND"
manf# = "S10B-ZR-SM4A-TF(LF)(SN)"

This create some ambiguous interpretation. Is this part composed by one subpart as `manf#` or by 3 as `digikey#`. You should leave `manf#` blank or specify the association order between manf#<->digikey#.

As far I search your parts:
digikey# = "455-1700-1-ND;455-1197-ND;10:455-1130-1-ND"
manf# = "S10B-ZR-SM4A-TF(LF)(SN);;"

So:
digikey = ["455-1700-1-ND", "455-1197-ND", "10:455-1130-1-ND"]
manfCode = ["S10B-ZR-SM4A-TF(LF)(SN)", "", ""]

This means, the manf# is associated with the first digikey code and I did type correspondence codes to the other ones.

Re-open if the issue persists.

nickoe, 2018-10-03:
Hmm, ok. I would have expected the digikey# to override the manf# always.

May I suggest kicost to throw a warning if there is ambiguity? Or handle this in some more user friendly way.

I tried to remove the manf# field and sure enough it shows all the parts in the list, but it does not use the correct qty. I would expect it to say 10 for the third part when the board number is 1.

hildogjr, 2018-10-03:
A was review the code and there two issues here:

1) The old KiCost (with scrape module) prioritize the distributor# over manf# to remove ambiguity.
E.g: `5001` is a manufacture code of a pin, but we have to look for `36-5001-ND`. But `5001` works on Mouser and others.

2) I check and there isn't a code that do that in the release API version.

I change the issue title here to deal withe the this. Could you create a other issue for this suggested warning? (It will better for me to track).

nickoe, 2018-10-03:
@hildogjr yes, I can do that. But I just wanted to test your change, but it seems like you committed it on master. Is it correct that there is no octopart branch anymore?

I then tried to install with
`pip install -U git+https://github.com/xesscorp/KiCost.git@master --user`

But the spreadsheet does not contain any parts now. :S

hildogjr, 2018-10-03:
Yes, octopart became the oficial release and scrape module is just kept as history/backup

hildogjr, 2018-10-03:
Not working in full test branch. It will need for coding.

hildogjr, 2018-10-03:
Try now. I changed some parameters on the installer (we I fix some installer documentation in the branch) and the octopart for others files test.

nickoe, 2018-10-03:
Mmm, why? I think the install went fine.

On e6b5a87ead990ed7b4b8db68dd84d4393a0637c3 I am back to a state similar to before my bug reprot. That is it uses the same part for all. And I have the manf# field populated as well. Which bugs are you expecting to be fixed by now?

hildogjr, 2018-10-03:
I fixed the issue related with the Octopart API, but you still have to use semicolon and "blank codes":
digikey# = "455-1700-1-ND;455-1197-ND;10:455-1130-1-ND"
manf# = "S10B-ZR-SM4A-TF(LF)(SN);;"
because KiCost correlate:
"455-1700-1-ND" <-> "S10B-ZR-SM4A-TF(LF)(SN)"
"455-1197-ND" <-> "" So prioritize digikey#
"10:455-1130-1-ND" <-> ""

I tested with this modify version of your BOM.
[subpartbom.xml.txt](https://github.com/xesscorp/KiCost/files/2444270/subpartbom.xml.txt)

closed 2018-10-03 by hildogjr

nickoe, 2018-10-03:
Ok. I confirm that it will now search for all the parts, after I added the extra semicolons to the manf# to make "empty" parts there.

hildogjr, 2018-10-03:
There is some other issues that I realize: the warning about the ;; would be good and the quantities are not ok in yout BoM (the actual script take the quantity of 'manf#' and not from 'digikey#').
If you agree, please open 2 issues to track these fix.