hildogjr/KiCost #327: Qty not respected for subparts

nickoe, 2018-10-03:
After fixing issue https://github.com/xesscorp/KiCost/issues/324 with e6b5a87ead990ed7b4b8db68dd84d4393a0637c3. There are still an issue left.

#### Issue / Problem report
1. The qty is not respected for each subpart. Test file https://github.com/xesscorp/KiCost/files/2439992/subpartbom.txt

This bug is created with the motivation from https://github.com/xesscorp/KiCost/issues/324#issuecomment-426830132

hildogjr, 2018-10-03:
This limitation is already known in the older version but needs to be fix in Octopart API.

hildogjr, 2018-10-05:
@nickoe, test this fix. But you still to use the blank at 'manf#'

closed 2018-10-05 by hildogjr

hildogjr, 2018-10-06:
I added also the warning tip when detected the  ambiguous case.

nickoe, 2018-10-07:
I have just tesed master including your commit. I can confirm that it works as expected and it also erros on mismatch between part fields. Everything is good now. Thanks. :)

hildogjr, 2018-10-07:
If you miss some message or tip that could help users. Give us a hit, we need the users feedback.

nickoe, 2018-10-08:
As it seems to work as expected now I don't see anything missing. One could add a note in the doc that for multiple subparts they are mapped one to one, hence one may need to add empty fields. But as that is an error thrown by the script they user will catch it anyways.

commit d3ef61b1 (2018-10-05): Fix #327