hildogjr/KiCost #333: Enhancement Request - User-defined mfr_num field name

JuliaTruchsess, 2018-10-24:
I have a substantial number of projects that use a field called "Manufacturer's Part Number" (because that's how I roll), and it's quite a tedious chore to manually go through the schematics adding a new field that conforms to KiCost's list of acceptable field names and copying the data from my old field to the new field. It's just a string that you're going to use to look up data from an XML file - why not let the user specify whatever string they want?

Apart from that, it's awesome so far, thanks!

hildogjr, 2018-10-24:
@JuliaTruchsess, I will mark this to solve soon.
A work around, in ` kicost\edas\tools.py` at your installation, add a new definition to `field_name_translations`. Example:

Before:

    field_name_translations = {
        'mpn': 'manf#',
        'pn': 'manf#',

Your definition:

    field_name_translations = {
        'mpn': 'manf#',
        'Manufacturer\'s Part Number': 'manf#', ## The \' here is to scape the interpreter.
        'pn': 'manf#',

hildogjr, 2018-10-27:
Use:

    kicost ... --translate_fields 'Manufacturer\'s Part Number' manf#

in the call. Or type `--translate_fields 'Manufacturer\'s Part Number' manf#` in the "Extra commands" of the GUI, this will keep saved in the configuration.

closed 2018-10-27 by hildogjr