hildogjr/KiCost #336: Field "Manufacturer" in the input is renamed to "manf" in input, but not in spreadsheet

timoalho, 2018-10-25:
Specifically: if I have a field called "Manufacturer", it gets renamed to "manf" by the mapping in 'tools.py:97'. However, if I use the option '--fields Manufacturer', both a field called 'manf' and 'Manufacturer' appear in the output spreadsheet, with the 'Manufacturer' column being empty.

The renaming should be done consistently everywhere, or alternatively should be disabled for fields passed as arguments to '--field', under the assumption that the user really wants to keep the name they're using.

hildogjr, 2018-10-25:
Linked with #333

hildogjr, 2018-10-27:
The renaming will need a code refacture of parts of KiCost. I will add an assert not not allow the user to rename this.

hildogjr, 2018-10-27:
If you want to use "Manufacturer" as a user field remove from the internal dictionary using `--translate_field Manufacturer`, command just used to solve #333, so it can be used as personal field now.

commit 0760c146 (2018-10-27): Fix #336
closed 2018-10-27 by hildogjr