hildogjr/KiCost #338: EUR pricing in Kicost - some prices still listed in USD

mdeweerd, 2018-10-25:
Kicost gets Euro prices from Octoparts - great ;-).
 
There are a few improvements that could be done related to pricing:
I have some prices expressed in USD that I specified in a kicost.supplier.pricing field .  Example: 500:USD2.12;1000:USD1.90 .  I noted that a currency module for python was installed during the installation process and I was hoping that the USD prices would be converted to EUR somehow.

The comment "popup" in excel shows the "$" sign before the prices while they are in EUR.  It is possibly on purpose, but only the first two price breaks are listed - it would be interesting to have the full list.
Here is a screenshot:
[image]

Unfortunately Octopart prices are not correct and have fewer prices breaks than the distributer (I attach screenshots of the Octopart prices and the Distributor prices taken at the same moment).   So a part looks much more expensive than it actually is.
Kicost is reusing the right values from Octopart though - so no action for the Kicost tool 
Screenshots of the pricing according to Octopart and Mouser:
[image]
[image]

hildogjr, 2018-10-25:
@mdeweerd, confirm me, there is 2 issues:
1. The symbol should be correlated with the converted currency and the values are fine (converted correctly);
2. It is not shown the complete price break list as in the web sites.

About 1, yet, it uses a currency converter, so could exist some minimum difference between the converted value present by KiCost and by Octopart (Octopart also convert the prices but not provide the converted values in the API).

hildogjr, 2018-10-25:
The symbols issue was fixed. MIssing fix the break price issue.

mdeweerd, 2018-10-26:
1. The currency symbol in the comment popup should be the same as the currency displayed.
This is/was not the case (you say you fixed it);
Possibly, if the currency was converted it would be a nice to have to show the value in the original currency and in the converted currency.
Example:  "10: 2.00 EUR (2.51USD)" 
2. The price break list in the popup should be complete.  Currently only 2 lines are shown.
The prices shown were already the same as the ones that are used in the spreadsheet, so that is ok.
3. When prices are entered in the specific Kicost field in USD, but the spreadsheet is in EUR, then the prices should be converted from EUR to USD.  This is not currently the case.
I specified "500:USD2.12;1000:USD1.90" in the "kicost.supplier.pricing" field of a component and these prices were interpreted as EUR.   Ideally in the comment I would have something like "500: 1.87 EUR (2.12USD)" and of course also the correct pricing for the BOM line and total.

hildogjr, 2018-10-30:
@mdeweerd, I checked I could get all the price breaks. Could you re-check, and re-open if still? The others improvements are in the list.
[image]

I am first modifying KiCost to respect the API terms of use. #331

mdeweerd, 2018-10-30:
Hi
I do not know what means exactly "The others improvements are in the list.", here are my observations:

The currency symbol is now € when EUR is specified.
However, there are still these two behaviors:
- kicost.supplier.pricing="500:USD2.12;1000:USD1.90" has two price breaks, the sheet shows two prices, but the interpretation is not ok.  Minimum order is 500, 100 price break is not shown: 
[image]
- "500:USD2.12;1000:USD1.90" is expressed in USD, but the prices shown in EUR are the same.  I thought there would be a currency conversion.

I had to disable octopart to make this test as my API has been disabled.

hildogjr, 2018-10-30:
More than one issue reported in one topic usually create some misunderstood. I fix the symbols in the title.
I will let you up-to-date.

commit 3ef6e7e6 (2018-10-25): Fix #338 symbols
closed 2019-03-16 by hildogjr