hildogjr/KiCost #339: Cannot run kicost on windows

romain145, 2018-10-31:
Following the instructions here for windows: https://xesscorp.github.io/KiCost/docs/_build/singlehtml/index.html

I installed ActivePython-3.5.4.3504-win64-x64-404899.exe then opened a powershell and installed kicost : 
```
easy_install kicost
```
Then running it I got the error:
```
kicost
You don't have the wxPython dependence to run the GUI interface. Run once of the follow commands in terminal to install them:
pip3 install -U wxPython # For Windows & macOS
```
Which I did, and now running kicost returns the following:
```
kicost
Traceback (most recent call last):
  File "C:\Python35\Scripts\kicost-script.py", line 11, in <module>
    load_entry_point('kicost==1.0.3', 'console_scripts', 'kicost')()
  File "c:\python35\lib\site-packages\kicost-1.0.3-py3.5.egg\kicost\__main__.py", line 215, in main
    kicost_gui() # Use the user guide if no input is given.
  File "c:\python35\lib\site-packages\kicost-1.0.3-py3.5.egg\kicost\kicost_gui.py", line 995, in kicost_gui
    frame = formKiCost(None)
  File "c:\python35\lib\site-packages\kicost-1.0.3-py3.5.egg\kicost\kicost_gui.py", line 493, in __init__
    self.set_properties()
  File "c:\python35\lib\site-packages\kicost-1.0.3-py3.5.egg\kicost\kicost_gui.py", line 805, in set_properties
    s=numbers.get_currency_symbol(currency),
  File "c:\python35\lib\site-packages\babel-2.6.0-py3.5.egg\babel\numbers.py", line 143, in get_currency_symbol
    return Locale.parse(locale).currency_symbols.get(currency, currency)
AttributeError: 'NoneType' object has no attribute 'currency_symbols'
```

Am I missing something?

hildogjr, 2018-10-31:
@romain45, I am not a Windows user. But try to install by `pip` instead of `easy_install`, it been more stable recently.

hildogjr, 2018-11-02:
@xesscorp, it is needed a Windows user help here.
This error is related if the system not setting the `Locale` (language and country) used by the `babel` library to correctly present the values.
https://stackoverflow.com/questions/11234843/how-to-set-python-default-locale-on-windows

hildogjr, 2018-11-06:
Check this answer
https://forum.kicad.info/t/kicost-octopart-api/12940/7

rubix1911, 2019-03-13:
Hi, I am facing the same issue. Tried looking on https://stackoverflow.com/questions/11234843/how-to-set-python-default-locale-on-windows ;but i am not getting it. Please guide me.

rubix1911, 2019-03-13:
What system language to be set for overcoming the error? I changed US(English) to UK(English) but the error is same.

hildogjr, 2019-03-13:
On KiCad forum there is this thread
https://forum.kicad.info/t/kicost-on-windows-cant-run-gui/15084/3
But I haven't a Windows system to test it.

rubix1911, 2019-03-14:
Ohh, If possible can you test it on my windows system via teamviewer?

xesscorp, 2019-06-11:
I just ran the latest version of KiCost on my Win7 PC and it completes successfully on the one example I tried. The Digikey, Mouser and Farnell catalog numbers and available quantities were populated. However, for Digikey and Mouser, all the reported prices were $0.00, and for Farnell all the prices were left blank.

My question: **What needs to be done to issue a new, working release of KiCost?** I only see one bug in the list of issues, but it looks like there's more to be fixed than that.

hildogjr, 2019-06-12:
@xesscorp, could you indicated me the XML used? (if in the examples)
I am already using in the project of the laboratory that I resulted fine.

For a functional release, [despite my road map](https://github.com/xesscorp/KiCost/milestones) , it is just missing fix the Windows issues. All others are new functionalities and enhancements.

xesscorp, 2019-06-12:
I tried it with all my test cases. When KiCost completes successfully, the parts all have quantities, catalog #'s, and links to their pages at the distributor website. But their costs are always either $0.00 (for Digikey & Mouser), or completely blank (for Farnell).

A number of my test cases fail to complete. Usually there's an exception for a local variable named `part_query` that's referenced before it's defined. Other times a subroutine is returning `None` when it shouldn't.

I'll check some of these and see if I can find the errors.

hildogjr, 2019-06-12:
The API can return the currency value in `QUERY_AVAIABLE_CURRENCIES = {'GBP', 'EUR', 'USD'}`. I reserved this definition but I am just requiring `USD`.
Could it be that the current part/distributor just have available in other currencies?

xesscorp, 2019-06-12:
The problem with the $0.00 prices was caused by the spreadsheet: I had to type a quantity into the "Board Qty" field and hit return and then all the prices for Digikey and Mouser showed up. The Farnell prices are still missing and aren't being provided by kitspace, but all the other part data is there. I'll have to check that further and also look at the causes of the exceptions.

hildogjr, 2019-06-12:
I see that like me, you are using LibreOffice. By default it not evaluate the Excel formulas (with xlsx file) in the startup. I discovered this in a forum and can be change in the options.

xesscorp, 2019-06-12:
You're right! It took me forever to find that option even after you told me about it.

hildogjr, 2019-06-13:
Something to mention: I just leave the available distributors in PartInfo API
https://github.com/kitspace/partinfo/issues/7

xesscorp, 2019-06-13:
OK, I have all the errors fixed on my test suite. Now I can start on this problem with the GUI version.

easyw, 2019-06-13:
@hildogjr 
> I see that like me, you are using LibreOffice. By default it not evaluate the Excel formulas (with xlsx file) in the startup. I discovered this in a forum and can be change in the options.

Would it be possible to link the forum post or to post the way to enable it?

hildogjr, 2019-06-13:
Tools >> Options >> LibreOffice Calc >> Formula >> Recalculation on File Load >> Excel 2007 and never
[image]

easyw, 2019-06-13:
Thx :smile:

xesscorp, 2019-06-14:
I've checked in the changes. KiCost now works with both Python 2 & 3 on Windows.

hildogjr, 2019-06-14:
Nice, @xesscorp . Please, re-add future as requirement in `setup.py`, I made a big clean up some time ago and removed some packages not used (and updated the `tqdm` version due a reported issue).

xesscorp, 2019-06-14:
Done.

closed 2019-06-14 by xesscorp