hildogjr/KiCost #342: Octopart can return items not exactly matching part number

anderwm, 2018-12-13:
It happens on this part number '1.5SMC68CA' that multiple 'items' are returned by Octopart. One of the items returned is actually part number  '1.5SMC6.8CA' which is a different component entirely. I'm not sure if it isn't handling the dot incorrectly or what, but the other two 'items' returned are correct. However, in this case Kicost used this incorrect value to populate the spreadsheet.
Recommend a check like

>                 # Loop through the offers from various dists for this particular part.
                for item in result['items']:
                    if item['mpn'].upper() != parts[i].fields['manf#'].upper():
                        continue

at line 273 in dist_octopart.py

hildogjr, 2018-12-13:
If we look direct at the site
https://octopart.com/search?q=1.5SMC68CA&start=0

Yes, this issue need a check.

hildogjr, 2019-03-16:
Duplicated from #319

closed 2019-03-16 by hildogjr