hildogjr/KiCost #343: KiCost reports missing wxPython asking to install when it is already installed (Windows 7)

micronpn, 2018-12-21:
I installed:
1. Python 3.7.1
2. Kicost: easy_install kicost
then I executed kicost and it reports for missing wxPython
then I installed wxPython:
3. pip3 install -U wxPython
everything went well so I executed kicost but it still reports for missing wxPython.

What went wrong?

hildogjr, 2018-12-21:
Do you have more than one Python installation? If yes, please, execute

     kicost -wi %YOUR_BOM_FILE% --debug 8

you should have some response like:

     Started KiCost v.1.0.4 on Linux-4.4.0-140-generic-x86_64-with-Ubuntu-16.04-xenial(64bit) Python 3.5.2

Other point is, use `pip3 install kicost` instead `easy_install kicost` I are having less issues with the first installation.

micronpn, 2018-12-21:
No, I installed only Python 3

micronpn, 2018-12-21:
I retry using pip3: same situation, same result.

hildogjr, 2018-12-21:
Can you execute:

     python3
     >>> import wx
     >>> wx.version()

? Whats the answer?

micronpn, 2018-12-21:
kicost -wi %YOUR_BOM_FILE% --debug 8 runs in text mode and seems it needs a valid key for Octopart. I have mine but I don't know how give it to kicost.
But the main point is that it runs... so the problem seems limited to run the GUI.

hildogjr, 2018-12-21:
Yes. Appear that KiCost is not recognizing the wxPython installation or it is not correct.

A work around, if you known Python, is use your KEY in `kicost.py` file  (use `apiKeys = {'octopart':'YOUR_KEY'}` after the python function declaration).

If your not familiarized with the language, we need to know the reason of not understand the installation.

Can you run the `import wx` as above?

hildogjr, 2018-12-21:
Try to replace the installed files by that ones downloaded here (may be a old problem that I already solved but not release the new code due the KEY problem that we are experiencing).

With you have the Git installed on your machine `pip3 install -U git+https://github.com/xesscorp/KiCost.git` does this procedure.

micronpn, 2018-12-21:
I did 2 "experiments":
first
1. install python 2.7.15
2. install wxPython on python 2.7.15
3. test wxPython installation using import and running wxDemo: both working
4. install kicost using easy_install: a lot of warnings but the installation (seems) completed without any error
5. launch kicost: returns an error in kicost-scrypt.py line 11 and others
second
1-3. as before
4. install kicost using pip install -U kicost: a lot of warnings but the installation (seems) completed without any error
5. launch kicost: returns an error in runpy.py line 174 and others (errors are different then those of the first experiment).

Seems to me that you have many problems related to Python compatibility. Is there any kicost release working with Python 2.7.15?

hildogjr, 2018-12-21:
Yes, @micronpn, the new code (since we migrate from web scrape to API) is not (yet) compatible with Python 2. Something regard the class and definitions.

I am try to fix some issues, but nowadays the development team are just 2 persons. So, I am focus, in my rest time, on other bugs that affect the functionality even in Python 3.

I appreciate if you can fix some... https://forum.kicad.info/t/kicost-call-of-developers/13271

Did you tried to replace the code installed by `pip` by that one that you can download here?

Could you attache some of this errors/warnings?

micronpn, 2018-12-21:
I would like to attach the errors but unfortunately redirecting the output to a file captures only few lines,  I see many more in the console.
I just look at your code and my opinion is that before to introduce the new API you should had fix a "stable release" then refactor it to be Python agnostic (2.7.x and 3.x) and then API agnostic. Only after that introducing a new API should be done comfortably without breaking too much the existent code. 

So, the point is: is there a "stable" release to test or at least to start with? Possibly without problems running on Python 2.7 considering that wxPython has serious problems to run on Python 3 (at the moment at least).

hildogjr, 2018-12-21:
The older version (web scrape version) was stable in both Py 2 and 3. The problem was that Digikey and other distributors web sites were in constant changes (in them HTML) also adding javascript to block robots (and scrapes).
So we are forced to release quick the API version and fix the issues on the fly (I don't like this method neither, but again, it was coded by just 2 persons and asked by many users).
Conclusion: the old version is not working, not because of the code, but because of the changes in the HTMLs.

micronpn, 2018-12-21:
So, as today, there is no working release. What is the last release that worked until Digikey and others changed their code?

hildogjr, 2018-12-21:
https://github.com/xesscorp/KiCost/tree/scrape_modules

closed 2019-03-16 by hildogjr