hildogjr/KiCost #344: Can't use both GUI and command line

tom88jerry, 2019-01-06:
Hi,

I have problem using kicost. It's my first time and I have problem using it.

First I tried to use kicost gui but it showed me this.
C:\Users\ezzsl3>kicost
Traceback (most recent call last):
  File "c:\python36\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\python36\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Python36\Scripts\kicost.exe\__main__.py", line 9, in <module>
  File "c:\python36\lib\site-packages\kicost\__main__.py", line 223, in main
    kicost_gui() # Use the user guide if no input is given.
  File "c:\python36\lib\site-packages\kicost\kicost_gui.py", line 1024, in kicost_gui
    frame = formKiCost(None)
  File "c:\python36\lib\site-packages\kicost\kicost_gui.py", line 518, in __init__
    self.set_properties()
  File "c:\python36\lib\site-packages\kicost\kicost_gui.py", line 836, in set_properties
    s=numbers.get_currency_symbol(currency),
  File "c:\python36\lib\site-packages\babel\numbers.py", line 143, in get_currency_symbol
    return Locale.parse(locale).currency_symbols.get(currency, currency)
AttributeError: 'NoneType' object has no attribute 'currency_symbols'

I already checked the local set up and it's English (UK). 

Then I tried to use C:\Users\ezzsl3>kicost -i "C:\Users\ezzsl3\Desktop\Matrix converter\Matrix converter\Matrix converter.xml"
Progress:   0%|                                                                               | 0/49 [00:00<?, ?part/s]Traceback (most recent call last):
  File "c:\python36\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\python36\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Python36\Scripts\kicost.exe\__main__.py", line 9, in <module>
  File "c:\python36\lib\site-packages\kicost\__main__.py", line 267, in main
    apiKeys=None)
  File "c:\python36\lib\site-packages\kicost\kicost.py", line 226, in kicost
    dist_octopart.query_part_info(parts, distributor_dict, currency, apiKeys.get('octopart'))
  File "c:\python36\lib\site-packages\kicost\distributors\dist_octopart.py", line 394, in query_part_info
    get_part_info(octopart_query, parts)
  File "c:\python36\lib\site-packages\kicost\distributors\dist_octopart.py", line 270, in get_part_info
    results = dist_octopart.query(query, apiKey)
  File "c:\python36\lib\site-packages\kicost\distributors\dist_octopart.py", line 174, in query
    raise Exception('Octopart KEY invalid, registre one at "https://www.octopart.com".')
Exception: Octopart KEY invalid, registre one at "https://www.octopart.com".

It gave me this error.

Can anyone help me?

hildogjr, 2019-01-06:
For the command line (also to the GUI), you have to register you own Octopart KEY, as the last line message. Also, replacing the files by the last version here will provide you better message.
More information in https://github.com/xesscorp/KiCost/issues/331

About the specific GUI problem here, could you provide the exactly configuration/file used?

tom88jerry, 2019-01-06:
@hildogjr 
Thank you for your answer.
I just applied Octopart KEY but I don't know how to use it. I followed the link that you gave me. It said patch KiCost that replace hardcoded apikey to my apikey in "def query(query):" at "kicost/distributors/dist_octopart.py". it works correctly. I couldn't find the directory of my Kicad.

In my C:\Python36\Scripts only has Kicost application.

I couldn't install using both easy_install kicost and pip install kicad so I googled and tried this python -m pip install kicost. It works perfectly. 

About Gui problem. I install python 3.6 and installed using above method. When I type kicost it showed that error.

hildogjr, 2019-01-06:
GUI: maybe you could try update the package `babel` used...

KEY: There is a place in the configuration tab of the GUI or add `apiKeys={'octopart':'YOUR_KEY_HERE_IN_QUOTE'}` after the line 224 of the `kicost.py` file.

We still have to fix some issues and remove workarounds (this included)...

tom88jerry, 2019-01-07:
@hildogjr How do i access the kicost.py? I couldn't find the directory to this.

hildogjr, 2019-01-07:
Judging by your Windows log:
     
      c:\python36\lib\site-packages\kicost\kicost.py

To know, if you are not a python programmer. The line that I said have to be vertical aligned with the others. E.g.:

    # Get the distributor pricing/qty/etc for each part.
    apiKeys={'octopart':'YOUR_KEY_HERE_IN_QUOTE'}
    if dist_list:
        #distributor.get_dist_parts_info(parts, distributor_dict, dist_list, currency)
        #TODO The calls bellow should became the call above of just one function in the `distributors` pachage/folder.
        dist_local_template.query_part_info(parts, distributor_dict, currency)
        if apiKeys and apiKeys.get('octopart'):
            dist_octopart.query_part_info(parts, distributor_dict, currency, apiKeys.get('octopart'))
        else:
            logger.warning("Octopart KEY not informed.")

hildogjr, 2019-01-07:
Other warning, this is your personal KEY, do not post it here (we had a issue about) and if it stop to work you have to mail to Octopart, as in the other thread (https://github.com/xesscorp/KiCost/issues/331).

tom88jerry, 2019-01-07:
Now I received this error.

C:\WINDOWS\system32>kicost -i "C:\Users\ezzsl3\Desktop\Matrix converter\Matrix converter\Matrix converter.xml"
Progress:  39%|███████████████████████████▏                                          | 19/49 [00:01<00:02, 12.14part/s]Traceback (most recent call last):
  File "C:\Python36\Scripts\kicost-script.py", line 11, in <module>
    load_entry_point('kicost==1.0.4', 'console_scripts', 'kicost')()
  File "C:\Python36\lib\site-packages\kicost-1.0.4-py3.6.egg\kicost\__main__.py", line 267, in main
    apiKeys=None)
  File "C:\Python36\lib\site-packages\kicost-1.0.4-py3.6.egg\kicost\kicost.py", line 226, in kicost
    dist_octopart.query_part_info(parts, distributor_dict, currency, apiKeys.get('octopart'))
  File "C:\Python36\lib\site-packages\kicost-1.0.4-py3.6.egg\kicost\distributors\dist_octopart.py", line 394, in query_part_info
    get_part_info(octopart_query, parts)
  File "C:\Python36\lib\site-packages\kicost-1.0.4-py3.6.egg\kicost\distributors\dist_octopart.py", line 270, in get_part_info
    results = dist_octopart.query(query, apiKey)
  File "C:\Python36\lib\site-packages\kicost-1.0.4-py3.6.egg\kicost\distributors\dist_octopart.py", line 176, in query
    raise Exception('Octopart error.')
Exception: Octopart error.

hildogjr, 2019-01-07:
This error could be a bad type in the manf# codes (or even a not recognized character, I though that I had it fix). If possible, attach your XML file.

tom88jerry, 2019-01-07:
@hildogjr 
I used bom2grouped_csv plugin in kicad to generate the bom file.
Here is the link to bom file https://ufile.io/dqgml

tom88jerry, 2019-01-10:
Sorry, I set it to be available for just 1 day. I attached the file. 
[Matrix converter.zip](https://github.com/xesscorp/KiCost/files/2745369/Matrix.converter.zip)

hildogjr, 2019-01-10:
@tom88jerry , it's working here on Linux, both GUI and CLI (terminal), with minor issues that I am without time to fix them.
Check the file [Matrix converter.xlsx](https://github.com/xesscorp/KiCost/files/2747353/Matrix.converter.xlsx). I had to remove the distributors link because they keep my personal KEY and can generate problems with Octopart company.

Check this file `kicost.py` attached [kicost.py.zip](https://github.com/xesscorp/KiCost/files/2747355/kicost.py.zip). Look for `YOUR_KEY_HERE_IN_QUOTE` and replace by your registered key between `'`.
Check in Octopart.com if your key remains active, if don't you have to contact them to active.

Please reopen this thread if the problem remains.

closed 2019-01-10 by hildogjr