hildogjr/KiCost #346: Switch KiCost to using Partinfo API

kasbah, 2019-02-22:
As I mentioned [on the forum](https://forum.kicad.info/t/kicost-going-quiet/15000/10) I'd be happy for KiCost users to make use of the Partinfo API. 

> I am happy for people working on BOM tools, especially KiCost users and developers, to use (but not abuse) the Partinfo endpoint right now, specifically: https://dev-partinfo.kitspace.org/graphql. 

>It's currently an open endpoint so anyone can access it. Docs are a bit thin but you are able to browse the schema using the "Documentation Explorer" on the right on the above URL and you can jump on [our chat](https://riot.im/app/#/room/#kitspace:matrix.org) or the GitHub issue tracker if you have any questions. You don't need any special GraphQL client to use it, here is the example query using curl:

```
curl  -H 'Content-Type: application/json' -X POST -d '{"query": "{ part(mpn: {part: \"NE555P\", manufacturer: \"Texas Instruments\"}) { datasheet description type offers { sku { vendor part } prices { USD EUR GBP SGD } } } }"}' https://dev-partinfo.kitspace.org/graphql | python -m json.tool
```

To cover the API costs when Octopart does ask for payment I suggest we pool money in [opencollective.com/kitspace](https://opencollective.com/kitspace). I estimate that $100 a month will be enough to cover current KiCost users (but we will see of course).

hildogjr, 2019-03-16:
@kasbah, the workaround using your proxy is finished (and other improvements on the spreadsheet). I am waiting @devbisme / @xesscorp for checking.

I didn't undestand how to use your API, could you explain with some Python / address that I could direct pass to my browner ?

kasbah, 2019-03-16:
So the API uses GraphQL. There are some [client libraries available for Python](https://graphql.org/code/#python-1) that may make things easier but the basic principle is that you make a `query` defining the data you want and making use of variables that you define separately in `variables`. 

Here is an example using `requests` that gets back all the available info about NE555P. Below you could define the `input` variable to another part you want information about and also reduce the `query` to just the information you want. 

```python
import requests

query = """
    query ($input: MpnInput!) {
      part(mpn: $input) {
        mpn {
          manufacturer
          part
        }
        type
        datasheet
        description
        image {
          url
          credit_string
          credit_url
        }
        specs {
          key
          name
          value
        }
        offers {
          sku {
            vendor
            part
          }
          description
          moq
          in_stock_quantity
          stock_location
          image {
            url
            credit_string
            credit_url
          }
          specs {
            key
            name
            value
          }
          prices {
            GBP
            EUR
            USD
          }
        }
      }
    }
"""

variables = '{"input": {"part": "NE555P", "manufacturer": "Texas Instruments"}}'

r = requests.post("https://dev-partinfo.kitspace.org/graphql", {"query": query, "variables": variables})

print(r.json())

```

Here is the same [example in the interactive GraphiQL interfaces](https://dev-partinfo.kitspace.org/graphql?query=query%20(%24input%3A%20MpnInput!)%20{%0A%20%20part(mpn%3A%20%24input)%20{%0A%20%20%20%20mpn%20{%0A%20%20%20%20%20%20manufacturer%0A%20%20%20%20%20%20part%0A%20%20%20%20}%0A%20%20%20%20type%0A%20%20%20%20datasheet%0A%20%20%20%20description%0A%20%20%20%20image%20{%0A%20%20%20%20%20%20url%0A%20%20%20%20%20%20credit_string%0A%20%20%20%20%20%20credit_url%0A%20%20%20%20}%0A%20%20%20%20specs%20{%0A%20%20%20%20%20%20key%0A%20%20%20%20%20%20name%0A%20%20%20%20%20%20value%0A%20%20%20%20}%0A%20%20%20%20offers%20{%0A%20%20%20%20%20%20sku%20{%0A%20%20%20%20%20%20%20%20vendor%0A%20%20%20%20%20%20%20%20part%0A%20%20%20%20%20%20}%0A%20%20%20%20%20%20description%0A%20%20%20%20%20%20moq%0A%20%20%20%20%20%20in_stock_quantity%0A%20%20%20%20%20%20stock_location%0A%20%20%20%20%20%20image%20{%0A%20%20%20%20%20%20%20%20url%0A%20%20%20%20%20%20%20%20credit_string%0A%20%20%20%20%20%20%20%20credit_url%0A%20%20%20%20%20%20}%0A%20%20%20%20%20%20specs%20{%0A%20%20%20%20%20%20%20%20key%0A%20%20%20%20%20%20%20%20name%0A%20%20%20%20%20%20%20%20value%0A%20%20%20%20%20%20}%0A%20%20%20%20%20%20prices%20{%0A%20%20%20%20%20%20%20%20GBP%0A%20%20%20%20%20%20%20%20EUR%0A%20%20%20%20%20%20%20%20USD%0A%20%20%20%20%20%20}%0A%20%20%20%20}%0A%20%20}%0A}%0A&variables={%22input%22%3A%20{%22part%22%3A%20%22NE555P%22%2C%20%22manufacturer%22%3A%20%22Texas%20Instruments%22}}) where it may be easier to play around a bit.

hildogjr, 2019-03-18:
Interesting, start to understanding. Two comments:
1. How can I request more than one part? Or have I to request one-by-one?
2. How to just specify the mpn code and not the Manufacture name? (I tried `variables = '{"input": {"part": "NE555P"}}'`
3. The datasheet result was `'datasheet': 'https://????.com/click/track?ak=????&ct=datasheets&sig=????&at=physicalpart&sid=????&ppid=?????&hlid=??????????'`. Please, resolve this link (and others) before pass through the API.

kasbah, 2019-03-19:
Thanks for looking into it.

1. There is also a [`match` query](https://github.com/kitspace/partinfo/blob/1700dbce57ff3ac04cbc431a0b23054d36429891/src/schema.js#L26) where you enter multiple `parts` (and a `search` query for text input)
2. You currently have to use an empty string `{"part": "NE555P", "manufacturer": ""}`, we should fix that
3. > The datasheet result 

We can work on that I guess, is it important?

4.
> specify the vendors
 Makes sense, that requires a bit of work on the Partinfo site. We are currently restricted to the 5 vendors supported by Kitspace. 

5. moq stands for minimum-order-quantity,  the smallest quantity you can buy of a particular part

FYI Partinfo also doesn't currently have pagination thus max search results returned are currently 20, is that a problem?

hildogjr, 2019-03-19:
1. I still need to learn how to use, I am planning it to future #17;
2. Yes, maybe passing just `{"part": XXX} `, Kitspace could understand a empty `manufacturer` string;
3. May be just my opinion. But I would like to remove each trace of dependence in the KiCost;
4. Witch vendors are now supports? (KiCost was using Digikey, Mouser, Farnell, Newark, RS, TME and Arrow. It is interesting provide these at next KiCost update);
5. Interesting information! I am planning to use.

It is not a problem just return 20 by time. KiCost just have to ask twice...

I am a quite of time now, I already update something at `spreadsheet.py` but the big works still need to be done at `distributors` folder (using `dist_octopart.py`as template).

hildogjr, 2019-03-24:
@xesscorp / @devbisme I just finished some improvements at `spreadsheet.py` on the branch Octopart.
Now this branch is froze and we can move into Partinfo API on the main branch.

hildogjr, 2019-03-25:
@kasbah, I am having problem if the API Mouser response (testing in the query system yet) for the part
IPW60R160P6FKSA1
CGA9N2X7R2A105M230KA

So far I checked, it appear be related to the price break, even on the web page those appear different. But I think is not your code layer.

kasbah, 2019-03-26:
Sorry, what is the problem exactly?

hildogjr, 2019-03-26:
For example IPW60R160P6FKSA1 returned as "0 in stock".
But the web page says the opposite
https://www.mouser.it/ProductDetail/Infineon-Technologies/IPW60R160P6FKSA1?qs=sGAEpiMZZMshyDBzk1%2FWizrSbRfT%2Fm2Uc4Oi9zmiFXNiGpJHGELpdQ==

kasbah, 2019-03-26:
Just checked, that's down to Octopart unfortunately, they have the wrong stock information for that part at the moment. 

> It is not a problem just return 20 by time. KiCost just have to ask twice...

FYI, there is no way to get more than 20 results even if you ask twice.

hildogjr, 2019-03-26:
I imagined that. So don't worry, nothing we can do for this.

> There is no way to get more than 20 results even if you ask twice.
If I have a list of 35 items, I can't request 20, wait and request again with the other 15? KiCost do this with Octopart.

kasbah, 2019-03-26:
In the future we can figure out an additional data source for Mouser. Partinfo does support the Farnell API as well, and could improves the stock information for that, but the API limit prevents me from turning it on at the moment. Hopefully I will be able to fix that soon.

No, currently it's a hard limit of 20 items for the `search` and `match` queries. `part` queries only return a single result anyway.

hildogjr, 2019-03-30:
@kasbah, for use the match query I:

1) Changed the query to:

    query ($input: MpnOrSku!) {
        match(parts: $input) {

2) The variable to:

    {
       "input": {
                        "mpn": "RNCP1206FTD33R2",
                        "sku": ""
                    }
     }

What I am doing wrong?

And 2 question:

1) How to request more than one match?
2) The `search` query? Is it possible to request more than one search by request? Could you provide me some example of this two?

kasbah, 2019-03-30:
You need a list with the proper structure as input. [Here you go](https://dev-partinfo.kitspace.org/graphql?query=query%20(%24input%3A%20[MpnOrSku]!){%0A%20%20match(parts%3A%20%24input)%20{%0A%20%20%20%20description%0A%20%20}%0A}&variables={%0A%22input%22%3A%20[{%22mpn%22%3A%20{%22part%22%3A%20%22RNCP1206FTD33R2%22%2C%20%22manufacturer%22%3A%20%22%22}}]%0A}). I don't think there is a limit on the size of the `match` list at the moment. 

`search` is currently one per request. [Here is an example search query](https://dev-partinfo.kitspace.org/graphql?query=query%20(%24input%3A%20String!)%7B%0A%20%20search(term%3A%20%24input)%20%7B%0A%20%20%20%20description%0A%20%20%7D%0A%7D&variables=%7B%0A%20%20%22input%22%3A%20%22resistor%201k%22%0A%7D).

hildogjr, 2019-05-31:
@kasbah , please gimme a tip / have a look at line 193 of `api_partinfo_kitspace.py`.

I have the query type

    query ($input: [MpnOrSku]!){ match(parts: $input) {mpn{manufacturer,part},type,datasheet,description,image{url,credit_string,credit_url},specs{key,name,value},offers{sku{vendor,part},description,moq,in_stock_quantity,stock_location,image{url,credit_string,credit_url},specs{key,name,value},prices{GBP,EUR,USD}}} }

the components query

    { "input": [{"mpn":{"part":"RMCF0805JT10K0","manufacturer":""}},{"mpn":{"part":"1375820-3","manufacturer":""}},{"mpn":{"part":"1N4728ATR","manufacturer":""}},{"mpn":{"part":"ECE-A1VKA100B","manufacturer":""}},{"mpn":{"part":"C0805C105K5RACTU","manufacturer":""}},{"mpn":{"part":"RMCF0805JT150R","manufacturer":""}},{"mpn":{"part":"MOC3021S-TA1","manufacturer":""}},{"mpn":{"part":"RMCF0805JT4K70","manufacturer":""}},{"mpn":{"part":"640456-4","manufacturer":""}},{"mpn":{"part":"ECQ-E4225KF","manufacturer":""}},{"mpn":{"part":"C0805C224K5RACTU","manufacturer":""}},{"mpn":{"part":"RMCF0805JT330R","manufacturer":""}},{"mpn":{"part":"MC7805CDTG","manufacturer":""}},{"mpn":{"part":"ISO1042BDWR","manufacturer":""}},{"mpn":{"part":"ISO1042BDWR","manufacturer":""}},{"mpn":{"part":"BC847BLT1G","manufacturer":""}},{"mpn":{"part":"LL4148","manufacturer":""}},{"mpn":{"part":"RMCF0805JT120R","manufacturer":""}},{"mpn":{"part":"BTA204-600E","manufacturer":""}},{"mpn":{"part":"RMCF0805JT470R","manufacturer":""}}]}

That works at your [GraphiQL](https://dev-partinfo.kitspace.org/graphql?query=query%20(%24input%3A%20%5BMpnOrSku%5D!)%7B%20match(parts%3A%20%24input)%20%7Bmpn%7Bmanufacturer%2Cpart%7D%2Ctype%2Cdatasheet%2Cdescription%2Cimage%7Burl%2Ccredit_string%2Ccredit_url%7D%2Cspecs%7Bkey%2Cname%2Cvalue%7D%2Coffers%7Bsku%7Bvendor%2Cpart%7D%2Cdescription%2Cmoq%2Cin_stock_quantity%2Cstock_location%2Cimage%7Burl%2Ccredit_string%2Ccredit_url%7D%2Cspecs%7Bkey%2Cname%2Cvalue%7D%2Cprices%7BGBP%2CEUR%2CUSD%7D%7D%7D%20%7D%0A&variables=%7B%0A%20%20%22input%22%3A%20%5B%7B%22mpn%22%3A%7B%22part%22%3A%22RMCF0805JT10K0%22%2C%22manufacturer%22%3A%22%22%7D%7D%2C%7B%22mpn%22%3A%7B%22part%22%3A%221375820-3%22%2C%22manufacturer%22%3A%22%22%7D%7D%2C%7B%22mpn%22%3A%7B%22part%22%3A%221N4728ATR%22%2C%22manufacturer%22%3A%22%22%7D%7D%2C%7B%22mpn%22%3A%7B%22part%22%3A%22ECE-A1VKA100B%22%2C%22manufacturer%22%3A%22%22%7D%7D%2C%7B%22mpn%22%3A%7B%22part%22%3A%22C0805C105K5RACTU%22%2C%22manufacturer%22%3A%22%22%7D%7D%2C%7B%22mpn%22%3A%7B%22part%22%3A%22RMCF0805JT150R%22%2C%22manufacturer%22%3A%22%22%7D%7D%2C%7B%22mpn%22%3A%7B%22part%22%3A%22MOC3021S-TA1%22%2C%22manufacturer%22%3A%22%22%7D%7D%2C%7B%22mpn%22%3A%7B%22part%22%3A%22RMCF0805JT4K70%22%2C%22manufacturer%22%3A%22%22%7D%7D%2C%7B%22mpn%22%3A%7B%22part%22%3A%22640456-4%22%2C%22manufacturer%22%3A%22%22%7D%7D%2C%7B%22mpn%22%3A%7B%22part%22%3A%22ECQ-E4225KF%22%2C%22manufacturer%22%3A%22%22%7D%7D%2C%7B%22mpn%22%3A%7B%22part%22%3A%22C0805C224K5RACTU%22%2C%22manufacturer%22%3A%22%22%7D%7D%2C%7B%22mpn%22%3A%7B%22part%22%3A%22RMCF0805JT330R%22%2C%22manufacturer%22%3A%22%22%7D%7D%2C%7B%22mpn%22%3A%7B%22part%22%3A%22MC7805CDTG%22%2C%22manufacturer%22%3A%22%22%7D%7D%2C%7B%22mpn%22%3A%7B%22part%22%3A%22ISO1042BDWR%22%2C%22manufacturer%22%3A%22%22%7D%7D%2C%7B%22mpn%22%3A%7B%22part%22%3A%22ISO1042BDWR%22%2C%22manufacturer%22%3A%22%22%7D%7D%2C%7B%22mpn%22%3A%7B%22part%22%3A%22BC847BLT1G%22%2C%22manufacturer%22%3A%22%22%7D%7D%2C%7B%22mpn%22%3A%7B%22part%22%3A%22LL4148%22%2C%22manufacturer%22%3A%22%22%7D%7D%2C%7B%22mpn%22%3A%7B%22part%22%3A%22RMCF0805JT120R%22%2C%22manufacturer%22%3A%22%22%7D%7D%2C%7B%22mpn%22%3A%7B%22part%22%3A%22BTA204-600E%22%2C%22manufacturer%22%3A%22%22%7D%7D%2C%7B%22mpn%22%3A%7B%22part%22%3A%22RMCF0805JT470R%22%2C%22manufacturer%22%3A%22%22%7D%7D%5D%7D%0A) platform. But I am getting response 400 (Bad request) from my Python code.

hildogjr, 2019-05-31:
Forgot @kasbah, it was just missing `"` in `"input"`.

hildogjr, 2019-06-01:
@kasbah, Are the server offline? I am getting "504 Gateway Timeout" (also with https://dev-partinfo.kitspace.org/graphql)

kasbah, 2019-06-01:
Yes, there was an issue. Should be fixed now.

hildogjr, 2019-06-02:
Some alpha code is ready but still need some fix to the cases the PartInfo return no available data for one specific part in one distributor.

kasbah, 2019-06-02:
Can you give an example?

hildogjr, 2019-06-02:
There are a bunch of in https://github.com/xesscorp/KiCost/tree/master/tests.
I am testing with my personal project on run (so I have in mind the components and possible problems).
For now, I am not getting the distributor part page link.

My QUERY configuration is

     QUERY_ANSWER = '''
         mpn{manufacturer, part},
         type,
         datasheet,
         description,
         image{url},
         specs{key, value},
         offers{
             sku {vendor, part},
             description,
             moq,
             in_stock_quantity,
             stock_location,
             image {url, credit_string, credit_url},
             specs {key, name, value},
             prices{''' + DEFAULT_CURRENCY + '''}
             }
'''

I will remove something from it, but I missing the distributor URL.

kasbah, 2019-06-02:
Sorry, I am still struggling to understand the problem you are describing. 

1. What are you doing?
2. What is the expected result?
3. What is the actual result?

hildogjr, 2019-06-02:
As far I tested (last commit right now), I got all information that KiCost use (and more that I intend do implement some validation) except the part part URL in the distributor web site.
So, I though that I forgot something the the query configuration to request this from your server.

kasbah, 2019-06-02:
You mean a link like https://uk.farnell.com/texas-instruments/ne555p/timer-single-precision-dip8-0/dp/9589899 ?

hildogjr, 2019-06-02:
Exactly!

hildogjr, 2019-06-02:
The branch is going to pre-beta status and related issues may appear. The base functionalities are all coded though the new API.

kasbah, 2019-06-02:
I think https://github.com/kitspace/partinfo/issues/7 is still a show stopper.

hildogjr, 2019-06-21:
To register and order the API uses:

- [Specific manufacture code](https://dev-partinfo.kitspace.org/graphql?query=query%20(%24input%3A%20MpnInput!)%20%7B%0A%20%20part(mpn%3A%20%24input)%20%7B%0A%20%20%20%20mpn%20%7B%0A%20%20%20%20%20%20manufacturer%0A%20%20%20%20%20%20part%0A%20%20%20%20%7D%0A%20%20%20%20type%0A%20%20%20%20datasheet%0A%20%20%20%20description%0A%20%20%20%20image%20%7B%0A%20%20%20%20%20%20url%0A%20%20%20%20%20%20credit_string%0A%20%20%20%20%20%20credit_url%0A%20%20%20%20%7D%0A%20%20%20%20specs%20%7B%0A%20%20%20%20%20%20key%0A%20%20%20%20%20%20name%0A%20%20%20%20%20%20value%0A%20%20%20%20%7D%0A%20%20%20%20offers%20%7B%0A%20%20%20%20%20%20sku%20%7B%0A%20%20%20%20%20%20%20%20vendor%0A%20%20%20%20%20%20%20%20part%0A%20%20%20%20%20%20%7D%0A%20%20%20%20%20%20description%0A%20%20%20%20%20%20moq%0A%20%20%20%20%20%20in_stock_quantity%0A%20%20%20%20%20%20stock_location%0A%20%20%20%20%20%20image%20%7B%0A%20%20%20%20%20%20%20%20url%0A%20%20%20%20%20%20%20%20credit_string%0A%20%20%20%20%20%20%20%20credit_url%0A%20%20%20%20%20%20%7D%0A%20%20%20%20%20%20specs%20%7B%0A%20%20%20%20%20%20%20%20key%20name%0A%20%20%20%20%20%20%20%20value%0A%20%20%20%20%20%20%7D%0A%20%20%20%20%20%20prices%20%7B%0A%20%20%20%20%20%20%20%20GBP%0A%20%20%20%20%20%20%20%20EUR%0A%20%20%20%20%20%20%20%20USD%0A%20%20%20%20%20%20%7D%0A%20%20%20%20%7D%0A%20%20%7D%0A%7D%0A&variables=%7B%0A%20%20%22input%22%3A%20%7B%0A%20%20%20%20%22part%22%3A%20%22NE555P%22%2C%0A%20%20%20%20%22manufacturer%22%3A%20%22ti%22%0A%20%20%7D%0A%7D)

- [Parts that match](https://dev-partinfo.kitspace.org/graphql?query=query%20(%24input%3A%20%5BMpnOrSku%5D!)%7B%0A%20%20match(parts%3A%20%24input)%20%7B%0A%20%20%20%20description%0A%20%20%20%20offers%20%7B%0A%20%20%20%20%20%20sku%20%7B%0A%20%20%20%20%20%20%20%20vendor%0A%20%20%20%20%20%20%20%20part%0A%20%20%20%20%20%20%7D%0A%20%20%20%20%7D%0A%20%20%7D%0A%7D&variables=%7B%0A%20%20%22input%22%3A%20%5B%0A%20%20%20%20%7B%0A%20%20%20%20%20%20%22mpn%22%3A%20%7B%0A%20%20%20%20%20%20%20%20%22part%22%3A%20%22RNCP1206FTD33R2%22%2C%0A%20%20%20%20%20%20%20%20%22manufacturer%22%3A%20%22%22%0A%20%20%20%20%20%20%7D%0A%20%20%20%20%7D%2C%0A%20%20%20%20%0A%20%20%20%20%7B%0A%20%20%20%20%20%20%22mpn%22%3A%20%7B%0A%20%20%20%20%20%20%20%20%22part%22%3A%20%22SI8621BB-B-IS%22%2C%0A%20%20%20%20%20%20%20%20%22manufacturer%22%3A%20%22%22%0A%20%20%20%20%20%20%7D%0A%20%20%20%20%7D%0A%20%20%5D%0A%7D)

- [Search part by description (it will be used at #17 )](https://dev-partinfo.kitspace.org/graphql?query=query%20(%24input%3A%20String!)%7B%0A%20%20search(term%3A%20%24input)%20%7B%0A%20%20%20%20description%0A%20%20%20%20offers%20%7B%0A%20%20%20%20%20%20sku%20%7B%0A%20%20%20%20%20%20%20%20vendor%0A%20%20%20%20%20%20%20%20part%0A%20%20%20%20%20%20%7D%0A%20%20%20%20%7D%0A%20%20%7D%0A%7D&variables=%7B%0A%20%20%22input%22%3A%20%22resistor%201k%200805%205%25%22%0A%7D)

kasbah, 2019-06-22:
Hey, Octopart has just turned of my API key and asked for payment. So far only one person is donating $5 to the Kitspace collective specifically for KiCost. I think we need to do a bit of a funding drive and find more donors otherwise this isn't going to work.

xesscorp, 2019-06-24:
I figured this would become a problem. I don't see a funding drive as being a solution. My old quote from Octopart was $100/month for 10,000 queries. So you would need $1200/year. I can't see the KiCad community donating that much consistently, year over year, for a single add-on tool. Even if they did, somebody could easily consume all the allowed queries and leave everyone else with nothing, so we'd need some way to allocate the queries to users. That involves accounts, privacy, security, etc. It's just a mess.

The only long-term solution (I think) is to use public APIs for places like Digi-Key and Mouser.

hildogjr, 2019-06-24:
Agree with @xesscorp, may be complicated achieve this.

I kept the `api_octopart.py` file fully functional in the current branch by this reason (as an academic guy, I have this kind of access and others user could prefer to pay Octopart due stability and queries that it may use).

My plan is, in the future, integrate different APIs / scrape module (stating with Digikey and Mouser) and allow the user to select which to use (in the GUI maybe, by on table, adding the personal token key or activating the correspondent modules). But I am stuck in this personal road map due my need help in GUI.

kasbah, 2019-06-24:
I agree that it's not a long term solution but I had hoped that there would be a bit more funding coming in from KiCost community for the short term. To be fair, I don't think anyone mentioned we were seeking donations besides me in some hidden away comments.

I should be able to get the Partinfo up again very soon. I am happy to use the money we do have in the Kitspace collective towards keeping it running, but we do need a bit more in the short term I think so any funding you can drive towards this would help a lot.

I think we can easily put some basic precautionary measures in place against abuse (requests per IP) and think about something more complicated if we run into problems. I will also improve the caching strategy in Partinfo and, as you know, keep adding additional sources (I now have a trial access from https://www.oemsecrets.com/api which seems to have more comprehensive pricing info actually).

xesscorp, 2019-06-24:
OK, once @hildogjr has a new release of KiCost on PyPI, then we can announce it has re-awakened and it is using Partinfo as its engine. Then we can request donations for Partinfo's upkeep. (If we get more than $100, I'd be *very* surprised.)

@hildogjr , with regards to creating new API interfaces and scraping modules for KiCost, I think your effort would be better spent doing that inside the Partinfo framework. That way, the new interfaces will be available to any and all tools that use Partinfo. That may lead to more maintainers and quicker bug fixes, particularly for any web scrapers.

hildogjr, 2019-06-26:
I almost solve #353, by using `kicost --setup`.

[image]

kasbah, 2019-06-26:
Service is up again just now!

hildogjr, 2019-06-27:
I got passed in all test file but I got some empty prices. I am checking but appear to but old parts / not found by the services used in PartInfo.
@kasbah, please check the documentation of KiCost. At the main page, in the figure above and the GUI, I put a mention about. Is this correct?
@xesscorp, should be nice improve the video example #249.

I may release KiCost until the weekend.

kasbah, 2019-06-27:
Could you link to the text you mean? I am not sure what branch/file it is.

hildogjr, 2019-06-27:
Everything in the main branch:
The text that appear in Eeschema plugin list is on [`kicost.py`](https://github.com/xesscorp/KiCost/blob/master/kicost/kicost.py).
The [main page](https://github.com/xesscorp/KiCost/blob/master/README.md) and the GUI use the [image here](https://github.com/xesscorp/KiCost/blob/master/kicost/logo_kitspace.png), also in the [functional diagram](https://github.com/xesscorp/KiCost/blob/master/docs/block_diag.svg).

hildogjr, 2019-06-27:
@xesscorp additionally #249, as Windows user, could you test the feature `kicost --setup` and `kicost --unsetup` as in https://forum.kicad.info/t/bom-from-eschema/16536/45?

kasbah, 2019-06-27:
It's ok, there's a lack of borders on the top and bottom and it maybe it's a bit clearer if it mentions the API? How about using this and also mentioning we are [seeking donations](https://opencollective.com/kitspace)? 

[image]

hildogjr, 2019-06-27:
For now, clicking on Kitspace logo on the GUI the user is redirect to https://kitspace.org/ (we can change to https://opencollective.com/kitspace).
We could add some message there, in the GUI, and change the plugin description to something like:
"Nowadays, KiCost is powered by PartInfo API (https://kitspace.org/) kept by donations (https://opencollective.com/kitspace)."

Please, @kasbah. Provide a push changing the logo in both figures and also changing the description in `kicostp.py` to registry your participation.

kasbah, 2019-06-27:
No probs on the logo will make a pull-request. What do you mean by "description in kicostp.py"? (Please link to lines of code when you refer to them -- I  don't know my way around the project very well.)

FYI, unless we get some more donations and can afford a higher request limit I will have to limit the number of octopart requests to 1000 per month. 

Also, am I right in thinking you can not make a release until https://github.com/kitspace/partinfo/issues/7 is solved?

hildogjr, 2019-06-27:
https://github.com/xesscorp/KiCost/blob/master/kicost/kicost.py
Lines 26-38

@xesscorp, please contribute with this specific revision. Since it will become the official description of KiCost.

kasbah, 2019-06-28:
https://github.com/kitspace/partinfo/issues/7 isn't really about any bottle neck, it's more about supporting all the different distributors. Currently only 5 are supported.

xesscorp, 2019-06-28:
I have tested and made many fixes to kicost_config.py to make the --setup and --unsetup options work. The KiCost shortcut now works, but the context menu integration does not. Oh well.

hildogjr, 2019-06-28:
This command doesn't affect the KiCost motor at all, I just though it would be nice to call more users to the tool.
I use your (@xesscorp ) recommendation at https://xesscorp.github.io/KiCost to create the context integration. But have been some time that I do not use Windows.

I also would like some Mac OS user help on this.

hildogjr, 2019-06-30:
@xesscorp I update de way to generate the Windows context menu, please, try.
I am also trying to create the documentation by `make docs`  using `sphinx` and trying to fix #289. Could you help / gimme some directives?

xesscorp, 2019-07-01:
I made some small fixes to your calls to `ConnectRegistry()`. After I did that, `kicost --setup` and `kicost --unsetup` both fail with errors in `remove_bom_plugin_entry()` and `add_bom_plugin_entry()`. I looked at the code and I can see why it is failing (indexing the BOM plugins incorrectly), but I assume you tested this on linux and it worked correctly. I don't know why it would fail on Windows since it's not OS-specific.

I changed some of the documentation to make it compile. All you have to do is go into the docs directory and use the command `make singlehtml`.

hildogjr, 2019-07-01:
@xesscorp I tested in some Win virtual machine. I am having some regular expression issue due Windows use "\" and unix based use "/" to path.

hildogjr, 2019-07-05:
@xesscorp, I think that I have fixed the "add to BOM list" Windows issue. Missing the context menu.

xesscorp, 2019-07-07:
I downloaded the latest KiCost and ran it under Windows 7 with KiCad 5.1.2.

When I ran `kicost --setup`, I got this output:
```
KiCost identified at 'c:\xesscorp\kicad\tools\kicost\kicost', proceding with it configuration in file 'C:\Users\DEVB\AppData\Roaming\kicad'...
GUI requirements (wxPython) identified.
Creating app shortcuts...
Check your desktop for the KiCost shortcut.
Creating OS context integration...
GUI requirements (wxPython) identified.
Failed to create KiCost OS context menu integration.  <== ???
Setting the GUI to display the NEWS message...
The will display the NEWS message on first staturp.
Creating KiCad integration...
Failed to add KiCost as KiCad BOM plugin.  <== ???
KiCost setup configuration finished.
```
After that, I ran KiCad and EESCHEMA. KiCost was in the BOM list and it started running when I clicked on it. So I'm not sure if it actually failed to be added. I also saw KiCost in the context menu when I right-clicked on an .xml file.

It's possible KiCost had already been added before I ran the --setup, so I decided to run `kicost --unsetup` to remove them and got the following:
```
Removing BOM plugin entry from Eeschma configuration...
Traceback (most recent call last):
  File "C:\Python37-64\Scripts\kicost-script.py", line 11, in <module>
    load_entry_point('kicost', 'console_scripts', 'kicost')()
  File "c:\xesscorp\kicad\tools\kicost\kicost\__main__.py", line 187, in main
    kicost_unsetup()
  File "c:\xesscorp\kicad\tools\kicost\kicost\kicost_config.py", line 485, in kicost_unsetup
    remove_bom_plugin_entry(kicad_config_path, 'KiCost')
  File "c:\xesscorp\kicad\tools\kicost\kicost\kicost_config.py", line 167, in remove_bom_plugin_entry
    if re.findall(name, plugin[1], re_flags):
  File "c:\python37-64\lib\re.py", line 223, in findall
    return _compile(pattern, flags).findall(string)
TypeError: expected string or bytes-like object
```
If you can get the `--unsetup` to work, I will try again.

hildogjr, 2019-07-08:
@xesscorp, I had fix the Win issue related do path named at the `add_bom_plugin_entry` function and forgot at `remove_bom_plugin_entry`. Now I propagate the changes. Please, check again.

When you use the context-menu in a .xml file, did the GUI open with the specified file?

hildogjr, 2019-07-12:
@xesscorp, I create the documentation by your make file (I don't know how to publish it after). Please check it and the `--setup`/`--unsetup` commands, I will release on PyPI after.

xesscorp, 2019-07-12:
--unsetup still fails:
```
# kicost --unsetup
Removing BOM plugin entry from Eeschma configuration...
Traceback (most recent call last):
  File "C:\Python37-64\Scripts\kicost-script.py", line 11, in <module>
    load_entry_point('kicost', 'console_scripts', 'kicost')()
  File "c:\xesscorp\kicad\tools\kicost\kicost\__main__.py", line 187, in main
    kicost_unsetup()
  File "c:\xesscorp\kicad\tools\kicost\kicost\kicost_config.py", line 504, in kicost_unsetup
    remove_bom_plugin_entry(kicad_config_path, 'KiCost')
  File "c:\xesscorp\kicad\tools\kicost\kicost\kicost_config.py", line 171, in remove_bom_plugin_entry
    search = plugin[1].replace("\\",'/')
AttributeError: 'Symbol' object has no attribute 'replace'
```

hildogjr, 2019-07-15:
Fixed the issue and already in
https://pypi.org/project/kicost/

@xesscorp, release or gimme so guidance to release/update https://xesscorp.github.io/KiCost.

hildogjr, 2019-07-23:
@kasbah, KiCost with PartInfo API is released https://forum.kicad.info/t/new-kicost-release/8074/2
Please, keep us updated about the server use and some statistics (for example country of access, ...). If necessary, we can add a "&tool=kicost" in the request link to help you with the statistics.

kasbah, 2019-07-31:
FYI there have been over 700 octopart requests resulting from KiCost since I put the service up again.

xesscorp, 2019-07-31:
Over what time period?

hildogjr, 2019-07-31:
Do you have some statics of how much of them was new (not replicated or already cached by you)?
Because, the code take a full BOM and broke into maximum 25 parts request. Because I was testing, I may be responsible by some replicated access.

kasbah, 2019-07-31:
Since 27th of June. This only counts cache misses i.e. when it actually makes requests to Octopart. Oh, and it actually also includes people using https://bom-builder.kitspace.org.

It does seem like a lot. I should set up some more sensible logging.

[10 more replies]

commit 5e65c9f7 (2019-06-02): Fix #346
commit 73c169bb (2019-07-15): Fix #346 unsetup
commit 950bdb98 (2019-08-11): Fix #346
commit d3d19a7f (2019-08-11): Fix #346
closed 2019-08-11 by hildogjr