hildogjr/KiCost #350: Non-ASCII character '\xc3' in file ... but no encoding declared

mdeweerd, 2019-04-21:
SyntaxError: Non-ASCII character '\xc3' in file /usr/lib/python2.7/site-packages/kicost-2.0-py2.7.egg/kicost/__main__.py on line 3, but no encoding declared; see http://python.org/dev/peps/pep-0263/ for details

Solution:  Add a header line like 

`# -*- coding: utf-8 -*-`

hildogjr, 2019-04-23:
Thanks for the suggestion.
Are you able to still using KiCost at Python 2? In last code changes I was using my energy to the feature and less to the Py2 compatibility.

mdeweerd, 2019-04-25:
Apparently there were too many ongoing changes so I reverted to an earlier version

hildogjr, 2019-04-25:
The main branch here is in development and not functional yet.
The last, frozen, functional branch is Octopart.

hildogjr, 2019-05-09:
Fixed in main branch

closed 2019-05-09 by hildogjr