hildogjr/KiCost #351: KiCost 2.0 - Some syntax issues (Python 2.7)

mdeweerd, 2019-04-21:
### *sorted() calls:
\*sorted(...)  results in invalid syntax

My workaround: comment these lines

### Python3 print:
`print('{} = '.format(f), end=' ')` gives invalid syntax (line 231, kicost.py) on th e"=" symbol for end

My workaround: replaced with `print '{} = '.format(f),`

### urllib import issue
```
from urllib.parse import quote_plus as urlquote
ImportError: No module named parse
```
My solution: replace with
```
try:
    from urllib.parse import quote_plus as urlquote
except ImportError:
     from urllib import quote_plus as urlquote
```
### dist_local_template import issue
`from .dist_local_template import dist_local_templates`

My solution: replace with
`from .dist_local_template import dist_local_template`

### Import issue in GUI code
- kicost_gui.py
    `from .distributors import init_distributor_dic`

My solution: commented the import of the gui:
	   `from .kicost_gui import *` etc.
	   as I am not using the gui.

### Issue removing (unexisting) distributor:
`dist_list.remove(d)
ValueError: list.remove(x): x not in list`

I had an invalid distributor in the exception list.

My solution: I added a try: except: pass arround the remove.

### Invalid call
`  File "kicost.py", line 215, in kicost
    dist_local_template.query_part_info(parts, distributor_dict, currency)
TypeError: unbound method query_part_info() must be called with dist_local_template instance as first argument (got list instance instead)
`

No solution, stuck here.

hildogjr, 2019-04-23:
@mdeweerd, this fix is related with the [Octopart branch](https://github.com/xesscorp/KiCost/tree/octopart) that is been decrepited.
Now, we are moving (and needing help) to Partinfo API #346 to provide better API to user, not dependent to Octopart Key. For now, since Python2 is in retire countdown, I am not using time for Py2 compatibility.

mdeweerd, 2019-04-25:
Hi
I understand - I upgraded my KiCost installation as I wanted to check how evolutions were going, but then it was dysfunctionnal, even when disabling all distributors.
I am quite busy myself, I always hope that sharing difficulties and workaround is a help to the team.
I am not so much using pull requests, etc.
I tried to make some contributions in the past, but KiCost was evolving too fast.

hildogjr, 2019-04-25:
The focus of KiCost now is make it work with PartInfo API, not being dependent of Octopart KEY anymore.

As secondary issues, we have: improve the GUI, improve the installation to create a program icon and KiCad plugin at Pcbnew.

mdeweerd, 2019-04-26:
With regards to the code: I know how to pull/push - as said KiCost evolves too much fort that.  I had quite some changes in a fork I have now deleted that became incompatible by the time it was ready for sharing.

Some brainstorming suggestion for the logo.

[image]

I'll think about documentation/plugin - share my method of using KiCost.

hildogjr, 2019-04-28:
Unfortunately the first step of KiCost (get the information through API / scrape) is being re-write.
The spreadsheet have few improvements: fix Digikey "txt importer" and its was add some checks to avoid misunderstood in the purchase act.

The icon:
I liked it, main for a logo. Maybe I give some more attention (uppercase) to the "C / €".
The icon should fit as shortcut in the desktop / start menu and the program window. For now, we I are using https://github.com/xesscorp/KiCost/blob/master/kicost/kicost.ico
I need a 16x16 icon version for use as KiCad menu plugin. I was thinking: some coin with "Ki" inside and the border/shadow making the "C".

mdeweerd, 2019-04-28:
Icons can be considered minimum 32x32 nowadays.
It could look like a coin rather than a bill.
Here is a another brainstorm suggestion - the SVG is in the zip.
[image]
[KicostLogoSuggestion.zip](https://github.com/xesscorp/KiCost/files/3125561/KicostLogoSuggestion.zip)

hildogjr, 2019-04-28:
The KiCad documentation recommends a 26x26px:
http://docs.kicad-pcb.org/doxygen/md_Documentation_development_pcbnew-plugins.html

I was think some compact for the coin icon:
[image]
also that the border could remember a coin, in perspective.
[image]

hildogjr, 2019-04-28:
Actual icon used:
[image]

mdeweerd, 2019-04-29:
I like the dime in perspective.  The "K$" should be a bit bigger on the coin so that it is more readable at 24x24 - similar to the flat K$.

hildogjr, 2019-04-29:
@mdeweerd, totally agree. The perspective dime cold be used to derivate a shadow "=", making an ambiguous "C" / "€" symbol and over the "K$".
(I am not good with graphics, so mine is just a draft).
The KiCost logo proposed (black letters with currency symbols) will be not visible at black/dark desktops or KiCad theme (I think this is in them roadmap). The green bill solve this problem, my propose is merge these ideas, with you agree.

mdeweerd, 2019-04-29:
I think that the two horizontal bars are a good suggestion of the other currency symbols - there is no need to overload the logo/icon.
I only now understand the suggestion of the € in the flat logo - I think this can be reminded in the same way by having a darker border/background to represent the € symbol.

hildogjr, 2019-05-09:
@mdeweerd, could provide this 2 icons version? The coin (`kicost_coin.svg`/png) and the bank note (`kicost.svg`).

mdeweerd, 2019-05-09:
HI
I saw that you assigned me this and I will do something, but I have some other priorities.
Please be patient ;-).

hildogjr, 2019-05-09:
No problem.
I am concerned now to make the new API before release and found someone (in KiCad forum) to help with the Windows / GUI issues.

mdeweerd, 2019-08-27:
Hi

I made some time to work on the logo.

These are the 24x24 and 512x512 versions.
I think that the 24x24 version is acceptable.

[image]
[image]

This is the SVG in a ZIP file:
[Kicost_Dime.zip](https://github.com/xesscorp/KiCost/files/3546701/Kicost_Dime.zip)

mdeweerd, 2023-04-27:
Almost 4 years - closing.

closed 2023-04-27 by mdeweerd