hildogjr/KiCost #357: No information about current state of KiCost

rheasman, 2019-05-23:
Hi,

I really can't tell by looking at this repo, or at documentation, what the current state of KiCost is.

I have used it extensively in the past, for multiple boards, and now I need it again to update those boards.

I installed it using pip3 and got something that failed because I have no Octopart API Key, so I went to Octopart but it looks like they are issuing keys to developers of apps, and not end users. Is this true?

On top of that, it looks like KiCost is moving away from Octopart, judging from conversations in the issue tracker? It's hard to tell.

On top of that, I don't think scraping works any more?

If I am:

1. Willing to install any version of KiCost necessary.
2. Willing to pay someone money (Octopart, whoever, I don't care).

Is it possible to get a working version of KiCost? I absolutely love its output, and I have 5 boards where manf# etc are already filled in, and now I need someway to get usable BOMs.

If it is possible, could you how to make KiCost work to the readme?

Thanks.

xesscorp, 2019-05-24:
Hi. I'm glad KiCost helped you in the past. I wish it could help you now. But that probably won't happen.

We moved KiCost to Octopart because I had to continually change the scraping code to account for the changes in the web page formats used by each distributor (in addition, Mouser started using Distil which completely thwarted any scraping).

After three months, Octopart revoked my free license and wouldn't issue any licenses to anybody to use KiCost. They also stopped issuing free licenses to anybody but students. Octopart did offer to sell me a $1200 yearly license limited to 10,000 HTTP accesses per month, which I declined.

Now the plan is to move KiCost to using the Partinfo interface at kitspace.org. Partinfo will probably use some mixture of web scraping and distributor's public APIs to get pricing information (as well as a lot of other stuff). Up to now, I haven't been involved in this very much but I don't think it's functional.

So your current options are:

1. Install KiCost 0.1.47 and see if the web scraping will pull any data from any distributor. That version is from August of 2018, so all the distributor web pages have probably mutated enough that this will not work.
2. Get a recent version of KiCost with the Octopart interface and either try to buy a license key from them or find a university student who can request a license for you.

This all assumes you need *pricing* information. If all you need is a BOM, it would seem your old KiCost spreadsheets would still have everything you needed except current prices. And there are plenty of BOM tools for KiCad now.

hildogjr, 2019-05-24:
I'm more involved in the Partinfo migration (into the KiCost main branch). It is not operational now because is missing finish the `api_partinfo_kitspace.py` code (all changes necessary in the other files are finished, I think).
The current status is available in the milestone status
https://github.com/xesscorp/KiCost/milestone/1

The issues that need more help now is Windows issue #339 (more information in https://forum.kicad.info/t/bom-from-eschema/16536/37). This have been difficult to solve without help of Windows user / programmers.

kasbah, 2019-05-24:
In the meantime, why not release a version of KiCost using our temporary proxy?

hildogjr, 2019-05-24:
The Octopart branch tree here have this workaround, if not detected any user KEY (and other small fixes).
Check line 160 in https://github.com/xesscorp/KiCost/blob/octopart/kicost/distributors/dist_octopart.py

I am just corned about release this, the implication about Octopart use term and that the users will stay inerts about the necessity of KiCost changes.

kasbah, 2019-05-24:
I would say go for the release. I don't think it's violating any terms of use. 

If Octpart ask Kitspace for money we can use the money from our Opencollective: https://opencollective.com/kitspace (please also donate if you can).

hildogjr, 2019-05-24:
I will wait for @devbisme (Author) decision.
Nevertheless #339 is the worst error and have to be fixed before the release (I need some Windows user to help fix).

rheasman, 2019-05-24:
Thank you for all the immediate and helpful replies. 

I'll make do for now, but know that I'll be back as soon as you have something figured out. :-)

w6ipa, 2019-05-24:
I can confirm that Octopart will not issue a license to users without a license fee of a minimum of $25 per month.

kasbah, 2019-05-24:
Yes, but if we pool the money through the Kitspace collective we can make it work.

hildogjr, 2019-06-02:
For report here, @rheasman KiCost have the new API implemented.
But still issues for release official (https://github.com/xesscorp/KiCost/milestone/1) and we wait for users to helps us to fix.

closed 2019-06-02 by hildogjr