hildogjr/KiCost #358: Keep DNP/NOPOP in output with Qty=0 and column DNP

mdeweerd, 2019-05-24:
To make sure that no component was forgotten during the analysis, it is common practice to also list the DNP/NOPOP components in the BOM file on separate lines with quantity 0 and a DNP/NOPOP mention somewhere.

So the feature would be to group all DNP/NOPOP components at the end of the list - with Qty 0 and possibly with the pricing info (not required).
Qty 0 is already an indication that the component is not mounted, but an extra DNP column may be usefull for some.

hildogjr, 2019-05-24:
First study:
This should be activated by a `--show_dnp` and may be required some changes in the group routines.

hildogjr, 2020-05-21:
Needed #359 fix for not workaround.

mdeweerd, 2023-04-27:
Stale, closing (limiting my issue backlog).

closed 2023-04-27 by mdeweerd