hildogjr/KiCost #362: Fails to parse spaces in .xml name using %I wildcard in BOM dialogue

Gasman2014, 2019-06-05:
Using an xml file containing spaces in the name and the following dialogue in the KiCad BOM generator dialogue
`kicost -i %I`
results in the BOM name being split at the 1st word.

eg `1GHz probe.xml` will throw an error `FileNotFoundError: [Errno 2] No such file or directory: ‘/***/***/1GHz.xml’` and, if the xml is renamed to 1GHz.xml, will the throw an error `FileNotFoundError: [Errno 2] No such file or directory: ‘probe.xml’`.

- Use a simple and clear title.
- Does not mak as "ISSUE", "FEATURE" on title, this will be taged by the programming group.

#### Issue / Problem report
Add:
1. Terminal error report / txt;
2. Use ``--debug`` option on KiCost call;

    kicost -w -i %YOUR_FILE% --degub 8
    kicost -w -i %YOUR_FILE% --degub 8 -s # For serial scrape, which give us a better debug in case of distributors scrape fails.

3. Write the command used to call KiCost (or the graphical interface configuration);
4. One BoM to reproduce the error (with the EDA version).

#### Doubts

#### Feature / Enhancement request
- Explain it will help the other users.

The team thanks your contribution.

hildogjr, 2019-06-05:
@Gasman2014, which system are you using? Your report is quite confusion and haven't any report log file as said.
Try use `kicost -i "%I"`, I think both Windows and Linux (I use like this) need this to scape the space in the command line interpretation.
You also can use (with your system have wxPython library) `kicost --guide "%I"` to open a KiCost dialog to choose option such distributors to use, currency and do on.

Gasman2014, 2019-06-05:
Using this on macOS. Yes, if I use it on the command line I can get it to work if I escape the spaces. However the %I substitution does not correctly pass the spaces if i invoke kickstart from within the Kicad ‘generate Bill of Materials’ dialogue. Perhaps this is more an issue with Kicad than with kicost which is otherwise working fine. Please feel free to close.

hildogjr, 2019-06-05:
I think KiCad make a raw replace and execute `kicost -wi %I` like "I" being a variable into the terminal (on Linux `kicost -wi $I`, this line with the "%" appear like a Windows command). This justifies the occurred.

closed 2019-06-05 by hildogjr