hildogjr/KiCost #365: Using digikey# returns the part, but not the exact digikey number given

nickoe, 2019-07-11:
I had a part with the digikey# being 631-1072-1-ND. When I open the spreadsheet it shows 631-1072-**2**-ND. (note the 2, which is not what I specified.

[vhf_uhf_com.xlsx](https://github.com/xesscorp/KiCost/files/3384253/vhf_uhf_com.xlsx)
[vhf_uhf_com.txt](https://github.com/xesscorp/KiCost/files/3384258/vhf_uhf_com.txt)

Command used:
`kicost -i %I --overwrite --currency DKK --debug 2`

Using KiCad 5.1 branch, Version: (5.1.2-191-g5b68cfa01), release build 

The part in question if of course FT5HNBPK19.44-T1 from the bom.

BOM snippet
```
<comp ref="X1">                                                             
  <value>FT5HNBPK19.44-T1</value>                                                                            
  <datasheet>http://www.foxonline.com/pdfs/fox924.pdf</datasheet>           
  <fields>                                                                  
    <field name="digikey">631-1072-1-ND</field>                             
    <field name="digikey#">631-1072-1-ND</field>                            
    <field name="manf#">FT5HNBPK19.44-T1</field>                            
    <field name="todo">yes</field>                                          
  </fields>                                                                 
  <libsource lib="aausat" part="ASEM-xxxMHz" description="3.3V CMOS SMD Crystal Clock Oscillator, Abracon"/>
  <sheetpath names="/RX path/" tstamps="/5CA93143/"/>                       
  <tstamp>5CB0DC43</tstamp>                                                 
</comp>                                                                     
```

hildogjr, 2019-07-12:
@nickoe, KiCad version doesn't matter for KiCost.
Please, type `kicost --version` at your terminal and reply here.

hildogjr, 2019-07-12:
As far I tested, your part is:
manf# FT5HNBPK19.44-T1
digikey# 631-1072-1-ND

The proclaiming is that KiCost is returning 631-1072-2-ND. But my version returns 631-1072-1-ND, correctly.  Check that "both part" are the same for Digikey:
[631-1072-1-ND](https://www.digikey.com/products/en?keywords=%20631-1072-1-ND) until 999 unit
[631-1072-2-ND](https://www.digikey.com/products/en?keywords=%20631-1072-2-ND) for more than 999 units, with lead time

May be related with the information in last API. Waiting the version check.

I see that your are using the `currency` functionality. This is a tool that I write (and improve in last version). Is it working fine for you?

nickoe, 2019-07-12:
@hildogjr Sorry, I forgot to post the version info of kicost. I use master as of yesterday. add6294. You can see in the attached xlsx that it returns the dash 2 version.

hildogjr, 2019-07-12:
@nickoe, I gotcha. Test the new code and also the new commands `kicost --setup` / `kicost --unsetup` that add and remove KiCost from  KiCad BOM plugin list.
The new KiCost use the PartInfo by kitspace.org API as a collaborative funding but I also updated the legacy Octopart API abou this issue.
When this 2 commands above were confirmed we will officially release the new version.

closed 2019-07-12 by hildogjr

nickoe, 2019-07-12:
What is the point of the `--setup` and `--unsetup`? How are they related to my issue. What commit should I make sure I test?

hildogjr, 2019-07-12:
It is not related, I change the code in the API do fix your issue. Those commands are just the development news.

nickoe, 2019-07-13:
Ok, thank you. It seems like your fix worked 3f64ffd?

hildogjr, 2019-07-15:
The fix that really affect your issue was https://github.com/xesscorp/KiCost/commit/2fe8cad91e07cf924cb6bfe29e537c0b608eb956
The other is documentation and cosmetics before official release.