hildogjr/KiCost #368: list index out of range - when using dist# and kicost:dist:pricing

mdeweerd, 2019-08-27:
KiCost version 1.1.1 (installed using pip)

I get error
  File "/usr/lib/python3.7/site-packages/kicost/distributors/api_partinfo_kitspace.py", line 300, in query_part_info
    part_code_dist = list({k for k, v in dist_xlate.items() if v == part_catalogue_code_dist})[0]
IndexError: list index out of range

I found that this is because I have kicost:dist:pricing for one component and dist# for another component.

I enclose an example which fails and two variants of the example that pass.
The executed commands are in "tests.sh", the log is in tests.log.

mdeweerd, 2019-09-02:
Hi Hildago

I was convinced that I attached this example:
[testcase.zip](https://github.com/xesscorp/KiCost/files/3566530/testcase.zip)

Hopefully it is ok for you.

hildogjr, 2019-09-02:
It was due the "tme#". The new API Kitspace doesn't support it and it was trying to use.
Now the code check if the code is supported before use.

commit 5aa6cd8b (2019-09-02): Fix #368 - check by accepetd stock codes
commit 5a86ae1e (2019-09-02): Fix #368 - check by accepetd stock codes - history
closed 2019-09-02 by hildogjr