hildogjr/KiCost #371: Feature for Variant combinations

mdeweerd, 2019-10-05:
I propose to add a new kind of selector to the variant field "+" and "-" or "!" (I prefer the "-" notation).

The purpose is to be able to set fields according to the presence and/or absence of one or multiple variant values.

Examples:
- Multiple connectors (with different footprints) are possible for an input.  I need at least one of both, and one of them is the default.  Example: a terminal block is the default, a BNC input is the alternative.   I would set the variant "-inp1bnc" on the terminal block and "inp1bnc" on the BNC.
- I need to add a terminal block only if the specific analog input is actif and no BNC is mounted for that input.   I use "inp1" to enable the input and "inp1bnc" to enable the BNC, the terminal block should be added when "inp1-inp1bnc" or when "inp1!inp1bnc". 
- I need or can add the BNC only when input1 is active.  By specifying the other required variants it is easier to detect that some variants were forgotten and they allow for more complex component specifications..
I would specify "+inp1+inp1bnc"  or "inp1+inp1bnc"or to indicate that both variants are required for this component (the first "+" token is optional). 

This would of course also be available for "old style variant specifications" (see #370 ).

hildogjr, 2019-10-05:
Interesting concept. This would be applied to the "variant" field ("new" variant method used by KiCost).

mdeweerd, 2023-04-27:
Stale, closing (limiting my issue backlog).

closed 2023-04-27 by mdeweerd