hildogjr/KiCost #372: Cannot install Kicost on Windows

romain145, 2019-10-20:
and documentation link gives a 404 error.

#### Steps performed:
1. clone the Kicost repo
2. invoke python setup.py build
3. invoke python setup.py install
4. Invoke kicost

```
Z:\git\KiCost>kicost
Traceback (most recent call last):
  File "C:\Python37\Scripts\kicost-script.py", line 6, in <module>
    from pkg_resources import load_entry_point
  File "C:\Python37\lib\site-packages\pkg_resources\__init__.py", line 3191, in <module>
    @_call_aside
  File "C:\Python37\lib\site-packages\pkg_resources\__init__.py", line 3175, in _call_aside
    f(*args, **kwargs)
  File "C:\Python37\lib\site-packages\pkg_resources\__init__.py", line 3204, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "C:\Python37\lib\site-packages\pkg_resources\__init__.py", line 583, in _build_master
    ws.require(__requires__)
  File "C:\Python37\lib\site-packages\pkg_resources\__init__.py", line 900, in require
    needed = self.resolve(parse_requirements(requirements))
  File "C:\Python37\lib\site-packages\pkg_resources\__init__.py", line 786, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'babel>=2.6' distribution was not found and is required by kicost
```

Note: Wheel==0.23.0 has already been installed from requirement.py.
Are the other packages also required?

```
Z:\git\KiCost>pip install babel==2.6
Collecting babel==2.6
  Downloading https://files.pythonhosted.org/packages/b8/ad/c6f60602d3ee3d92fbed87675b6fb6a6f9a38c223343ababdb44ba201f10/Babel-2.6.0-py2.py3-none-any.whl (8.1MB)
     |████████████████████████████████| 8.1MB 3.3MB/s
Requirement already satisfied: pytz>=0a in c:\python37\lib\site-packages (from babel==2.6) (2019.2)
ERROR: kicost 1.1.2 requires beautifulsoup4>=4.3.2, which is not installed.
ERROR: kicost 1.1.2 requires CurrencyConverter>=0.13, which is not installed.
ERROR: kicost 1.1.2 requires future, which is not installed.
ERROR: kicost 1.1.2 requires lxml>=3.7.2, which is not installed.
ERROR: kicost 1.1.2 requires requests>=2.18.4, which is not installed.
ERROR: kicost 1.1.2 requires tqdm>=4.30.0, which is not installed.
ERROR: kicost 1.1.2 requires XlsxWriter>=0.7.3, which is not installed.
Installing collected packages: babel
Successfully installed babel-2.6.0
```

Should all these be added to the requirements.txt file?

romain145, 2019-10-20:
Note: after installing all these packages by hand plus wxPython, KiCost starts successfully.

hildogjr, 2019-10-20:
All KiCost requirements are listed by `requirements` variable at `setup.py` file. But I am not a expert in Python package and this file was present some issues, since I added the `post_install_setup()` function to run some after installation configurations (this version is beta and not officially on PyPI).

Basically, now it is needed to install the `pywin32` in the Windows case and run `kicost --setup` command line, that make all the configuration integration Eeschema-KiCost (add KiCost to BOm list, add the standards fields, save the typing from user, add a shortcut to KiCost in case of GUI installation, ...).

Is the command `kicost --setup` working for you? Do you mind to give a look on `setup.py`?

romain145, 2019-12-03:
Sorry I'm no expert in Python...
By following this, it installs the dependencies: https://stackoverflow.com/questions/1471994/what-is-setup-py

Now only wsPython is missing, which can be solved by uncommenting the line in setup.py.
After doing this and running `pip install .` KiCost can be launched.

hildogjr, 2019-12-03:
@romain145, which line did you uncomment? which OS did you try?
If Windows, did I automatically get `pywin32` python package? (`wsPython` is not automatically installed, some users ask me to do that, but I have to be asked on KiCost installation with you want to download thi package).

romain145, 2020-06-10:
Installation with pip fails on a new windows system.
Steps to reproduce:

1. Install Python 3.8.3, and add it to the environment variables
2. pip install kicost #this installs kicost1.1.4 and its dependencies
3. run kicost #this gives the following error
```
C:\Users\Romain>kicost
You don't have the wxPython dependence to run the GUI interface. Run once of the follow commands in terminal to install them:
pip3 install -U wxPython # For Windows & macOS
pip install -U -f https://extras.wxpython.org/wxPython4/extras/linux/gtk3/ubuntu-16.04 wxPython # For Linux 16.04
Or download from last version from <https://wxpython.org/pages/downloads/>
```
4. pip3 install -U wxPython
5. run kicost #this gives the following error
```
C:\Users\Romain>kicost
Traceback (most recent call last):
  File "C:\Users\Romain\AppData\Local\Programs\Python\Python38-32\Scripts\kicost-script.py", line 11, in <module>
    load_entry_point('kicost==1.1.4', 'console_scripts', 'kicost')()
  File "c:\users\romain\appdata\local\programs\python\python38-32\lib\site-packages\kicost\__main__.py", line 253, in main
    kicost_gui() # Use the user gui if no input is given.
  File "c:\users\romain\appdata\local\programs\python\python38-32\lib\site-packages\kicost\kicost_gui.py", line 1065, in kicost_gui
    frame = formKiCost(None)
  File "c:\users\romain\appdata\local\programs\python\python38-32\lib\site-packages\kicost\kicost_gui.py", line 367, in __init__
    fgSizer1.Add(self.m_gauge_process, 1, wx.ALL|wx.EXPAND|wx.ALIGN_CENTER_HORIZONTAL|wx.ALIGN_CENTER_VERTICAL, 5)
wx._core.wxAssertionError: C++ assertion "!(flags & (wxALIGN_BOTTOM | wxALIGN_CENTRE_VERTICAL)) || !(flags & (wxALIGN_RIGHT | wxALIGN_CENTRE_HORIZONTAL))" failed at ..\..\src\common\sizer.cpp(1453) in wxGridSizer::DoInsert(): wxEXPAND flag will be overridden by alignment flags```

hildogjr, 2020-06-10:
Please, report `kicost --info` information (with some library versions).

romain145, 2020-06-10:
`C:\Users\Romain>kicost --info
KiCost v.1.1.4.at Python 3.8.3.on Windows-10-10.0.19041-SP0(32bit).Graphical library: 4.1.0 msw (phoenix) wxWidgets
3.1.4.`

hildogjr, 2020-06-10:
Are KiCost working fine as CLI (terminal) interface?

FlYnN91, 2020-09-29:
@hildogjr I've just installed KiCost and seeing the same issue. Installed via pip. Seems like a long time for the Kitspace server to be offline?

`KiCost v.1.1.4.at Python 3.8.6.on Windows-10-10.0.19041-SP0(64bit).Graphical library: 4.1.0 msw (phoenix) wxWidgets
3.1.4.`

hildogjr, 2020-09-29:
Yes, before was reported two behaviours:

1. Not running even in CLI mode (server problem).
2. The GUI needs to be fix, so this issue here is duplicated from #412

hildogjr, 2020-09-29:
I will close this, since already exist a ticket open for.

closed 2020-09-29 by hildogjr