hildogjr/KiCost #374: Export component type (THT/SMD/Virtual)

mdeweerd, 2019-11-20:
I do not think there is currently a way to export the component type to the BOM output.
That would be usefull as some prototyping companies ask the number of THT/SMD lines/components or ask to identify them in the BOM after uploading that to their service (where they could extract the data from the BOM if present).

hildogjr, 2019-11-21:
Unfortunately the XML file from KiCad and others BOM files (Altium / CSV / Eagle / Proteus) doesn't have such information, making impossible to get the "virtual". About the THT/SMD, yes, would be very useful (I also confront this exactly need).
There a plan (roadmap) to include the [electro-grammar](https://github.com/kitspace/electro-grammar) into KiCost making possible (internal) check footprint and values and implement #4. I am just delayed with the current bugs.

About count the parts, I tried an approach one month ago using `regex(foo)` function but despise available at Google Spreadsheets and LibreOffice Calc, it is not on Microsoft Excel (KiCost spreadsheet should to be opened in those three software). If you give some idea how to dynamic count R/C/... in the spreadsheet (if a delete a line, the value should be updated), should be very useful to this issue.

romain145, 2019-12-04:
@mdeweerd All that's available in the XML file (generated from Eeschema) is the footprint link, where the information you're looking for is embedded within the footprint itself.

As a workaround, you way want to have a look at the exports from Pcbnew. For example: the .rpt , .cmp and in particular the .pos which only contains SMD parts.

mdeweerd, 2019-12-04:
I did not examping the xml format in detail - there is some logic in the fact that this information is not in the xml file, but it is useful to have it in the BOM.
The information does seem to be included in the report file (*.rpt).

mdeweerd, 2023-04-27:
Stale, closing (limiting my issue backlog).

closed 2023-04-27 by mdeweerd